espressif/esp-idf #9625: Unable to configure UART1 for CONSOLE output

aggaddam, 2022-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Environment:

    Development Kit: ESP32-C3
    Kit version: DevKitM-1: v1
    Module or chip used: ESP32-C3-MINI-1
    IDF version: v4.4.2(stable)
    Build System: idf.py
    Operating System: Ubunut
    Using an IDE?: No
    Power Supply: USB

Problem Statement:
I am unable to configure the UART1 for console output using menuconfig. Here is the screenshot of settings from menuconfig

[image]

Though I selected custom, it is not allowing me to select the UART port. It always points to default port UART0. Is there any limitation in using the second UART port i.e. UART1 in ESP32 C3 ?

igrr, 2022-08-24:
This feature is indeed not supported on ESP32-C3 or ESP32-C2. However since the two UARTs are identical, why do you need to use specifically UART1 for the console output? The pins for both UARTs can be freely configured. If you don't want to see the boot ROM output on your console UART, this can be disabled via an eFuse.

aggaddam, 2022-08-24:
@igrr I need to use two uart ports for my application. 1) for console output for debugging 2) Communication with another mcu

igrr, 2022-08-24:
Yes, on ESP32-C3 and ESP32-C2 only the UART0 can be used for the console output. I think you can keep using UART0 for console output (and set up the TX/RX pins as necessary). The UART1 can then be used for communication with another MCU.

aggaddam, 2022-08-24:
@igrr Got it. Will try it. Appreciate your fast response!

aggaddam, 2022-08-24:
@igrr It's working fine. I am closing this issue

closed 2022-08-24 by aggaddam