espressif/esp-idf #9626: IDF 5.0 beta1 Monitor Colors

ifedotov1984, 2022-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.0-beta1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expected to get colorized output of logs (like at esp-idf-v4.4.1)

### What is the actual behavior?

[image]

Instead of colorized logs (as worked in IDF 4.4.1) I got codes [0;33m and similar ([0;32m ... etc)

### Steps to reproduce.

1. clone esp-idf-v5.0-beta1 from git
2. install esp-idf-tools-setup-online-2.16.exe with local idf from step1 (install IDF IDE, command prompt,  drivers and chip targets)
3. run install.bat and export.bat
4. from IDF CMD run 'idf.py monitor' (in project folder)

igrr, 2022-08-24:
This is possibly the same bug as reported in #9610.

dobairoland, 2022-08-25:
I'm closing this as duplicate. Please follow the linked issue for updates.

closed 2022-08-25 by dobairoland