espressif/esp-idf #9629: Read/Write data to ESP32 from PC over USB

tobyloki, 2022-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to build a desktop application that will be able to retrieve details such as the device id, hardware version, firmware version, and model from an ESP32 device. How would I be able to do that?

chegewara, 2022-08-25:
https://github.com/espressif/esp-idf/tree/master/examples/system/console

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/console.html

0xjakob, 2022-09-07:
@tobyloki I don't have a lot to add to @chegewara's comment. Some basic information (MAC, flash ID, etc.) can be retrieved with `esptool.py`. Of course, you can implement your own protocol over UART to communicate with an ESP as well, but the linked console application might be more straight-forward.

If you need more support, feel encouraged to post in the [ESP Forum](https://esp32.com) as this is meant to discuss programming questions like these. You'll probably get more help there.

Besides the implementation, if you have issues communicating with the ESP-chip please let us know! Otherwise, we would like to close this issue.

0xjakob, 2022-11-30:
@tobyloki we're closing this issue. Feel free to re-open if the aforementioned approaches didn't work for you.

closed 2022-11-30 by 0xjakob