espressif/esp-idf #9630: Add the mqtt component when using the idf_as_lib example,cmake Error .

mrheqie, 2022-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/4.4

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### What is the expected behavior?

Add the mqtt component when using the idf_as_lib example and  buid ok

### What is the actual behavior?

cmake reports an error: (Target "idf_as_lib.elf" links to target "idf::mqtt" but the target was not found )

### Steps to reproduce.

```
cmake_minimum_required(VERSION 3.5)

project(idf_as_lib C)

list(APPEND DH_MODEL System)

if("${TARGET}" STREQUAL "esp32c3")
    # Include for ESP-IDF build system functions
    include($ENV{IDF_PATH}/tools/cmake/idf.cmake)
    include_directories(dahao)
    # Create idf::esp32 and idf::freertos static libraries
    idf_build_process("${TARGET}"
                    # try and trim the build; additional components
                    # will be included as needed based on dependency tree
                    #
                    # although esptool_py does not generate static library,
                    # processing the component is needed for flashing related
                    # targets and file generation
                    COMPONENTS "${TARGET}" freertos esptool_py
                    SDKCONFIG ${CMAKE_CURRENT_LIST_DIR}/sdkconfig
                    BUILD_DIR ${CMAKE_BINARY_DIR})
                    
                    foreach(_dh_model ${DH_MODEL})
                        add_subdirectory(dahao/${_dh_model})
                    endforeach(_dh_model)
else()
    # Create stubs for esp32 and freertos, stub::esp32 and stub::freertos
    add_subdirectory(stubs/esp32)
    add_subdirectory(stubs/freertos)
    add_subdirectory(stubs/spi_flash)
endif()

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

set(elf_file ${CMAKE_PROJECT_NAME}.elf)
add_executable(${elf_file} main.c)

# Link the static libraries to the executable
if("${TARGET}" STREQUAL "esp32c3")
    target_link_libraries(${elf_file} "idf::${TARGET}"  idf::freertos idf::spi_flash  idf::mqtt ${DH_MODEL})
    # Attach additional targets to the executable file for flashing,
    # linker script generation, partition_table generation, etc.
    idf_build_executable(${elf_file})
else()
    target_link_libraries(${elf_file} stub::esp32 stub::freertos stub::spi_flash)
endif()
```

### Build or installation Logs.

```plain
-- Adding linker script /home/dahao/prj/esp_idf/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Configuring done
CMake Error at CMakeLists.txt:36 (add_executable):
  Target "idf_as_lib.elf" links to target "idf::mqtt" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
```

### More Information.

none

DNedic, 2022-08-29:
Thank you for submitting your issue.

It appears you forgot to add mqtt to the COMPONENTS argument of the idf_build_process() function.

There however does seem to be an issue on our end also requiring manual adding of the tcp_transport component, so please add it as well until we investigate this.

mrheqie, 2022-08-29:
> Thank you for submitting your issue.
> 
> It appears you forgot to add mqtt to the COMPONENTS argument of the idf_build_process() function.
> 
> There however does seem to be an issue on our end also requiring manual adding of the tcp_transport component, so please add it as well until we investigate this.

thanks

DNedic, 2022-10-06:
@mrheqie The issue has been fixed and is being backported to ESP-IDF 4.4 and 4.3 as well, you should no longer need to add the tcp-transport component manually.

dobairoland, 2022-10-10:
Fixes will be included in the next bugfix releases of 4.4 and 5.0; and in the initial release of v5.1. Probably no fix will be included for v4.3 because only high priority issues are accepted there as per the maintenance period policy. 

Please report separately any possible future build system warnings. We will fix them instead of adding possibility to ignoring them.

closed 2022-10-10 by dobairoland