espressif/esp-idf #9631: Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled in ESP32C3

aggaddam, 2022-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2 (stable)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

ESP32C3 crashes when connecting to Wi-Fi AP. This shouldn't happen

### What is the actual behavior?

ESP32 C3 crashes with load access fault when configured as station and tries to connect to the Access point. (Same code works in ESP32 DEV kit without any issues). Any thoughts on this issue?

SDK Config:

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_RISCV=y
CONFIG_IDF_TARGET="esp32c3"
CONFIG_IDF_TARGET_ESP32C3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0005

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="riscv32-esp-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
[... 1515 lines cut ...]
3fc9f450: 0x00000005 0x00000000 0x3fc94da8 0x420606fe 0x00000001 0x00000001 0x3fca5458 0x00000000
3fc9f470: 0x00000001 0x00000010 0x3fca5458 0x40387da0 0x00000001 0xffffffff 0x00000000 0x00000000
3fc9f490: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000001 0x00000040 0x00000000 0x00000000
3fc9f4b0: 0x00000040 0x3fc8fdbc 0x3fc8fdc8 0x420605be 0x3fc72d00 0x3fca554c 0x00000001 0x00000001
3fc9f4d0: 0x00000000 0x00000000 0x3fc98ae4 0x00000000 0x00000008 0x00000008 0x3fc9f7a8 0x3fca54e8
3fc9f4f0: 0x00000000 0x00000040 0x00000008 0x3fca554c 0x00000000 0x00000040 0x3fca54e0 0x4205e112
3fc9f510: 0x00000004 0x00000000 0x3fc98ae4 0x00001000 0x00000003 0x3fc93114 0x00000008 0x3fca554c
3fc9f530: 0x3fc9313b 0x3fc9f674 0x00000000 0x420570be 0x3fc9fa08 0x00000001 0x3fc98ae4 0x403878f8
3fc9f550: 0x0000000a 0x3fc96216 0x0000000a 0x00000036 0x3fc9fa08 0x00000001 0x3fc98ae4 0x403878f8
3fc9f570: 0x00000000 0x3fc96216 0x3fc8ea08 0x420034ce 0x00000044 0xffffffff 0x3fc93217 0x00000014
3fc9f590: 0x00000020 0x3fc9f674 0x3fc9313b 0x4205810c 0x00000036 0x3fc9fa64 0x00000008 0x3fc9313b
3fc9f5b0: 0x00000000 0x3fc96206 0x3fc96070 0x01000000 0x0000000a 0x00000000 0xc0380000 0x3fc95c60
3fc9f5d0: 0x00000084 0x00000080 0x3fc95580 0x4038bfe0 0x0000000a 0x3fc9fa64 0x3fc95edc 0x00000080
3fc9f5f0: 0x00000004 0x00000080 0x00000804 0x40380a7c 0x00000000 0x00000000 0x3fc96070 0x00000000
3fc9f610: 0x00000008 0x00000008 0x3fc9f7a8 0x3fc95000 0x00000001 0x3fc9313b 0x3fc93217 0x00001000
3fc9f630: 0x00000804 0x00000080 0x00000000 0x40380a96 0x00000001 0x00000008 0x3fc9f7a8 0x3fc95000
3fc9f650: 0x00000001 0x3fc9313b 0x3fc93217 0x00001000 0x00000003 0x3fc93114 0x3fc9313b 0x42025caa
3fc9f670: 0x00000001 0x3c0c6c3c 0x3fca54e0 0x3fca554c 0x00000001 0x3fc9313b 0x3fc937ec 0x3fc9313b
3fc9f690: 0x3fc93110 0x3fc9525c 0x3fc95000 0x4208b5ee 0x00000000 0x3fca5418 0x3fc95580 0x00000003
3fc9f6b0: 0x00000003 0x3fc9313b 0x3fc937ec 0x3fc93114 0x00000003 0x3fc95000 0x3fc9525c 0x4208b844
3fc9f6d0: 0x00ce0000 0x00000000 0x3fc94ab8 0x3c0ba0c0 0x00000004 0x00000018 0x00000000 0x4209dfc6
3fc9f6f0: 0x00000004 0x3fc9f75c 0x420277f8 0x00000000 0x3fcdf934 0x3fce0000 0x00000001 0x3fca4d94

ELF file SHA256: 69f45d3de5c717d5
```

aggaddam, 2022-08-25:
Here is the SDK config settings from menuconfig

SDK Config:

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_RISCV=y
CONFIG_IDF_TARGET="esp32c3"
CONFIG_IDF_TARGET_ESP32C3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0005

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="riscv32-esp-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x0
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
[... 1247 lines cut ...]
CONFIG_TCP_WND_DEFAULT=5744
CONFIG_TCP_RECVMBOX_SIZE=6
CONFIG_TCP_QUEUE_OOSEQ=y
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

DirtYiCE, 2022-12-07:
Just run into this issue myself. For me it crashes here: https://github.com/espressif/esp-idf/blob/454aeb3a48ac2b92cfa9d8b6a01d1b53179ec50a/components/esp_hw_support/dma/gdma.c#L289-L290
`%zu` is C99 which is not supported in nano format. I think it just ignores the `%zu` and moves on to the next `%s` and crashes because the `sram_alignment` is not a valid pointer to a c-style string...
I can workaround this issue by setting the log level higher, but a quick search on esp-idf came up with a bunch of results for `%zu`. I think the only safe way to use CONFIG_NEWLIB_NANO_FORMAT right now is to disable logging completely.

AxelLin, 2023-03-01:
Any update for the fix? @zikalino

dmezh, 2023-03-27:
Pretty sure I ran into this too (ESP32-S3, NEWLIB_NANO_FORMAT=y, log level = verbose). This seems like it's happening in the middle of an `esp_ota_set_boot_partition` op:

```
D (1704) gdma: new group (0) at 0x3fcee660
D (1704) gdma: new pair (0,0) at 0x3fcee69c
D (1704) gdma: new tx channel (0,0) at 0x3fcee630
D (1704) gdma: new rx channel (0,0) at 0x3fcee6bc
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40056ea0  PS      : 0x00060130  A0      : 0x8200a756  A1      : 0x3fc98f60
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3fc99003
A6      : 0x42009a2c  A7      : 0x00000000  A8      : 0x3fcf0721  A9      : 0x3fcf0722
0x42009a2c: __sfputs_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/nano-vfprintf.c:407

A10     : 0x3fcee02c  A11     : 0x00000020  A12     : 0x3fcf0388  A13     : 0x3fc99003
A14     : 0x3fc98fd9  A15     : 0x00000005  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000001  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x40056e9d:0x3fc98f60 |<-CORRUPTED
```

aggaddam, 2023-03-27:
Based on my past experience, Disabling NEWLIB_NANO_FORMAT might fix your issue

AxelLin, 2023-04-10:
@zikalino @igrr 
Can someone help to fix this panic?

Hit below panic with v5.1-dev-4375-g42261df71f (ESP32-C3, NEWLIB_NANO_FORMAT=y, log level = verbose)
D (7804) gdma: new group (0) at 0x3fcbf4ac
D (7808) gdma: new pair (0,0) at 0x3fcbf4e0
D (7812) gdma: new tx channel (0,0) at 0x3fcbf47c
D (7817) gdma: new rx channel (0,0) at 0x3fcbf500
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x40058766  RA      : 0x420d7ab6  SP      : 0x3fcb65a0  GP      : 0x3fc94400
0x420d7ab6: _printf_i at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/nano-vfprintf_i.c:221

TP      : 0x3fc72c68  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0xffffffff
S0/FP   : 0x3fcb65e0  S1      : 0x3fcb69a4  A0      : 0x00000001  A1      : 0x00000000
A2      : 0x00000000  A3      : 0x3fcb66c4  A4      : 0x3fcb65d8  A5      : 0x3fcb66c0
A6      : 0x3c12f050  A7      : 0x00000073  S2      : 0x3fc9e394  S3      : 0x420d6d48
0x420d6d48: __sfputs_r at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/nano-vfprintf.c:407

S4      : 0x00000001  S5      : 0x00000001  S6      : 0x3c12f000  S7      : 0x3c12f000
S8      : 0x420d7000  S9      : 0x00000025  S10     : 0x3c12f000  S11     : 0x00000001
0x420d7000: _vfprintf_r at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/nano-vfprintf.c:609

T3      : 0x00000000  T4      : 0x00001dce  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000001
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Backtrace:

0x40058766 in ?? ()
#0  0x40058766 in ?? ()
#1  0x420d7ab6 in _printf_i (data=0x3fcb69a4, pdata=0x3fcb65e0, fp=0x3fc9e394, pfunc=0x420d6d48 <__sfputs_r>, ap=<optimized out>) at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/nano-vfprintf_i.c:219
#2  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: 6c518fd84ce5f65b

Rebooting...
ESP-ROM:esp32c3-api1-20210207

aggaddam, 2023-04-10:
@AxelLin 

NEWLIB_NANO_FORMAT=y might be causing this issue. 
Please try disabling NEWLIB_NANO_FORMAT i.e NEWLIB_NANO_FORMAT=n

AxelLin, 2023-04-10:
> @AxelLin
> 
> Enabling nano lib format i.e.NEWLIB_NANO_FORMAT=y might be causing this issue. Please try disabling NEWLIB_NANO_FORMAT i.e NEWLIB_NANO_FORMAT=n

@aggaddam NEWLIB_NANO_FORMAT=n is a workaround, not a real fix!

closed 2023-05-08 by espressif-bot

josef2600, 2023-05-10:
hi, i am working with **Arduino**, and nop! it is not fixed. i have it when using info.
`Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.`
i don't know how can i disable `NEWLIB_NANO_FORMAT` either. any help would be appreciated. thanks.

aggaddam, 2023-05-11:
@josef2600 You can try the below steps to disable it

1.  idf.py menuconfig  → Component config → Newlib 
2. Uncheck Enable Nano Formatting if checked already

[image]

ESP-Marius, 2023-05-11:
@josef2600 As far as I know the arduino library does not come with `NEWLIB_NANO_FORMAT` active. Are you sure this is the same issue?

But yes, in general any fix merged into esp-idf will need some time before they appear in Ardunio.

igrr, 2023-05-11:
To add to what @ESP-Marius said: In general, "load access fault" occurs in every case when the program tries to read from an invalid memory address. Please see [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/fatal-errors.html#instruction-access-fault-load-access-fault-store-access-fault) for how to interpret the panic handler output.

josef2600, 2023-05-12:
> @josef2600 You can try the below steps to disable it
> 
>     1. idf.py menuconfig  → Component config → Newlib
> 
>     2. Uncheck Enable Nano Formatting if checked already
> 
> 
> [image]

thanks, but i am not on IDF, i work with Arduino.

> @josef2600 As far as I know the arduino library does not come with `NEWLIB_NANO_FORMAT` active. Are you sure this is the same issue?
> 
> But yes, in general any fix merged into esp-idf will need some time before they appear in Ardunio.

thanks, and yes it has it. i found them in libraries, they are using them to see if it is defined, then change some registers. but i didn't fined where the main thing is defined. just found thinks like` #ifdef NEWLIB_NANO_FORMAT` .

> To add to what @ESP-Marius said: In general, "load access fault" occurs in every case when the program tries to read from an invalid memory address. Please see [...]

thank you for your info, but this is a little beside the point. i am using uart in my code and using it for debug (info) too. it hangs while using the serial port. read this issue and you get the point. i have search for a week  in internet and in datasheet and nothing works for preventing for clashing the uart. since esp32 uses RTOS and it is using serial port and i am using serial port, it is possible that program tries to access the serial port from different parts of program in the same time and it causes the chip to crash. at lest, it is my theory! in other microcontrollers, i can check to see if uart is busy or not then send data to it. but in Arduino, i couldn't find the damn thing!

ESP-Marius, 2023-05-15:
> thank you for your info, but this is a little beside the point. i am using uart in my code and using it for debug (info) too. it hangs while using the serial port. read this issue and you get the [...]

Not sure exactly what you are trying to do here, but normally if you want to share and protect a resource, e.g. UART, in  an RTOS you would use a [mutex](https://www.freertos.org/a00113.html)

Seems like that for UART the ardunio code already does take a mutex for you, e.g. see [this](https://github.com/espressif/arduino-esp32/blob/3ec5f4efa1de4342aaab742008dc630091e5e035/cores/esp32/esp32-hal-uart.c#L419).

If you have question specifically about the arduino port then you might be better of asking at https://github.com/espressif/arduino-esp32

josef2600, 2023-05-16:
thank you. it is fixed now. i used xQueueSend and xQueueReceive, and now it is perfectly working. and it is fast! so far no problem with any debug or anything eles.