espressif/esp-idf #9632: USB issue on ESP32S2 MINI1

lalith10112000, 2022-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have tried to run the example files for "USB DEVICE" like usb_serial_device and usb_console on ESP32-S2FH4. I flashed the code and connected the device to my PC. I got an error "CODE 43 : DEVICE DESCRIPTOR REQUEST FAILED". I tried with multiple PCs, with multiple ESP32-S2FH4 boards and with multiple USB cables. The issue remains the same. I went in depth into the example files and there was a task that logs the "event-id" of the USB.

Event id is 1 in case of new USB connection, 6 in case PC sends data to the device, 7  if data from the ESP chip is transmitted. What I observed from the logs is that event id is continuously remaining "1" even after connection has been made and not switching to "6". This is a clear indication that ESP is not taking the request sent by PC. How can I resolve this?

Below is the link of the examples projects I used:

[https://github.com/espressif/esp-idf/tree/ae256e9ca23a85e05f6a9ff31805028d3e823f4d/examples/peripherals/usb/device](url)

tore-espressif, 2022-08-29:
Hello @lalith10112000 , 

I'm sorry to hear that it doesn't work for you. Did you make any changes in the example projects? Could you please set TinyUSB log level to 3 in menuconfig and share a full log?

And please double check your wiring; D+ is GPIO20 and D- is GPIO19.

lalith10112000, 2022-08-29:
The issue is resolved. Its just that we dumped 0 ohms resistors and capacitors on the d+ and d- lines along with a diode IC which led to this issue! Removing all of them solved the issue we faced.

closed 2022-08-29 by lalith10112000