espressif/esp-idf #9633: BLE Mesh Send and receive String MSG to all nodes

JegadeeshXaglabs, 2022-08-25:
### Is your feature request related to a problem?

Hi Developers, 
I need a help in ESP32 BLE mesh. I need to send and receive String MSG in all nodes .

What i have done:
I tried on off server and client. So i had a basic knowledge in BLE mesh. I can create a mesh network which can control all nodes with the client.

What i need 
 i have to send and receive a string msg to all nodes .i tried vendor module and sensor module with that example i can only send msg from client to server. Even with that i receive client msg in the last server node which is connected to the client.

hipal31, 2022-08-25:
Hello @JegadeeshXaglabs 
I'm currently using vendor model for communication(string messages) between nodes.
Using `esp_ble_mesh_server_model_send_msg()` function, all the nodes can communicate with each other.

JegadeeshXaglabs, 2022-08-26:
> Hello @JegadeeshXaglabs I'm currently using vendor model for communication(string messages) between nodes. Using `esp_ble_mesh_server_model_send_msg()` function, all the nodes can communicate with each other.
i tried vendor module and sensor module with that example i can only send msg from client to server. Even with that i receive client msg in the last server node which is connected to the client. it

It shows  BLE_MESH: No matching TX context for ack

hipal31, 2022-08-26:
Did you add opcode of both request and response in vendor model in all node?

If possible please share code snippet of vendor model and custom callback.

JegadeeshXaglabs, 2022-08-26:
> Did you add opcode of both request and response in vendor model in all node?
> 
> If possible please share code snippet of vendor model and custom callback.

Thanks for your reply.

Can you tell me how to add op code.
Actually i am new to BLE mesh.

hipal31, 2022-08-26:
[Check this example](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_vendor_model/vendor_server/main/main.c#L62)

JegadeeshXaglabs, 2022-08-26:
Thank you i have added opcode but i got these error
 BLE_MESH: Too short message for OpCode 0x00c002e7

hipal31, 2022-08-26:
`    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_SEND, 2)`

set minimum message size while registering opcode using ESP_BLE_MESH_MODEL_OP()
Keep it 0 for testing

JegadeeshXaglabs, 2022-08-26:
> ` ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_SEND, 2)`
> 
> set minimum message size while registering opcode using ESP_BLE_MESH_MODEL_OP() Keep it 0 for testing

if i put 0 again i get **BLE_MESH: No matching TX context for ack** in the 1st node

hipal31, 2022-08-26:
Actually you'll have to add every opcode in vnd_op[].

something like this
```
static esp_ble_mesh_model_op_t vnd_op[] = {
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_SEND, 0),
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_STATUS, 0),
// add more here
    ESP_BLE_MESH_MODEL_OP_END,
};
```

JegadeeshXaglabs, 2022-08-26:
> 
[image]
This is the error i got

hipal31, 2022-08-26:
You'll have to bind that model also,

Ignore that error right now.

JegadeeshXaglabs, 2022-08-26:
> > 
> 
> [image] This is the error i got

@

> You'll have to bind that model also,
> 
> Ignore that error right now.

hipal31, 2022-08-26:
@JegadeeshXaglabs Did it work?

JegadeeshXaglabs, 2022-08-26:
Nope @hipal31 accidently closed that issue i am reopening it.
 i bind the all the keys

hipal31, 2022-08-26:
please share a initialization code snippet

JegadeeshXaglabs, 2022-08-26:
[image]
server side

hipal31, 2022-08-26:
send root_models elements

JegadeeshXaglabs, 2022-08-29:
> send root_models elements
Thanks @hipal31 
I am using vendor module example. Only enable .relay function and added op codes

JegadeeshXaglabs, 2022-09-02:
Is anyone tried send and receive string msg

JegadeeshXaglabs, 2022-09-06:
> send root_models elements

static esp_ble_mesh_model_t root_models[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
};

hipal31, 2022-09-06:
Did you add elements ?
static esp_ble_mesh_elem_t elements[] = {
    ESP_BLE_MESH_ELEMENT(0, root_models, vnd_models),
};

JegadeeshXaglabs, 2022-09-06:
> Did you add elements ? static esp_ble_mesh_elem_t elements[] = { ESP_BLE_MESH_ELEMENT(0, root_models, vnd_models), };

No

hipal31, 2022-09-06:
Can you share your whole code? Because I'm not able to understand implementation done by you.

JegadeeshXaglabs, 2022-09-07:
> Hello @JegadeeshXaglabs I'm currently using vendor model for communication(string messages) between nodes. Using `esp_ble_mesh_server_model_send_msg()` function, all the nodes can communicate with each other.

Can you share your code

JegadeeshXaglabs, 2022-09-19:
I just added all the node address in this void example_ble_mesh_send_vendor_message(bool resend); function. Thanks for the help

closed 2022-09-19 by JegadeeshXaglabs

Ambika-Devi-shobha, 2022-12-13:
@hipal31 can you please help me out  in adding unprovisioned device to ble mesh network. which model is best suited from examples? I m using fast provisioning model. Can you plzz give some idea so that I can provisioning more than one device at a time

Ambika-Devi-shobha, 2022-12-13:
> I just added all the node address in this void example_ble_mesh_send_vendor_message(bool resend); function. Thanks for the help

We're be able get this working

hipal31, 2022-12-13:
> @hipal31 can you please help me out in adding unprovisioned device to ble mesh network. which model is best suited from examples? I m using fast provisioning model. Can you plzz give some idea so that I can provisioning more than one device at a time

@Ambika-Devi-shobha I haven't tried multiple device provisioning at a time yet.

Ambika-Devi-shobha, 2022-12-13:
> @Ambika-Devi-shobha I haven't tried multiple device provisioning at a time yet.

Thanks, but is it possible? 
One more help while using fast provisioning model of esp-idf I'm getting this error
" BLE_MESH:Too short message for opCode"
Plz help me with this error...

hipal31, 2022-12-13:
@Ambika-Devi-shobha Yes, it's possible but haven't tried yet.

for the error, please modify message length provided in esp_ble_mesh_model_op_t structure variable.

Ambika-Devi-shobha, 2022-12-13:
> @Ambika-Devi-shobha Yes, it's possible but haven't tried yet.
> 
> for the error, please modify message length provided in esp_ble_mesh_model_op_t structure variable.

Thank you so much for help.. 
If possible can you please suggest me some reference for provisioning more devices. I tried in vendor model but unsuccessful...!

hipal31, 2022-12-13:
> Thank you so much for help.. If possible can you please suggest me some reference for provisioning more devices. I tried in vendor model but unsuccessful...!

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_fast_provision
Have you tries this?

Ambika-Devi-shobha, 2022-12-13:
> https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_fast_provision Have you tries this?

Yes what I did is modified max_node_num=0x02
So that I can connect to two server

hipal31, 2022-12-13:
> Yes what I did is modified max_node_num=0x02 So that I can connect to two server

In the given example it's 6 so it should work

Ambika-Devi-shobha, 2022-12-13:
> In the given example it's 6 so it should work
didn't understand can elaborate nore plzz

Ambika-Devi-shobha, 2023-01-05:
did anyone try to add more than one node to ble_mesh_vendor_model in order to achieve this kind of communication using BLE mesh?
[image]

Ambika-Devi-shobha, 2023-01-05:
> did anyone try to add more than one node to ble_mesh_vendor_model in order to achieve this kind of communication using BLE mesh? [image]

JegadeeshXaglabs, 2023-01-06:
In void e**xample_ble_mesh_send_vendor_message(bool resend)** function  you have to pass the nodes ID in **store.server_addr** using For loop.Like if your 1st node has a address 0X0005 and 2nd node has a 0X0006 like wise for 5 nodes
you have to write a for loop which can run 5 times each time you have to increase and store address in (**store.server_addr++** )

Ambika-Devi-shobha, 2023-01-06:
> In void e**xample_ble_mesh_send_vendor_message(bool resend)** function you have to pass the nodes ID in **store.server_addr** using For loop.Like if your 1st node has a address 0X0005 and 2nd node [...]

Thank you @JegadeeshXaglabs I will try it and get back if doubt still persist.

Ambika-Devi-shobha, 2023-01-18:
> In void e**xample_ble_mesh_send_vendor_message(bool resend)** function you have to pass the nodes ID in **store.server_addr** using For loop.Like if your 1st node has a address 0X0005 and 2nd node [...]

@JegadeeshXaglabs is it possible for you to elaborate more on this, please? I am bit confused.

Ambika-Devi-shobha, 2023-01-25:
> In void e**xample_ble_mesh_send_vendor_message(bool resend)** function you have to pass the nodes ID in **store.server_addr** using For loop.Like if your 1st node has a address 0X0005 and 2nd node [...]

I tried using for loop but not working. I have attached the code snippet where I made changes can you plz check if I am doing it right or not? can I get help in this regard, please, and should I keep the uuid of all nodes same?

for(int i=0;i<5;i++)
        {
            ctx.addr = store.server_addr;
            if (resend == false) 
            {
                store.vnd_tid++;
            }
            err = esp_ble_mesh_client_model_send_msg(vendor_client.model, &ctx, opcode,
            sizeof(store.vnd_tid), (uint8_t *)&store.vnd_tid, MSG_TIMEOUT, true, MSG_ROLE);
            store.server_addr++;
        }

Ambika-Devi-shobha, 2023-03-02:
hello people,
I am still stuck with this can someone please give me some suggestions?
I wanted to make this kind of connection using BLE mesh between (A, B, C, D, E, and Q). currently, I am trying to establish this network using the vendor model but am unable to do. since I don't know much about BLE mesh, I couldn't solve this.
Any people who worked on the BLE mesh projects here.. can I get help with that regard, please.

hoping for a quick reply!! ASAP

[image]