espressif/esp-idf #9634: BLE warning after connection try -> disconnected

jrHernandezSio, 2022-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

ENVIRONMENT:

espressif32@3.3.2 -> framework-espidf @ 3.40300.0 (4.3.0)
BLE driver: Bluedroid (loaded dynamically)

DESCRIPTION

Sometimes I get a warning when trying to connect adn smartphoe to ESP32.

I have never got this problem until I have started to run BLE in a parallel task.

See below how after 2,3 sec. connection is up the device get disconnected.

[8/25 13:15:23.6]-DEBUG-: [Thu Aug 25 11:15:20 2022] BLE stat: BLUDRD(0) BT (0) 
[8/25 13:15:23.6]-DEBUG-: [Thu Aug 25 11:15:20 2022] SESSION (2): RFID 
[8/25 13:15:23.6] ID usuario []
[8/25 13:15:23.6]-DEBUG-: [Thu Aug 25 11:15:20 2022] Getting picture...
[8/25 13:15:23.6]-DEBUG-: [Thu Aug 25 11:15:20 2022] Expecting ini...
[8/25 13:15:24.0]-DEBUG-: [Thu Aug 25 11:15:20 2022] 
[8/25 13:15:24.0]Token_encrypt(Reg) = 0x E6 F6 5A D0 48 8F 9A D7 D1 F8 8B 4A 5E 63 A7 C3 CD 86 87 A8 48 4F 51 A3 
[8/25 13:15:24.0]-DEBUG-: [Thu Aug 25 11:15:20 2022] 
[8/25 13:15:24.0]-DEBUG-: [Thu Aug 25 11:15:20 2022] BLE stat: BLUDRD(2) BT (2) 
[8/25 13:15:24.0]-DEBUG-: [Thu Aug 25 11:15:20 2022] BLE connecting...
[8/25 13:15:24.0]-DEBUG-: [Thu Aug 25 11:15:20 2022] BLE conn ......... (0)
[8/25 13:15:24.1]-DEBUG-: [Thu Aug 25 11:15:20 2022] Expecting ini. buff (len=51): VC0703 1.00
[8/25 13:15:24.1]Ctrl infr exist
[8/25 13:15:24.1]OV7720
[8/25 13:15:24.1]525
[8/25 13:15:24.1]Init end
[8/25 13:15:24.1]-DEBUG-: [Thu Aug 25 11:15:20 2022] Camera Found
[8/25 13:15:24.2]-DEBUG-: [Thu Aug 25 11:15:20 2022] BLE conn ......... (0)
**[8/25 13:15:24.3]-DEBUG-: [Thu Aug 25 11:15:20 2022] set 3 for gatts_if_for_indicate**
[8/25 13:15:24.4]-DEBUG-: [Thu Aug 25 11:15:20 2022] BLE conn ......... (0)
[8/25 13:15:24.6]-DEBUG-: [Thu Aug 25 11:15:21 2022] BLE conn ......... (0)
[8/25 13:15:24.8]-DEBUG-: [Thu Aug 25 11:15:21 2022] BLE conn ......... (0)
[8/25 13:15:24.9]-DEBUG-: [Thu Aug 25 11:15:21 2022] RFID init...
[8/25 13:15:24.9]-DEBUG-: [Thu Aug 25 11:15:21 2022] RFID_READER init...
[8/25 13:15:24.9]-DEBUG-: [Thu Aug 25 11:15:21 2022] OK -> Cfg: burst (0xFF) | recal (20)
[8/25 13:15:24.9]-DEBUG-: [Thu Aug 25 11:15:21 2022] RFID detecting...CalVal: 0x70 
[8/25 13:15:25.0]-DEBUG-: [Thu Aug 25 11:15:21 2022] TIMEOUT: SPI device timeout
[8/25 13:15:25.0]-DEBUG-: [Thu Aug 25 11:15:21 2022] RFID stop...
[8/25 13:15:25.0]-DEBUG-: [Thu Aug 25 11:15:21 2022] BLE conn ......... (0)
[8/25 13:15:25.2]-DEBUG-: [Thu Aug 25 11:15:21 2022] BLE conn ......... (0)
[8/25 13:15:25.4]-DEBUG-: [Thu Aug 25 11:15:21 2022] INT_L1 pin: 1. INT_RFID: 0 Cause (13)
[8/25 13:15:25.4]-DEBUG-: [Thu Aug 25 11:15:21 2022] LED dc: 50 %
[8/25 13:15:25.4]-DEBUG-: [Thu Aug 25 11:15:21 2022] pin_mask: 0xFF , current_mask: 0xEF
[8/25 13:15:25.4]-DEBUG-: [Thu Aug 25 11:15:21 2022] Getting BarCode...
[8/25 13:15:25.4]-DEBUG-: [Thu Aug 25 11:15:21 2022] BLE conn ......... (0)
[8/25 13:15:25.6]-DEBUG-: [Thu Aug 25 11:15:22 2022] BLE conn ......... (0)
[8/25 13:15:25.8]-DEBUG-: [Thu Aug 25 11:15:22 2022] BLE conn ......... (0)
[8/25 13:15:26.0]-DEBUG-: [Thu Aug 25 11:15:22 2022] BLE conn ......... (0)
[8/25 13:15:26.2]-DEBUG-: [Thu Aug 25 11:15:22 2022] BLE conn ......... (0)
[8/25 13:15:26.4]-DEBUG-: [Thu Aug 25 11:15:22 2022] BLE conn ......... (0)
**[8/25 13:15:26.6]W (11666) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x1f**
[8/25 13:15:26.6]-DEBUG-: [Thu Aug 25 11:15:23 2022] BLE conn ......... (0)
[8/25 13:15:26.8]-DEBUG-: [Thu Aug 25 11:15:23 2022] BLE conn ......... (0)
[8/25 13:15:27.0]-DEBUG-: [Thu Aug 25 11:15:23 2022] BLE conn ......... (0)
[8/25 13:15:27.2]-DEBUG-: [Thu Aug 25 11:15:23 2022] BLE conn ......... (0)
[8/25 13:15:27.4]-DEBUG-: [Thu Aug 25 11:15:23 2022] BLE conn ......... (0)

Can someone tell me what rsn =0x1f means? I could not find it in the enum.

Regards, Ramon.

jrHernandezSio, 2022-08-25:
With info log enabled:

[image]

[image]

jrHernandezSio, 2022-09-01:
Hi,

I have also tried with 4.4.1 idf version and no success.

Can someone please tell the meaning of rsn = 0x1f ?

Any clue will be appreciated.

Regards, Ramon.

chegewara, 2022-09-01:
https://github.com/chegewara/esp32-ble-wiki/issues/5

jrHernandezSio, 2022-09-01:
Thanks for the link.

Unfortuantely, `0x1F Unspecified Error` does not tell me a lot ;). 

The problem only happens sometimes (1/15) and not sure how to reproduce it.

I will try some changes in my end.

Regards, Ramon.

jrHernandezSio, 2022-09-05:
Hi, 

I found the root cause: after connection my FW moved into **ligth sleep** for a while and that was causing the lose BT connectivity without a reason.

Regards, Ramon.

closed 2022-09-05 by jrHernandezSio