espressif/esp-idf #9636: RTC WDG Reset when GNSS is started

RilabsAutomotive, 2022-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dev-3290-g01d014c42d

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board with ESP32S3

### Power Supply used.

Battery

### What is the expected behavior?

The device is able to connect to MQTT backend server via Modem and GNSS started.
If a reset happens, it is expected to have some more info. Currently it is only said that it is RTC WDG reset.

### What is the actual behavior?

The device resets 40-80 sec after MQTT server is connected via Modem and GNSS is started 

### Steps to reproduce.

Power Off and then Power On the SIMCOM or RESET
Switch the ESP in 2 wire mode as the SIMCOM always starts in 2 wire mode
Switch the SIMCOM to 4 wire mode while the ESP is still in 2 wire mode
Switch the ESP to 4 wire mode
Config the modem
Get operator name, signal strength etc.
Start CMUX mode
Receive PDP context
Connect to Mosquitto Broker
Subscribe to a MQTT topic
Start GNSS
Check fix status every 100ms , but fix status is "INVALID"
after 40-80s , the device is reset due to RTCWDT_RTC_RST

### Debug Logs.

```plain
} (76030) command_lib: Command: {AT+CGNSINF
} (76040) command_lib: Token: {
I (76040) command_lib: Token: {+CGNSINF: 1,,,0.000000,0.000000,-18.000,,,1,,0.1,0.1,0.1,,,,9999000.0,6000.0}
} (76040) command_lib: Token: {
I (76040) command_lib: Token: {OK}
} (76130) command_lib: Command: {AT+CGNSINF
} (76140) command_lib: Token: {
I (76140) command_lib: Token: {+CGNSINF: 1,,,0.000000,0.000000,-18.000,,,1,,0.1,0.1,0.1,,,,9999000.0,6000.0}
} (76140) command_lib: Token: {
I (76140) command_lib: Token: {OK}
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x2f (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd03e0,len:0x32b0
load:0x403b6000,len:0xbb8
load:0x403b9000,len:0x4c0c
entry 0x403b6260
W (30) boot.esp32s3: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (46) boot: ESP-IDF 01d014c-dirty 2nd stage bootloader
I (47) boot: compile time 20:26:10
I (47) boot: chip revision: 0
I (49) boot.esp32s3: Boot SPI Speed : 80MHz
I (54) boot.esp32s3: SPI Mode       : DIO
```

### More Information.

The issue is only observed when the GNSS is started and there is no fix status!

If the GNSS is not started via _set_gnss_power_mode_(1), then there is no reset.
If the GNSS is started but the device is outside and gets fix status, then there is no reset.
If the GNSS is started and then stopped (after 30 sec) via _set_gnss_power_mode_(0), then there is a reset.

david-cermak, 2022-10-21:
There was a bug that might have caused similar behavior (fixed in https://github.com/espressif/esp-protocols/pull/157)
Could you please check if that fixes the problem you're reporting? @RilabsAutomotive

RilabsAutomotive, 2022-10-22:
We will integrate, this change. Thank you again , David!
The SIMCOM support confirmed that it is not possible to use GNSS and mobile data in the same time for SIM7070G. They promised to fix this issue on their side.

david-cermak, 2023-03-24:
Closing as it's not an issue on IDF side (feel free to reopen of create a new one if any other trouble)

closed 2023-03-24 by david-cermak