espressif/esp-idf #9639: ESP32-C3 light sleep power consumption increases after 1st resume

ValterMinute, 2022-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ENVIRONMENT: 
esp-if built from master branch
ESP32-C3-DevkitM-1 board stripped of all components but the module and equipped with a 32.768 xtal on pins GPIO0 and GPIO1.

DESCRIPTION:

I configured CONFIG_FREERTOS_USE_TICKLESS_IDLE and power management to have my MCU going in light sleep when idle.
My test application just configures PM to enable light sleep and runs a simple task that just sleeps for 1 seconds.
As expected the system goes to light sleep, wakes up briefly and then goes back to sleep for another second.
The issue is that the 1st time system goes into light sleep state it uses  around 100uA (as expected). From the 2nd time on this increased to more than 300uA.

[image]

EXPECTED BEHAVIOR:

the system uses around 130uA when in light sleep

ACTUAL BEHAVIOR:

after the first wake-up from light-sleep (due to timer set by freertos scheduler) the system uses 330uA in light sleep 

CODE:

See my code here:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_pm.h"

void dummyTask(void *pvParameters)
{
    for (;;)
    {
        vTaskDelay(1000);        
    }
}

void app_main(void)
{
    esp_pm_config_esp32c3_t cfg;

    esp_pm_get_configuration(&cfg);    

    cfg.light_sleep_enable=1;
    cfg.max_freq_mhz=80;
    cfg.min_freq_mhz=80;

    esp_pm_configure(&cfg);

    xTaskCreate(&dummyTask, "dummyTask", 2048, NULL, 5, NULL);     
}

```

As you can see esp_pm_configure is called only once at startup and my code does not change any clock/gpio/PM related setting.

This is my sdkconfig:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9425925/sdkconfig.txt)

NOTES:

I tried to use both the internal clock and an external 32KHz crystal (I need that to reduce power consumption when using BLE), but the behavior remains the same.
I changed the different PM/RTC related options that should improve power consumption, with no differences (and anyway the 1st time system goes into light sleep it is actually using the expected amount of power from the datasheet).
I tested on 2 different boards.

ValterMinute, 2022-08-30:
Just some more information.
The same thing happens even if, instead of relying on freeRTOS schedule, I invoke esp_light_sleep_start directly (after configuring a timeout with esp_sleep_enable_timer_wakeup).
I changed my code to this:

```

#include <stdio.h>
#include "esp_pm.h"
#include "esp_sleep.h"

void app_main(void)
{
    esp_pm_config_esp32c3_t cfg;

    esp_pm_get_configuration(&cfg);    

    cfg.light_sleep_enable=1;
    cfg.max_freq_mhz=80;
    cfg.min_freq_mhz=80;

    esp_pm_configure(&cfg);

    for (;;) {
        esp_sleep_enable_timer_wakeup(1000000);
        esp_light_sleep_start();
    }
}

```

I still see that from the 2nd time the MCU enter light sleep power consumption rises by 200uA.

ValterMinute, 2022-09-02:
I update my branch with all the latest commits from master and the issue is now fixed.
Thank you!

closed 2022-09-02 by ValterMinute

AxelLin, 2022-09-02:
Which commit fixed the issue? (I'd like to check if release branches have this issue or not.)

ValterMinute, 2022-09-02:
@AxelLin good question. I just noticed that my local master branch was lagging behind and I did a pull, I didn't check them one by one.
I think my branch was based on the latest release, but I've been testing a lot of other things, so I can't grant 100%. I am still in the evaluation phase and did not keep track of everything. Sorry for this.

AxelLin, 2022-12-28:
> I updated my branch with all the latest commits from master and the issue is now fixed. Thank you!

hi @ValterMinute 
Does v5.0 branch has this issue or not?

ValterMinute, 2023-01-10:
At the moment I don't have the equipment to do power consumption measurements available, so I can't confirm this.
I hope to be able to do some power consumption tests in a few weeks and I am currently on 5.0 stable.