espressif/esp-idf #9641: "No Toolchain found for Target Local" When trying to build projects downloaded from the web.

SteveCal3228, 2022-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Build Successfully. IDF is setup correctly if I load an example "File -> New -> Espressif IDF Project" and select from the list of example projects the build is successful. 

I cannot load any existing project from the filesystem though.

### What is the actual behavior?

Build fails with error "No Toolchain found for Target Local"

### Steps to reproduce.

1. Install ESP-IDF and Tools
2. Install Eclipse CDT Ver 2022-06 (4.24.0)
3. File-> Import -> Espressif -> Existing IDF Project
4. Next
5. Browse for existing project folder
6. "Select Folder"
7. "Finish"
8. Right Click Project in Project Explorer
9. "Build Project"

### Build or installation Logs.

```plain
16:14:37 **** Build of configuration Default for project Firebase-idf-master ****
make all 
Cannot run program "make": Launching failed

Error: Program "make" not found in PATH
PATH=[C:\Users\sc3228\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Users\sc3228\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\sc3228\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\sc3228\.espressif\tools\xtensa-clang\12.0.1-d9341b81fc\xtensa-esp32-elf-clang\bin;C:\Users\sc3228\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Users\sc3228\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\sc3228\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\sc3228\.espressif\tools\cmake\3.20.3\bin;C:\Users\sc3228\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Users\sc3228\.espressif\tools\ninja\1.10.2\;C:\Users\sc3228\.espressif\tools\idf-exe\1.0.3\;C:\Users\sc3228\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\sc3228\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\sc3228\.espressif\python_env\idf4.4_py3.10_env\Scripts;C:\Users\sc3228\esp\esp-idf\tools;C:/Program Files/Java/jdk-18.0.2/bin/server;C:/Program Files/Java/jdk-18.0.2/bin;C:\Program Files\Python310\Scripts\;C:\Program Files\Python310\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Users\sc3228\AppData\Local\Microsoft\WindowsApps;C:\Users\sc3228\AppData\Local\GitHubDesktop\bin;C:\Users\sc3228\.dotnet\tools;C:\Users\sc3228\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\sc3228\eclipse\cpp-2022-06\eclipse]

16:14:37 Build Failed. 1 errors, 0 warnings. (took 91ms)

No Toolchain found for Target Local
```

### More Information.

No Toolchain found for Target Local

klumw, 2022-12-21:
Open a console execute get_idf to get the idf settings. If you use eclipse or vscode for development start the editor from the console.

kolipakakondal, 2025-03-12:
With recent versions of[ espressif-ide](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.3.0), we only show esp targets for esp-idf projects, there is no scope for local target. Hence, closing this issue.

closed 2025-03-12 by kolipakakondal