espressif/esp-idf #9642: build error: 'sclk_freq' may be used uninitialized in this function

xobs, 2022-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-361-gae256e9ca2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

compile without error

### What is the actual behavior?

ccache C:\Users\Sean\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe  -IE:/Code/Farpatch/farpatch/build/config -IC:/Users/Sean/esp/esp-idf/components/driver/include -IC:/Users/Sean/esp/esp-idf/components/driver/deprecated -IC:/Users/Sean/esp/esp-idf/components/driver/esp32s3/include -IC:/Users/Sean/esp/esp-idf/components/driver/include/driver -IC:/Users/Sean/esp/esp-idf/components/newlib/platform_include -IC:/Users/Sean/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/Sean/esp/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/Sean/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/Sean/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/Sean/esp/esp-idf/components/esp_hw_support/include -IC:/Users/Sean/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/Sean/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -IC:/Users/Sean/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -IC:/Users/Sean/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -IC:/Users/Sean/esp/esp-idf/components/heap/include -IC:/Users/Sean/esp/esp-idf/components/log/include -IC:/Users/Sean/esp/esp-idf/components/soc/include -IC:/Users/Sean/esp/esp-idf/components/soc/esp32s3/. -IC:/Users/Sean/esp/esp-idf/components/soc/esp32s3/include -IC:/Users/Sean/esp/esp-idf/components/hal/esp32s3/include -IC:/Users/Sean/esp/esp-idf/components/hal/include -IC:/Users/Sean/esp/esp-idf/components/hal/platform_port/include -IC:/Users/Sean/esp/esp-idf/components/esp_rom/include -IC:/Users/Sean/esp/esp-idf/components/esp_rom/include/esp32s3 -IC:/Users/Sean/esp/esp-idf/components/esp_rom/esp32s3 -IC:/Users/Sean/esp/esp-idf/components/esp_common/include -IC:/Users/Sean/esp/esp-idf/components/esp_system/include -IC:/Users/Sean/esp/esp-idf/components/esp_system/port/soc -IC:/Users/Sean/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/Sean/esp/esp-idf/components/xtensa/include -IC:/Users/Sean/esp/esp-idf/components/xtensa/esp32s3/include -IC:/Users/Sean/esp/esp-idf/components/esp_timer/include -IC:/Users/Sean/esp/esp-idf/components/lwip/include -IC:/Users/Sean/esp/esp-idf/components/lwip/include/apps -IC:/Users/Sean/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Sean/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/Sean/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/Sean/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Sean/esp/esp-idf/components/esp_pm/include -IC:/Users/Sean/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/Sean/esp/esp-idf/components/efuse/include -IC:/Users/Sean/esp/esp-idf/components/efuse/esp32s3/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -O2 -fmacro-prefix-map=E:/Code/Farpatch/farpatch=. -fmacro-prefix-map=C:/Users/Sean/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.1-dev-361-gae256e9ca2-dirty\" -DESP_PLATFORM -DNDEBUG -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/driver/CMakeFiles/__idf_driver.dir/uart.c.obj -MF esp-idf\driver\CMakeFiles\__idf_driver.dir\uart.c.obj.d -o esp-idf/driver/CMakeFiles/__idf_driver.dir/uart.c.obj -c C:/Users/Sean/esp/esp-idf/components/driver/uart.c
C:/Users/Sean/esp/esp-idf/components/driver/uart.c: In function 'uart_set_baudrate':
C:/Users/Sean/esp/esp-idf/components/driver/uart.c:310:5: error: 'sclk_freq' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  310 |     uart_hal_set_baudrate(&(uart_context[uart_num].hal), baud_rate, sclk_freq);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/Sean/esp/esp-idf/components/driver/uart.c: In function 'uart_get_baudrate':
C:/Users/Sean/esp/esp-idf/components/driver/uart.c:327:5: error: 'sclk_freq' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  327 |     uart_hal_get_baudrate(&(uart_context[uart_num].hal), baudrate, sclk_freq);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/Sean/esp/esp-idf/components/driver/uart.c: In function 'uart_param_config':
C:/Users/Sean/esp/esp-idf/components/driver/uart.c:756:5: error: 'sclk_freq' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  756 |     uart_hal_set_baudrate(&(uart_context[uart_num].hal), uart_config->baud_rate, sclk_freq);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
[778/957] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_rgb_panel.c.objninja: build stopped: subcommand failed.

### Steps to reproduce.

Unclear

### More Information.

I'm unclear how to reproduce this. It happens on my Windows PC, but doesn't appear to happen in CI. It's truly bizarre, and I'm posting this in case anyone is experiencing something similar.

The following patch fixes things, but the compiler seems to be wrong here, unless I'm missing something:

```patch
diff --git a/components/driver/uart.c b/components/driver/uart.c
index b68b2900e1..b0f811db9c 100644
--- a/components/driver/uart.c
+++ b/components/driver/uart.c
@@ -300,7 +300,7 @@ esp_err_t uart_set_baudrate(uart_port_t uart_num, uint32_t baud_rate)
     ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");

     uart_sclk_t src_clk;
-    uint32_t sclk_freq;
+    uint32_t sclk_freq = 0;

     uart_hal_get_sclk(&(uart_context[uart_num].hal), &src_clk);
     esp_err_t err = uart_get_sclk_freq(src_clk, &sclk_freq);
@@ -317,7 +317,7 @@ esp_err_t uart_get_baudrate(uart_port_t uart_num, uint32_t *baudrate)
     ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");

     uart_sclk_t src_clk;
-    uint32_t sclk_freq;
+    uint32_t sclk_freq = 0;

     uart_hal_get_sclk(&(uart_context[uart_num].hal), &src_clk);
     esp_err_t err = uart_get_sclk_freq(src_clk, &sclk_freq);
@@ -746,7 +746,7 @@ esp_err_t uart_param_config(uart_port_t uart_num, const uart_config_t *uart_conf
         periph_rtc_dig_clk8m_enable();
     }
 #endif
-    uint32_t sclk_freq;
+    uint32_t sclk_freq = 0;
     esp_err_t err = uart_get_sclk_freq(uart_config->source_clk, &sclk_freq);
     assert(err == ESP_OK);
```

xobs, 2022-08-26:
I've reinstalled the toolchain based on `master`, and still have this problem.

One way to reproduce it may be to check out https://github.com/farpatch/farpatch/commit/84cbd00762bc1a95f7ce48197a8aa61256624364 on a Windows machine and run `idf.py` to build. Note that it requies node.js to be installed.

ESP-Marius, 2022-08-26:
Hi,

I see `DNDEBUG` in your output, which likely means you are compiling with `CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE` enabled.  With the assert checking the ret value of `uart_get_sclk_freq` gone the compiler is no longer sure that `sclk_freq` has been initialized.

Maybe you are running without  `CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE` in CI and that's why you are seeing different results.

We'll fix this on our end, thanks for reporting this!

xobs, 2022-08-26:
That's very good sleuthing -- yes, I was disabling those in a local configuration to improve performance, and that configuration didn't get checked into CI.

closed 2022-09-02 by espressif-bot