espressif/esp-idf #9643: crash on `esp_netif_internal_dhcpc_cb()` with master

trombik, 2022-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-361-gae256e9ca2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

not a devboard, a bare esp32

### Power Supply used.

USB

### What is the expected behavior?

the app does not crash. older releases are known to work, such as 4.4.x.

### What is the actual behavior?

with `master`, the app crashes.

### Steps to reproduce.

as i don't have minimal code to reproduce, you need to build the app.

1. clone https://github.com/trombik/esp_wireguard
2. cd examples/demo && idf.py menuconfig (edit SSID and password, other defaults are irrelevant)
3. idf.py build
4. idf.py flash monitor

### Debug Logs.

```plain
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=22a2ch (141868) map
I (162) esp_image: segment 1: paddr=00032a54 vaddr=3ffb0000 size=03e88h ( 16008) load
I (169) esp_image: segment 2: paddr=000368e4 vaddr=40080000 size=09734h ( 38708) load
I (185) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=86b08h (551688) map
I (385) esp_image: segment 4: paddr=000c6b30 vaddr=40089734 size=0b6e8h ( 46824) load
I (404) esp_image: segment 5: paddr=000d2220 vaddr=50000000 size=00010h (    16) load
I (415) boot: Loaded app from partition at offset 0x10000
I (415) boot: Disabling RNG early entropy source...
I (427) cpu_start: Pro cpu up.
I (427) cpu_start: Starting app cpu, entry point is 0x40081314
0x40081314: call_start_cpu1 at /usr/home/trombik/github/trombik/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (443) cpu_start: Pro cpu start user code
I (443) cpu_start: cpu freq: 160000000 Hz
I (443) cpu_start: Application information:
I (448) cpu_start: Project name:     demo
I (453) cpu_start: App version:      9befece-dirty
I (458) cpu_start: Compile time:     Aug 26 2022 10:15:38
I (464) cpu_start: ELF file SHA256:  4c2f2f6e2f5cfde8...
I (470) cpu_start: ESP-IDF:          v5.1-dev-361-gae256e9ca2
I (477) heap_init: Initializing. RAM available for dynamic allocation:
I (484) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[... 104 lines cut ...]
Rebooting...
```

### More Information.

the code that trigger is `netif_add()` at:
https://github.com/trombik/esp_wireguard/blob/9befece87ddbf2fbfd9942285abe60f3ecb20476/src/esp_wireguard.c#L180

```c
    /* Register the new WireGuard network interface with lwIP */
    wg_netif = netif_add(
            &wg_netif_struct,
            ip_2_ip4(&ip_addr),
            ip_2_ip4(&netmask),
            ip_2_ip4(&gateway),
            &wg, &wireguardif_init,
            &ip_input);
    if (wg_netif == NULL) {
        ESP_LOGE(TAG, "netif_add: failed");
        err = ESP_FAIL;
        goto fail;
    } 
```

the interface has a fixed IP address. i don't know why the callback is being called (ref 5b471a18489b056f65fe8dcbb2c992d27909ebc9).

trombik, 2022-08-27:
`ip_info` is NULL, thus the crash.

```console
(gdb) p esp_netif->ip_info 
$15 = (esp_netif_ip_info_t *) 0x0
```

https://github.com/espressif/esp-idf/blob/83b8556f107fc3d5fd411eb7a2bba185f988cc0f/components/esp_netif/lwip/esp_netif_lwip.c#L1096-L1118

david-cermak, 2022-10-21:
@trombik The problem is that your project uses lwIP directly and at the same time the default `esp-netif` layers. It is possible to use lwIP netifs directly, but in that case I'd recommend using only lwIP (without IDF wrappers), since esp-netif installs certain callbacks and relies on certain conditions that won't be met if you bypass the `esp-netif` API and create netifs partially with `esp-netif` and partially using lwip's `netif_add()`

trombik, 2022-10-22:
@david-cermak thanks for the explanation. do you (espressif) have a plan to abstract everything so that one can use `esp-netif` without `lwip`?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html says:

> ESP-IDF currently implements ESP-NETIF for the lwIP TCP/IP stack only. However, the adapter itself is TCP/IP implementation agnostic and different implementations are possible.

it looks to me, at the moment, one cannot write networking code without including `lwip` headers, like `lwip/netdb.h`. or, are you expecting users should use `esp-netif` just for higher protocol layers?

david-cermak, 2022-10-22:
Writing networking code is already possible if you're using standard socket/posix like API. It's recommended to include
```cpp
#include <sys/socket.h>
#include <netdb.h>
```
That will in turn include `lwip/sockets` and `lwip/netdb.h` from lwip, but the app *source* code would then be lwip independent.

As for the network interface code, it's *mostly* possible (if you're okay with IDF's `ethernetif` and `wlanif` layers, i.e. using `ESP_NETIF_NETSTACK_DEFAULT_...()` macro). Only if you need to create some special/custom interface, you'd have to include `lwip/esp_netif_net_stack.h` that uses `lwip/netif.h` and in that case a direct lwip dependency is necessary.

trombik, 2022-10-22:
@david-cermak thanks for the very clear answer!

closed 2022-10-22 by trombik