espressif/esp-idf #9646: `AttributeError: module 'esptool' has no attribute '_main'` at idf5.+

Coloryr, 2022-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

idf project can build with `idf5.0` on `vs code IDE`

### What is the actual behavior?

build fail with `idf5.0`, but use `idf4.4` can build successful

### Steps to reproduce.

1. install idf5.0
2. open project
3. run build command

dobairoland, 2022-08-26:
Hi @Coloryr. Please take a look at hints posted [here](https://github.com/espressif/esp-idf/issues/9451#issuecomment-1198096270)

Duplicate of https://github.com/espressif/esp-idf/issues/9451

dobairoland, 2022-08-26:
> build fail with idf5.0, but use idf4.4 can build successful

I think the setup of v4.4 is used with v5.0 (which is not compatible). Probably, you will have to re-run the setup so it would be re-generated for v5.0.

Coloryr, 2022-08-26:
> > build fail with idf5.0, but use idf4.4 can build successful
> 
> I think the setup of v4.4 is used with v5.0 (which is not compatible). Probably, you will have to re-run the setup so it would be re-generated for v5.0.

my `esp-idf` uses `ESP-IDF VS Code Extension` to install

Coloryr, 2022-08-26:
re-run `esp-idf\install.bat` can't fix

dobairoland, 2022-08-26:
> All files have been deleted and reinstalled

What about `E:\workload\.espressif\python_env`?

Coloryr, 2022-08-26:
> > All files have been deleted and reinstalled
> 
> What about `E:\workload\.espressif\python_env`?

Contains two folders:
[image]

Coloryr, 2022-08-26:
it may be a vs extension problem
idf5.+ can build successful with cmd
[image]

dobairoland, 2022-08-26:
You could try to remove the 4.4 version. It could be one more in `E:\workload\.espressif\vscode\python_env` as well. That is why `export.bat` has no effect. If you still have an issue then please report it in https://github.com/espressif/vscode-esp-idf-extension. This relates to Vscode setup. You can get better help there.

closed 2022-08-26 by dobairoland

Coloryr, 2022-08-26:
> You could try to remove the 4.4 version. It could be one more in `E:\workload\.espressif\vscode\python_env` as well. That is why `export.bat` has no effect. If you still have an issue then please [...]

thank you.
i will report to vscode extension.

kurau1, 2022-09-26:
Hello, @Coloryr .
Do you solve the problem?
i am facing the same issue. i reinstall windows, reinstall VScode, it didn't work.
i wanna use idf5! it can bulid project successfully but can't flash device. and give this information in terminal.
```
Traceback (most recent call last):
  File "f:\ESP32\idf\esp-idf\components\esptool_py\esptool\esptool.py", line 22, in <module>
    esptool._main()  # type: ignore
AttributeError: module 'esptool' has no attribute '_main'
```
i have reinstall windows and vscode so many times. i install esp-idf v4.4 in VScode, it worked, can build and flash. but when i install esp-idf v5. i can't flash! it's a torture.

Coloryr, 2022-10-06:
@kurau1 nope, but i have a better solution without use the vscode extension.

kurau1, 2022-10-17:
> 

haha, me too, i edit program by vscode and flash by powershell, it can work T T