espressif/esp-idf #9647: Adding msgpack-c library to MQTT project

marco711, 2022-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I am trying to add the MessagePack library https://github.com/msgpack/msgpack-c/tree/c_master to the MQTT-TCP example https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/tcp/main/app_main.c . I have installed the msgpack-c library in the tcp folder of the MQTT project example following the https://github.com/msgpack/msgpack-c/tree/c_master#install-from-git-repository procedure. I have also modified the CMakeLists.txt file within the main folder as follows:

```
idf_component_register(SRCS "app_main.c"
                    INCLUDE_DIRS "../msgpack-c/include/" 
                    )
```

Then, I added the library (`#include <msgpack.h>`) and I have modified the `app_main` function to call some of the msgpack-c functions as follows:
```
void app_main(void)
{
    ESP_LOGI(TAG, "[APP] Startup..");
    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());

    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set("MQTT_CLIENT", ESP_LOG_VERBOSE);
    esp_log_level_set("MQTT_EXAMPLE", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT_BASE", ESP_LOG_VERBOSE);
    esp_log_level_set("esp-tls", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT", ESP_LOG_VERBOSE);
    esp_log_level_set("OUTBOX", ESP_LOG_VERBOSE);

    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());

    mqtt_app_start();

   /* MessagePack functions. */
	msgpack_sbuffer *buffer = msgpack_sbuffer_new();
	msgpack_packer *pk = msgpack_packer_new(buffer, msgpack_sbuffer_write);
	#if 1
        for(int j = 0; j<23; j++) {
           /* NB: the buffer needs to be cleared on each iteration */
           msgpack_sbuffer_clear(buffer);

           /* serializes ["Hello", "MessagePack"]. */
           msgpack_pack_array(pk, 3);
           msgpack_pack_bin(pk, 5);
           msgpack_pack_bin_body(pk, "Hello", 5);
           msgpack_pack_bin(pk, 11);
           msgpack_pack_bin_body(pk, "MessagePack", 11);
           msgpack_pack_int(pk, j);

           /* deserializes it. */
           msgpack_unpacked msg;
           msgpack_unpacked_init(&msg);
           msgpack_unpack_return ret = msgpack_unpack_next(&msg, buffer->data, buffer->size, NULL);

           /* prints the deserialized object. */
           msgpack_object obj = msg.data;
           msgpack_object_print(stdout, obj);  /*=> ["Hello", "MessagePack"] */
           msgpack_unpacked_destroy(&msg);
           puts("");
        }

        /* cleaning */
        msgpack_sbuffer_free(buffer);
        msgpack_packer_free(pk);	
	

#endif
}
```
When I build the project with `idf.py build` it fails showing the following:
```

At top level:
../main/app_main.c:112:13: warning: 'mqtt_app_start' defined but not used [-Wunused-function]
 static void mqtt_app_start(void)
             ^~~~~~~~~~~~~~
[5/7] Linking CXX executable mqtt_tcp.elf
FAILED: mqtt_tcp.elf 
.
.
.
/home/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(app_main.c.obj): in function `app_main':
/home/esp/IoTliftv1_0_3/tcp/build/../main/app_main.c:195: undefined reference to `msgpack_unpack_next'
/home/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/esp/IoTliftv1_0_3/tcp/build/../main/app_main.c:199: undefined reference to `msgpack_object_print'
/home/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(app_main.c.obj): in function `msgpack_unpacked_destroy':
/home/esp/IoTliftv1_0_3/tcp/build/../msgpack-c/include/msgpack/unpack.h:260: undefined reference to `msgpack_zone_free'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
Then I've tried to solve the problem by modifying the CMakeLists.txt file in the tcp folder as follows:

```
# The following four lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

project(mqtt_tcp)
# (Not part of the boilerplate)
# This example uses an extra component for common functions such as Wi-Fi and Ethernet connection.
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

add_executable (${PROJECT_NAME} ${CMAKE_CURRENT_LIST_DIR}/../tcp/main/app_main.c)
target_link_libraries(${PROJECT_NAME} "-lmsgpackc")
```

But this time the build fails to find the esp_wifi.h file:
```
[1/2] Building C object CMakeFiles/mqtt_tcp.dir/main/app_main.c.obj
FAILED: CMakeFiles/mqtt_tcp.dir/main/app_main.c.obj 
/home/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc   -march=rv32imc -MD -MT CMakeFiles/mqtt_tcp.dir/main/app_main.c.obj -MF CMakeFiles/mqtt_tcp.dir/main/app_main.c.obj.d -o CMakeFiles/mqtt_tcp.dir/main/app_main.c.obj -c ../main/app_main.c
../main/app_main.c:15:10: fatal error: esp_wifi.h: No such file or directory
 #include "esp_wifi.h"
          ^~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
Has anybody added the msgpack-c library before or could help me to understand what am I doing wrong and how can I add this library to the MQTT project?

igrr, 2022-08-26:
Hi @marco711, I recommend you to follow the structure of this example: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_lib.

If you run into a problem and can't make it work based on this structure, please upload the project to Github and share the link, we will try to help.

Regarding why the existing project doesn't work, I think your project CMakeLists.txt file is not compatible with the IDF build system. Some edits and comments:
```cmake
cmake_minimum_required(VERSION 3.5)

# has to be before "include project.cmake"
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

# has to be before "project" call
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(mqtt_tcp)

# not necessary, the IDF build system does this for you
# add_executable (${PROJECT_NAME} ${CMAKE_CURRENT_LIST_DIR}/../tcp/main/app_main.c)

# not necessary in the project CMakeLists.txt file.
# If you build msgpackc as a static library (which you aren't doing now),
# you should use target_link_libraries(${COMPONENT_LIB} PRIVATE msgpackc) in component CMakeLists.txt file.
# target_link_libraries(${PROJECT_NAME} "-lmsgpackc")
```

But it's better to start from the import_lib project.

igrr, 2023-01-08:
Closing as there was no feedback. Please leave a comment if the issue is still valid.

closed 2023-01-08 by igrr