espressif/esp-idf #9649: crash in idf.py monitor after device disconnect

simpkins, 2022-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Relatively low priority, but I figured I would report this idf tool  exception.  I ran into a crash after leaving `idf.py monitor` running for a while after the device had been disconnected.  This was on Linux, close to the head of master (v5.0-dev-4038-g9961061811)

Just from glancing at the error, it looks like the idf tool code can't handle very long lines.  If the device is disconnected for long enough the idf monitor code will print enough `.` characters without a newline to crash the idf tool code that is reading from it.

```
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Traceback (most recent call last):
  File "/usr/lib/python3.8/asyncio/streams.py", line 540, in readline
    line = await self.readuntil(sep)
  File "/usr/lib/python3.8/asyncio/streams.py", line 618, in readuntil
    raise exceptions.LimitOverrunError(
asyncio.exceptions.LimitOverrunError: Separator is not found, and chunk exceed the limit

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/src/esp-idf/tools/idf.py", line 808, in <module>
    main()
  File "/usr/local/src/esp-idf/tools/idf.py", line 743, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "/home/simpkins/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/simpkins/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/simpkins/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1691, in invoke
    return _process_result(rv)
  File "/home/simpkins/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1628, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
  File "/home/simpkins/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/src/esp-idf/tools/idf.py", line 638, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "/usr/local/src/esp-idf/tools/idf.py", line 216, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "/usr/local/src/esp-idf/tools/idf_py_actions/serial_ext.py", line 159, in monitor
    RunTool('idf_monitor', monitor_args, args.project_dir, build_dir=args.build_dir, hints=hints)()
  File "/usr/local/src/esp-idf/tools/idf_py_actions/tools.py", line 146, in __call__
    process, stderr_output_file, stdout_output_file = asyncio.run(self.run_command(self.args, env_copy))
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/usr/local/src/esp-idf/tools/idf_py_actions/tools.py", line 180, in run_command
    await asyncio.gather(
  File "/usr/local/src/esp-idf/tools/idf_py_actions/tools.py", line 210, in read_and_write_stream
    out = await input_stream.readline()
  File "/usr/lib/python3.8/asyncio/streams.py", line 549, in readline
    raise ValueError(e.args[0])
ValueError: Separator is not found, and chunk exceed the limit
```

dobairoland, 2022-08-26:
Thanks, you are right @simpkins. We need nicer error messages than this.

simpkins, 2022-08-27:
Actually, the fix for #9432 looks like it might have fixed this already by adding interactive=True in serial_ext.py.  That change causes it to avoid the readline() call, at least for the `monitor` tool.  The commit where I saw this error did not yet include that fix.

closed 2022-09-06 by espressif-bot