espressif/esp-idf #9652: Ch340/41 usb driver for esp32 s3 Any one help me out on this?

THEDSP, 2022-08-27:
### Is your feature request related to a problem?

I just trying to communicate with device which has ch340/41 usb chip i want to esp32 s3 to make communicate with it as host 

- i already try esp-idf usb host cdc lib but it does not work the descriptor read is working but how to set up for device ch340 in simple word to write the driver for ch340 i trying to replicate the ftdi driver given by the esp but i cant make it working so there is any one who help me out on this.

AristarchosOfSamos, 2022-08-28:
-> [[https://github.com/espressif/esp-idf/issues/8951#issuecomment-1177577056](url)](https://github.com/espressif/esp-idf/issues/8951#issuecomment-1177577056)

tore-espressif, 2022-08-29:
Hello @THEDSP ,
As @AristarchosOfSamos pointed out, this is a duplicate of https://github.com/espressif/esp-idf/issues/8951 you can find link to CH34x driver there.

The reason why we didn't merge it to `master` branch of esp-idf is that we don't want to add device drivers into the esp-idf, but rather have separate projects for drivers such as this (although, we got a little delayed with this plan).

THEDSP, 2022-08-30:
thanks a lot I checked it out but i can not compile it succusses fully can  you help me on this tore

THEDSP, 2022-08-30:
> Hello @THEDSP , As @AristarchosOfSamos pointed out, this is a duplicate of #8951 you can find link to CH34x driver there.
>
> The reason why we didn't merge it to `master` branch of esp-idf is that we [...]

i can't compile this please help me out on this

tore-espressif, 2022-08-31:
From the log I can see that `cdc_acm_host.h` header file is included correctly, but `cdc_acm_host.c` is not compiled.

Navigate to your project root and `usb_host_cdc_acm` component to your project. run :
```
 idf.py add-dependency usb_host_cdc_acm
```

This will add the dependency to your `main` component. If you need to add the dependency to another_componenet run
```
 idf.py add-dependency usb_host_cdc_acm --component another_component
 ```

THEDSP, 2022-09-01:
hello 

> From the log I can see that `cdc_acm_host.h` header file is included correctly, but `cdc_acm_host.c` is not compiled.
>
> Navigate to your project root and `usb_host_cdc_acm` component to your project. [...]

Hello sir thanks 
Now i successfully compiled the code and attach usb to it But first the program is stuck at Opening CH340 device
and after time device rebooted .
I further debug it so I found out the PID of my ch341 is 5512 instead of 5522 mentioned in .hpp file so I changed it to my PID
now the code will stuck at Setting up line coding so what we can do to work it properly.

tore-espressif, 2022-09-01:
@THEDSP PID 0x5512 means that your CH341 is programmed for 'EPP/MEM/I2C mode, EPP/I2C adapter'.

The driver that I gave you is for 'UART' mode. Unfortunately, I don't have this device, so I can't help you much.

What you can do is to capture USB communication between the device and USB Host (PC), on Windows you can use Wireshark. 
And then replicate the communication with ESP, by modifing the CH340 UART driver

Regards, Tomas

THEDSP, 2022-09-01:
can you mailed me in personal I have some thing urgent to discuss with you?

THEDSP, 2022-09-01:
I will try to work on as you said and let you updated on that and if you contact me on mail we can discus more in details

THEDSP, 2022-09-01:
i found linux driver for ch341 for epp i2c mode so how can we port them in to the esp.
https://github.com/zoobab/ch341-parport/tree/master/CH341PAR_LINUX
where to start any hint ?

THEDSP, 2022-09-07:
Hello any update on this?

THEDSP, 2022-09-07:
Can you guide me for raw fiction to communicate with the usb host

THEDSP, 2022-09-07:
how to set up packet brrequest and so on

tore-espressif, 2022-12-06:
CH34x driver released https://components.espressif.com/components/espressif/usb_host_ch34x_vcp

The example will be updated soon

closed 2022-12-13 by espressif-bot

chegewara, 2023-02-25:
Hi @tore-espressif 
i see this note in component:
`Limited implementation only. The vendor does not provide full specification.`
I just found this piece of code and i believe it may help to reverse engineer CH341 and to improve espressif driver.:
https://github.com/openwch/ch583/blob/main/EVT/EXAM/USB/Device/COM/src/Main.c#L562-L588

I hope this help in future.

tore-espressif, 2023-03-08:
Hi @chegewara ,
Thank your for the link! Yes, most of the current driver is reverse-engineered from similar sources.

We are trying to define boundaries for support of these 3rd party components, ATM we provide the support only if more users are asking about the same feature. In case of CH341 it seems that the basic TX/RX support is enough for most usecases :)