espressif/esp-idf #9655: Secure boot check fail  / enabled secure boot v1 without flash encryption

ABHITHLALC, 2022-08-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf -release-v4.2 esp32-version

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

USB

### What is the expected behavior?

I expected to flash the code in secure boot enabled chip using same secure boot key.

### What is the actual behavior?

PS C:\Users\LENOVO\OneDrive\Desktop\abhith\template-app> idf.py flash
Executing action: flash
Serial port COM5
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Running ninja in directory c:\users\lenovo\onedrive\desktop\abhith\template-app\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Users\LENOVO\OneDrive\Desktop\abhith\templ...s/LENOVO/OneDrive/Desktop/abhith/template-app/build/template-app.bin" 
template-app.bin binary size 0x2fff4 bytes. Smallest app partition is 0x100000 bytes. 0xd000c bytes (81%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\LENOVO\OneDrive\Desktop\abhith\template-app\build\bootloader\esp-idf\esptool_py && C:\Users\LENOVO\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/LENOVO/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x10000 bootloader 0x1000 C:/Users/LENOVO/OneDrive/Desktop/abhith/template-app/build/bootloader/bootloader.bin"
Bootloader binary size 0xab90 bytes. 0x4470 bytes (29%) free.
[2/3] cmd.exe /C "cd /D C:\Users\LENOVO\esp\esp-idf\components\esptoo...Users/LENOVO/esp/esp-idf/components/esptool_py/run_serial_tool.cmake" 
esptool.py esp32 -p COM5 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x20000 template-app.bin 0x10000 partition_table/partition-table.bin
esptool.py v3.3.2-dev
Serial port COM5
Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:c4:49:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00020000 to 0x0004ffff...
Flash will be erased from 0x00010000 to 0x00010fff...
Compressed 196596 bytes to 92754...
Writing at 0x00020000... (16 %)
Writing at 0x00030f63... (50 %)
[... 305 lines cut ...]
load:0x40078000,len:14848
load:0x40080400,len:3792
0x40080400: _init at ??:?

secure boot check fail
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6664
load:0x40078000,len:14848
load:0x40080400,len:3792
0x40080400: _init at ??:?

secure boot check fail
ets_main.c 371 
**repeating**

**EFUSE SUMMARY**
[image]

Can anyone help me with this? What went wrong.

negativekelvin, 2022-08-28:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6664
load:0x40078000,len:14848
load:0x40080400,len:3792
0x40080400: _init at ??:?
```

```
rst:Oxc (SW_CPU_RESET), boot:0x13 (SPI_FAST_FLASH_BOOT) 
configsip: 0, SPIWP: Oxee 
clk_drv: 0x00,q_drv: 0x0,d_drv: Oxe, cs@_drv: 0x00, hd_drv: 0x00, wp_drv: 0x00 
mode:DIO, clock div:2 
load: Ox3fff0eb8, len: 10768 
load: 0x40078000, len:24604 
load: 0x48080400, len:3792 
0x40088400: init at ??:?
```

You must have accidentally reflashed the bootloader because the logs show two different bootloader sizes

ABHITHLALC, 2022-08-29:
Thank you for the detailed and sudden helpful response @negativekelvin .
I think I made a mistake by tried to confim if the secure boot enabled properly by flashing another code without using secure key. 

If you don't mind ,Some doubts which I want to clear.

1. Is it possible to reflash the bootloader after enabling secureboot ?
**Below  commad is shown after build**
```
One-time flash command is:
        C:/Users/LENOVO/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe  C:/Users/LENOVO/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 C:/Users/LENOVO/OneDrive/Desktop/abhith/template-app/build/bootloader/bootloader.bin
* IMPORTANT: After first boot, BOOTLOADER CANNOT BE RE-FLASHED on same device*
```

2. As the steps mentioned above(from the documment), I did not burn the key manually. Is it really necessary to burn the key with burn key command? If it is necessary to burn key manually then it should be after build project,right? 
3. What is the main diffrence between One-time flash and reflashable in secure boot v1.
4. After enabling secure boot can we flash any code with signing with our private key?

sachin0x18, 2022-08-29:
Hi @ABHITHLALC,

I can help clear your doubts,

1. It is possible to reflash the bootloader but for that you need to select the "Reflashable" option for "Secure bootloader mode". I would suggest you to go through the [Re-Flashable Software Bootloader](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/security/secure-boot-v1.html#re-flashable-software-bootloader) section from the documentation. It has all the required details
2. If you decide to use the Reflashable option then you need to burn the key manually after the build
3. In one-time flash, you won't be able to flash any other bootloader as the secure bootloader key is generated on the device. Whereas, in reflashable mode, you provide the secure bootloader key and hence you can flash any bootloader as long as you generate the digest using the same secure bootloader key. All these details are provided in the documentation linked above.
4. Yes, you can flash any application as long as you sign it with the same secure boot signing key.

ABHITHLALC, 2022-08-29:
Hello @sachin0x18 
Thanks for the reply and thank you for the information.

> 3. In one-time flash, you won't be able to flash any other bootloader as the secure bootloader key is generated on the device. Whereas, in reflashable mode, you provide the secure bootloader key and [...]

In this case I used One-time flash mode. Can you please tell me which step went wrong.

sachin0x18, 2022-08-29:
Like negativekelvin pointed out, did you reflash some other bootloader ?

ABHITHLALC, 2022-08-29:
I think may be I did. But if I try to reflash the other bootloader on a system with secure boot enabled, will it damage the chip? Is there any option to recover this module?

sachin0x18, 2022-08-29:
If you flash any other bootloader then it simply won't boot up because it doesn't have the same digest. If you still have the same bootloader that you flashed when enabling secure boot then you can flash it again and the device should boot fine

ABHITHLALC, 2022-08-29:
Which files should be kept safe after enabling secure boot?

sachin0x18, 2022-08-29:
Only the signing key should be kept safe. You need to ensure that you don't erase/reflash the bootloader.

ABHITHLALC, 2022-08-29:
Thank you for all the informations.  It's really helpful. I will it out on new chip.

closed 2022-08-29 by sachin0x18

ABHITHLALC, 2022-08-29:
You were right about bootloader @sachin0x18 @negativekelvin , I accidentally flashed another bootloader. Now I have the old bootloader file that flashed earlier. And I reflashed it as you said

> If you flash any other bootloader then it simply won't boot up because it doesn't have the same digest. If you still have the same bootloader that you flashed when enabling secure boot then you can flash it again and the device should boot fine

But the output is same secure boot check fail, but except the logs shows old bootloader logs.
But the problem is same.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff00b8,len:10768
load:0x40078000,len:24604
load:0x40080400,len:3792
0x40080400: _init at ??:?

secure boot check fail   
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff00b8,len:10768
load:0x40078000,len:24604
load:0x40080400,len:3792
0x40080400: _init at ??:?

secure boot check fail
ets_main.c 371
```

sachin0x18, 2022-09-14:
Hi @ABHITHLALC,

Sorry for the delayed response.

Just want to check few things,

Did you re-build the old bootloader file, if so then its digest would change as well.

If you did not rebuild, did you change the digest that is flashed at 0x0 ?

ABHITHLALC, 2022-09-15:
> Did you re-build the old bootloader file, if so then its digest would change as well.

Yes I rebuild the file. So maybe that will be the reson.

sachin0x18, 2022-09-15:
Yeah. Unfortunately, there is no way to recover this module now. You will have to try on a new chip.

Feel free to reopen or submit a new issue if you ever come across any other problem

ABHITHLALC, 2022-09-15:
Thank you for the support, really appreciated