espressif/esp-idf #9657: ubuntu 无法寻找到esp32c3的usb-jtag调试器

e2718281689, 2022-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

开发环境：idf4.4.1
开发系统：Ubuntu（vmware）
使用模组：安信可esp-c3-12
我最近在尝试使用esp32c3的usb调试器进行调试却无法进行调试，我发现Ubuntu系统只加载了usb-serial，可找到/dev/ttyACM0，但没有usb-jtag接口设备.
[image]
而我将开发板接入到win10系统中可以找到usb-serial和usb-jtag
[image]
我认为是Ubuntu中关于esp32-c3的usb-jtag的驱动没有加载
这是系统日志  [01.txt](https://github.com/espressif/esp-idf/files/9438975/01.txt)
我截取了你们可能感兴趣的部分
```
[    2.012822] usb 3-2: new full-speed USB device number 3 using xhci_hcd
[    2.315437] usb 3-2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[    2.315443] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.315445] usb 3-2: Product: USB JTAG/serial debug unit
[    2.315446] usb 3-2: Manufacturer: Espressif
[    2.315447] usb 3-2: SerialNumber: 84:F7:03:54:55:E8
[    2.451929] usb 3-3: new full-speed USB device number 4 using xhci_hcd
[    2.601037] usb 3-3: New USB device found, idVendor=0e0f, idProduct=0002, bcdDevice= 1.00
[    2.601042] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.601044] usb 3-3: Product: VMware Virtual USB Hub
[  163.914257] usb 3-2: USB disconnect, device number 3
[  167.028470] usb 3-2: new full-speed USB device number 6 using xhci_hcd
[  167.323418] usb 3-2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[  167.323426] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  167.323428] usb 3-2: Product: USB JTAG/serial debug unit
[  167.323430] usb 3-2: Manufacturer: Espressif
[  167.323431] usb 3-2: SerialNumber: 84:F7:03:54:55:E8
[  167.331482] cdc_acm 3-2:1.0: ttyACM0: USB ACM device
```
如果真是驱动的问题，请一定要告诉我如何安装驱动

WangShuoran, 2022-09-14:
看看这个回答能不能解决你的问题：https://www.esp32.com/viewtopic.php?t=28290

Alvin1Zhang, 2025-03-12:
Thanks for reporting and sharing the forum post link, seems the issue has been resolved per the post, feel free to reopen.

closed 2025-03-12 by Alvin1Zhang