espressif/esp-idf #9658: esp_partition_write_raw never returns

HondaRulez, 2022-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Writing partition in 4KB chunks.

after several calls using esp_partition_write_raw() never returns. (task watchdog raises)
using esp_partition_write() always works fine.

ESPIDF 4.4.1

Chip is ESP32-D0WD (revision 1)

HondaRulez, 2022-08-28:
Udate:

May important, but my app uses Bluetooth connection to read the data to be flashed.

I dig into the IDF source and did some testing:
 
as I determined all the partition_write() functions calling one of these functions depend on the configuration.

esp_flash_write();
spi_flash_write();

esp_flash_write_encrypted();
spi_flash_write_encrypted();

Then I called these functions directly from my source, **ALL fails randomly.**

Then I read some doc about the caching.

I decided my problem probably the flash is not ready to execute code after finishing the write.
so can't return to my app.

as a possible solution  I put a IRAM_ATTR flagged function between my writer function and the flash functions.

like this

esp_err_t IRAM_ATTR flash_write(void *dptr, uint32_t wpos, uint32_t wsz)
{
    esp_err_t err;

    err = esp_partition_write_raw(mempart, wpos, dptr, wsz);   // mempart is global var.
    return(err);
}

then my writer function calls the flash_write() instead of the esp_write functions.

This seems to solve my problem (at least not happened any freezes yet :) )

chipweinberger, 2022-08-29:
You should try v4.4.2! I think this is fixed.

chipweinberger, 2022-08-29:
https://github.com/espressif/esp-idf/issues/9332

HondaRulez, 2022-08-29:
> 

with my solutions works for a whole day :D then start the freezing again, then set the clock from 80Mhz -> 160Mhz now seems Ok, but I'm sure this is a timing issue somwhere. BTW the problem still exists with the master branch.