espressif/esp-idf #9660: SD VFS 'fseek' beyond a file end leads to inconsistent behavior if 'fstat' to obtain the file size

wuyuanyi135, 2022-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dev-436-g5c1044d84d

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

External 3.3V

### What is the expected behavior?

When seeking beyond the end of a file, the file size should not change if no modification is made, confirmed on Windows MINGW64 9.0 using the following code, where the output is `sz=0`, no matter `fflush` or not.
```c
#include <stdio.h>
#include <stdlib.h>
#include <sys/stat.h>
int main() {
  FILE *f = fopen("./hello.txt", "w+");  // hello.txt is not exist or is empty
  fseek(f, 120, SEEK_SET);
  //    fflush(f);
  struct stat sbuf;
  fstat(fileno(f), &sbuf);
  printf("sz=%ld\n", sbuf.st_size);
}
```

### What is the actual behavior?

On idf.py the similar code with sdcard  (SD SPI) VFS registered will report a inconsistent **`sz=120`** when executing the same code.

This lead to unexpected behavior when I am porting the SQLite3 library to my project. The library will try to seek by 24 bytes on the database file and perform some read when opened. This step is expected to fail since the database file is new. Then, it will perform some initialization on this created database file. However, on esp-idf, after `fseek`, the file is no longer 0 bytes (but 24 bytes), causing the sqlite3 library wrongly taken it as an initialized file and try to read again. Then it will report error code 26 (NOT_A_DB), because its just 24 bytes of zeros. 

### Steps to reproduce.

Run the following code (based on storage/sd_card/sdspi example)

```c
  sdcard_initialize();
  FILE* f = fopen(SDCARD_MOUNT"/hello.txt", "w+");
  fseek(f, 120, SEEK_SET);

  struct stat sbuf;
  fstat(fileno(f), &sbuf);

  ESP_LOGI(TAG, "sz=%ld", sbuf.st_size);  // will give sz=120
```

wuyuanyi135, 2022-08-28:
https://github.com/espressif/esp-idf/blob/5c1044d84d625219eafa18c24758d9f0e4006b2c/components/fatfs/src/ff.c#L4503-L4506

Implementation in FATFS: seek will change `obj.objsize` but not necessarily change the disk. stat will simply return `obj.objsize` as the file size, hence reporting the in-memory size rather than the file size.

wuyuanyi135, 2022-08-28:
I have managed to use 'stat' rather than 'fstat' to force the FATFS to read the file size from the filesystem. Obviously there may be some io bandwidth penalties if the size is request frequently. But this is the temporary workaround that at least allow me to run SQLite3 on my MCU.

adokitkat, 2023-05-22:
Hello. `fstat` and `fseek` are implemented in Newlib C standard library and like you mentioned, FATFS lseek `f_lseek` is indeed in implemented in FATFS. It really seems like this is a problem regarding FATFS `f_lseek` which behaves differently than POSIX/Linux/Windows. We are using FATFS just as a third-party library, it is not our code. I've written a comment on FATFS forum regarding this: http://elm-chan.org/fsw/ff/bd/?show=3743

adokitkat, 2023-05-30:
I'll close this issue. If you have any more questions related to this you can reopen it or if you have any other questions please open another issue. Thank you.

myfreax, 2024-11-17:
@wuyuanyi135 Can you tell me fix this issue

wuyuanyi135, 2024-11-19:
> @wuyuanyi135 Can you tell me fix this issue

@myfreax I was not happy with the overhead using ported SQLite so I designed a simple filesystem to store time-series blobs. From what I could recall, using stats instead of fstats should help.