espressif/esp-idf #9661: UART async sending random bits when Wi-Fi is connected

lawrenceko-intecular, 2022-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.2-1-gce1fde4b3b

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

When using example [uart_async_rxtxtasks](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/peripherals/uart/uart_async_rxtxtasks), it works perfectly fine with two ESP32 connected to each other. Adding `example_connect` to make the ESP32 connected to the Wi-Fi should have the UART async still working fine.

Expected behavior:
```
I (378572) RX_TASK: Read 11 bytes: 'Hello world'
I (378572) RX_TASK: 0x3ffb8cf8   48 65 6c 6c 6f 20 77 6f  72 6c 64                 |Hello world|
```

### What is the actual behavior?

Random bits are received on the other side when Wi-Fi is enabled.

Actual behavior:
```
I (381312) RX_TASK: Read 12 bytes: 'Hello world�'
I (381312) RX_TASK: 0x3ffb8cf8   48 65 6c 6c 6f 20 77 6f  72 6c 64 fc              |Hello world.|
```

### Steps to reproduce.

1. Add Wi-Fi example connect dependencies
```
/* Wi-Fi example_connect */
#include "esp_wifi.h"
#include "esp_event.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "protocol_examples_common.h" 
```
2. Add Wi-Fi initialization to the beginning of `app_main`
```
/* Initialize Wi-Fi */
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());
    /* --- end of Wi-Fi initialization --- */
```
4. `idf.py flash monitor` and check on the other ESP running the barebone example [uart_async_rxtxtasks](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/peripherals/uart/uart_async_rxtxtasks)
...
### 

### Debug Logs.

```plain
ESP32 w/ Wi-Fi enabled (Tx):

I (179582) TX_TASK: Wrote 11 bytes
```

ESP32 Rx:
```
I (362572) RX_TASK: Read 13 bytes: '��Hello world'
I (362572) RX_TASK: 0x3ffb8cf8   fe f0 48 65 6c 6c 6f 20  77 6f 72 6c 64           |..Hello world|
I (381312) RX_TASK: Read 12 bytes: 'Hello world�'
I (381312) RX_TASK: 0x3ffb8cf8   48 65 6c 6c 6f 20 77 6f  72 6c 64 fc              |Hello world.|
```

Note the additional '?' random bytes at the front or the end.
```

lawrenceko-intecular, 2022-08-28:
An example project can be found [here](https://github.com/lawrenceko-intecular/esp-idf/tree/master/examples/peripherals/uart/uart_sync_rxtxtasks_wifi)

lawrenceko-intecular, 2022-09-05:
Any updates on this? This should be a very important issue that may affect many applications

ginkgm, 2022-09-06:
Hi @lawrenceko-intecular , thanks for your report. Have you tried if this issue exist on master?

ginkgm, 2022-09-06:
Another thought (sorry I haven't got time to have a try), could it be a hardwar issue - the inrush current of Wi-Fi disturbs the UART signal?

To check this, you may:

1. add more grounding wires (try to add to 4  if you are using dupont wires) between the TX and RX devices
And/Or
2. add wires for VDD and GND directly from the power supply to each of the board, instead of sharing the power and GND wires between two boards

and see if it helps to suppress the issue

3. Or, use oscilloscope to test the RX pin and GND of the RX device, and see if the signal is perfect (no glitches between bytes).

Alvin1Zhang, 2022-09-21:
@lawrenceko-intecular Thanks for reporting, would you please help check if any updates for the issue? Thanks.

lawrenceko-intecular, 2022-09-21:
We found out that when using `UART_NUM_1` it will be okay, but using `UART_NUM_2` will cause the random bits to occur. Are there any limitations on UART ports?

ginkgm, 2022-10-31:
@law-ko 

There's no much differences between UART instances, except from:

1. IOMUX. IOMUX pins of UART1 are IO10, IO9, and UART2 are IO17 and IO16. But if you are using the pins described in the example (22, 23), There shouldn't be any problem
2. Pullup. There are some pins pulled up by default, but some are not. For example, pin 22, 23 are not pulledup by default, but IO 13 and 14 are. Usually the TX side should ensure the pins are pulled high properly even when nothing is being transmitted. Otherwise the garbage on the lines will still be treated as data. When the system goes into sleep modes, some pull-ups may be disabled, which may disturb the data.

Suggest connect an oscillator onto the wire to see if there is really something wrong, or we should check from the receiver side.

luacerbb, 2022-11-15:
I occur similar question,I use uart2 in esp32 chip,it can work well in sdk4.1.3 and 4.3.4,but can't work well in sdk4.4.3,it often receive UART_BREAK uart event type when data is coming. I don't known why.