espressif/esp-idf #9662: system/console/basic example does not work on ESP32-S3 using USB serial/jtag controller

davidzuhn, 2022-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-437-g5c1044d84d

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S3-DevKit-C v1.0

### Power Supply used.

USB

### What is the expected behavior?

I tried to run the system/console/basic example when configured for the USB serial/jtag controller.  I expected a working command line console over the USB port.  I've seen this work in previous snapshots of the ESP-IDF.  

### What is the actual behavior?

The board goes into a continuous loop with the following console output:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375af5
0x40375af5: esp_restart_noos_dig at /Users/zoo/esp32/esp-idf-latest-20220828/components/esp_system/esp_system.c:46 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1184
load:0x403c9700,len:0xa3c
load:0x403cc700,len:0x29f0
entry 0x403c9898
I (186) cpu_start: Pro cpu up.
I (186) cpu_start: Starting app cpu, entry point is 0x40375674
0x40375674: call_start_cpu1 at /Users/zoo/esp32/esp-idf-latest-20220828/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (196) cpu_start: Pro cpu start user code
I (196) cpu_start: cpu freq: 160000000 Hz
I (197) cpu_start: Application information:
I (198) cpu_start: Project name:     console
I (198) cpu_start: App version:      v5.1-dev-437-g5c1044d84d
I (198) cpu_start: Compile time:     Aug 28 2022 13:46:47
I (199) cpu_start: ELF file SHA256:  6edb499afddc010f...
I (200) cpu_start: ESP-IDF:          v5.1-dev-437-g5c1044d84d
I (201) heap_init: Initializing. RAM available for dynamic allocation:
I (202) heap_init: At 3FCA1188 len 00048588 (289 KiB): D/IRAM
I (203) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (204) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (204) heap_init: At 600FE03C len 00001FC4 (7 KiB): RTCRAM
I (206) spi_flash: detected chip: generic
I (206) spi_flash: flash io: dio
W (207) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (209) sleep: Configure to isolate all GPIO pins in sleep state
I (210) sleep: Enable automatic switching of GPIO sleep configuration
I (210) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (255) example: Command history enabled
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4206fcb9  PS      : 0x00060f30  A0      : 0x8200ef8e  A1      : 0x3fcecd30  
0x4206fcb9: setvbuf at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/setvbuf.c:119

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3fcecf64  A7      : 0x3c0ad120  A8      : 0x8206fcb9  A9      : 0x3fcecd10  
A10     : 0x00000000  A11     : 0x3fcecd30  A12     : 0x00000008  A13     : 0x3fcecf64  
A14     : 0x3fcecba0  A15     : 0x0000000b  SAR     : 0x00000005  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000064  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  

Backtrace: 0x4206fcb6:0x3fcecd30 0x4200ef8b:0x3fcecd60 0x40382231:0x3fcecd90
0x4206fcb6: setvbuf at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/setvbuf.c:119

0x4200ef8b: esp_console_repl_task at /Users/zoo/esp32/esp-idf-latest-20220828/components/console/esp_console_repl.c:490

0x40382231: vPortTaskWrapper at /Users/zoo/esp32/esp-idf-latest-20220828/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: 6edb499afddc010f

Rebooting...
```

### Steps to reproduce.

1. cd $IDF_PATH/examples/system/console/basic
2. idf.py set-target esp32s3
3. idf.py menuconfig   Under components | ESP System Settings, I set the value for ESP_CONSOLE_UART to ESP_CONSOLE_USB_SERIAL_JTAG
4.  idf.py build flash monitor

### More Information.

The usb serial/jtag controller works, as my own code (which doesn't use esp_console_repl_task, but a broken out variant) is working just fine with the same version of ESP-IDF.   

This is not the same as issue IDFGH-7840 (https://github.com/espressif/esp-idf/issues/9369), as the stack trace differs (and that bug was fixed).   

Running the system/console/basic example in v5.1-dev-97-g1b51788872 works just fine, so this is a regression.

davidzuhn, 2022-08-28:
Oh, and kudos to the new form for reporting bugs.  That's nice!

davidzuhn, 2022-08-28:
Also, while this is the simplest case, many of the other examples that build on this also fail (I found this by trying to run examples/peripherals/i2c/i2c_tools).

igrr, 2022-08-30:
Thank you for reporting the issue! The fix is in review, and should be merged soon. I'm also attaching the patch file.
[19866.patch.txt](https://github.com/espressif/esp-idf/files/9455257/19866.patch.txt)

closed 2022-09-05 by espressif-bot