espressif/esp-idf #9664: OTA image hash(RSA) fails on ESP32S2 but not on ESP32 (ECSDA hash) with identical code.

hairyhoncho, 2022-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32S2

### Power Supply used.

USB

### What is the expected behavior?

Expected OTA using signed image to complete on ESP32S2

### What is the actual behavior?

OTA download completes and is valid but RSA  signature check fails

### Steps to reproduce.

Build for ESP32S2
Load signed image to server
Initiate OTA from board

### Debug Logs.

```plain
I (39764) esp-x509-crt-bundle: Certificate validated
I (42344) HTTP_CLIENT: Body received in fetch header state, 0x3ffe9cff, 605
I (42344) esp_https_ota: Starting OTA...
I (42344) esp_https_ota: Writing to partition subtype 16 at offset 0x200000
I (101324) esp_image: segment 0: paddr=00200020 vaddr=3f000020 size=2df74h (188276) map
I (101354) esp_image: segment 1: paddr=0022df9c vaddr=3ffc4010 size=0207ch (  8316) 
I (101354) esp_image: segment 2: paddr=00230020 vaddr=40080020 size=8a260h (565856) map
I (101444) esp_image: segment 3: paddr=002ba288 vaddr=3ffc608c size=013bch (  5052) 
I (101454) esp_image: segment 4: paddr=002bb64c vaddr=40022000 size=1200ch ( 73740) 
I (101464) esp_image: segment 5: paddr=002cd660 vaddr=50000000 size=00010h (    16) 
I (101464) esp_image: segment 6: paddr=002cd678 vaddr=00000000 size=02958h ( 10584) 
I (101474) esp_image: Verifying image signature...
I (101484) secure_boot_v2: Take trusted digest key(s) from running app
I (101494) secure_boot_v2: #0 app key digest == #0 trusted key digest
I (101494) secure_boot_v2: Verifying with RSA-PSS...
E (101504) secure_boot_v2: Failed mbedtls_rsa_complete, err: -16512
E (101504) esp_image: Secure boot signature verification failed
I (101514) esp_image: Calculating simple hash to check for corruption...
W (101594) esp_image: image valid, signature bad
E (101594) OTA_CTL: Failed to finish up OTA
```

### More Information.

Identical code when built for ESP32 (ECSDA hash) works perfectly.

sachin0x18, 2022-08-29:
> E (101504) secure_boot_v2: Failed mbedtls_rsa_complete, err: -16512

The error code 16512 indicates that there is something wrong with the input data.

Can you please run 

```
espsecure.py verify_signature -v 2 -k <key.pem> <signed_firmware.bin>
```

and check if espsecure.py utility can verify the signature

hairyhoncho, 2022-08-29:
espsecure.py verify_signature --version 2  --keyfile secure_boot_signing_key.pem  esp32-scale.bin
espsecure.py v3.3.2-dev
Signature block 0 is valid (ECDSA). 
Signature block 0 verification successful with secure_boot_signing_key.pem (ECDSA).

Note, the problem is not with the ECDSA signature on the esp32 but with the RSA signature on esp32s2 which fails

sachin0x18, 2022-08-29:
Please check the signature of the ESP32-S2 image using the above command

hairyhoncho, 2022-08-29:
This is for the esp32-s2 image the bin file name is not changed between the esp32 and esp32S2 builds. According to sdk configuration editor this should be an RSA key.

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32s2"
CONFIG_IDF_TARGET_ESP32S2=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0002

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32s2-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config

#
# Security features
#
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_SUPPORTS_RSA=y
CONFIG_SECURE_TARGET_HAS_SECURE_ROM_DL_MODE=y
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_SIGNED_ON_UPDATE_NO_SECURE_BOOT=y
# CONFIG_SECURE_BOOT is not set
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="secure_boot_signing_key.pem"
# CONFIG_SECURE_FLASH_ENC_ENABLED is not set
# end of Security features

[image]
y

sachin0x18, 2022-08-29:
Are you using the same key for signing ESP32 and ESP32-S2 images ? I see `secure_boot_signing_key.pem` in the ESP32-S2 sdkconfig file and also in the espsecure.py command that you ran for ESP32.

Note, ESP32 Secure Boot v1 requires ECDSA key while ESP32-S2 requires RSA key.

Please generate a RSA key for signing ESP32-S2 chip by running 

```
espsecure.py generate_signing_key -v 2 esp32_s2_secure_boot_signing_key.pem
``` 

then update the key name in sdkconfig file and use this key to sign the S2 image.

hairyhoncho, 2022-08-29:
That was it. Generating new key fixed the problem. Thank-you.

closed 2022-08-29 by hairyhoncho