espressif/esp-idf #9668: esp_ota_begin called with large image size (1263952) causes a WDT reset whit CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE = y

ffamar, 2022-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.1 and 4.4.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

The ota partition is initialized

### What is the actual behavior?

WDT reset info

### Steps to reproduce.

1. Step  Set CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE = y
2.  Call esp_ota_begin with large image size (1263952)

### Debug Logs.

```plain
D (38517) CMD_MAN: Start erase partition firmwareLen: 1263952     <--This is printed before esp_ota_begin
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:14788
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (171) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (171) boot: compile time 16:42:46
I (172) boot: chip revision: 1
I (175) boot.esp32: SPI Speed      : 40MHz
I (180) boot.esp32: SPI Mode       : DIO
I (185) boot.esp32: SPI Flash Size : 16MB
W (189) boot.esp32: PRO CPU has been reset by WDT.
W (195) boot.esp32: WDT reset info: PRO CPU PC=0x400945e1
0x400945e1: _xt_context_save at /home/famar/Works/toolchain/esp/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_context.S:209

W (201) boot.esp32: WDT reset info: APP CPU PC=0x40081f6d
0x40081f6d: panicHandler at /home/famar/Works/toolchain/esp/esp-idf-v4.4.1/components/esp_system/port/panic_handler.c:213

I (207) boot: Enabling RNG early entropy source...
I (212) boot: Partition Table:
I (216) boot: ## Label            Usage          Type ST Offset   Length
I (223) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (231) boot:  1 nvs              WiFi data        01 02 0000b000 00505000
I (239) boot:  2 ota_0            OTA app          00 10 00510000 00500000
I (246) boot:  3 ota_1            OTA app          00 11 00a10000 00500000
```

### More Information.

I executed the same test using esp idf 4.4.2 and I got the same result.

```
st:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (171) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (171) boot: compile time 08:38:19
I (172) boot: chip revision: 1
I (175) boot.esp32: SPI Speed      : 40MHz
I (180) boot.esp32: SPI Mode       : DIO
I (185) boot.esp32: SPI Flash Size : 16MB
W (189) boot.esp32: PRO CPU has been reset by WDT.
W (195) boot.esp32: WDT reset info: PRO CPU PC=0x40081f91
0x40081f91: panicHandler at /home/famar/Works/toolchain/esp/esp-idf-v4.4.2/components/esp_system/port/panic_handler.c:213

W (201) boot.esp32: WDT reset info: APP CPU PC=0x40081f91
0x40081f91: panicHandler at /home/famar/Works/toolchain/esp/esp-idf-v4.4.2/components/esp_system/port/panic_handler.c:213

I (207) boot: Enabling RNG early entropy source...
```

**When CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE is not set, esp_ota_begin ends succesfully in 4.4 s** 
Thanks

AxelLin, 2023-05-05:
@ffamar 
1. Do you still hit the issue with latest v4.4.4 release?
2. Can you share your sdkconfig?

wmy-espressif, 2023-05-08:
Hi @ffamar , I would also like to ask the questions asked by @AxelLin . Besides, can you provide more contexts of your application? What happens before you call esp_ota_begin? Do you use Wi-Fi and Bluetooth, if yes, what are their activities before you call `esp_ota_begin`?

ffamar, 2024-02-28:
@esp-lrh I migrated my firmware to idf version 5.0.6 and the bug does not exist.

closed 2024-02-28 by ffamar