espressif/esp-idf #9671: Define NVS_NS_NAME_MAX_SIZE in nvs.h

nipil, 2022-08-29:
### Is your feature request related to a problem?

There is no define for NVS maximum namespace length, contrary to keys and partitions, as seen bellow

The capacity for namespace length is hardcoded to 16 in nvs.h

It would be nice (and easy) to add a define for the namespace length (see proposed solution)

And it is retro-compatible with earlier code.

```
#define NVS_PART_NAME_MAX_SIZE              16   /*!< maximum length of partition name (excluding null terminator) */
#define NVS_KEY_NAME_MAX_SIZE               16   /*!< Maximal length of NVS key name (including null terminator) */

typedef struct {
    char namespace_name[16];    /*!< Namespace to which key-value belong */
    char key[NVS_KEY_NAME_MAX_SIZE];               /*!< Key of stored key-value pair */
    nvs_type_t type;            /*!< Type of stored key-value pair */
} nvs_entry_info_t;
```

### Describe the solution you'd like.

```
#define NVS_NS_NAME_MAX_SIZE               16   /*!< Maximal length of NVS key name (including null terminator) */

typedef struct {
    char namespace_name[NVS_NS_NAME_MAX_SIZE];    /*!< Namespace to which key-value belong */
  ...
} nvs_entry_info_t;
```

### Describe alternatives you've considered.

When using multiple namespaces, or dynamically named namespaces, we need to allocate buffer to put the namespace name in. Currently the buffer has a fixed length of 16, and i define my own constant for the namespace length.

negativekelvin, 2022-08-30:
>Namespace identifiers (strings) are stored as keys of key-value pairs in namespace with index 0. Values corresponding to these keys are indexes of these namespaces.

So it should probably be defined as 

`#define NVS_NS_NAME_MAX_SIZE    NVS_KEY_NAME_MAX_SIZE`

closed 2022-10-27 by espressif-bot

0xjakob, 2022-11-04:
@nipil The new define is present in master now.