espressif/esp-idf #9672: deadlock after call esp_bluedroid_disable()

fzboffice, 2022-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-174-g134170a6fc

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

调用esp_bluedroid_disable()后应正常

### What is the actual behavior?

调用esp_bluedroid_disable()后被死锁

### Steps to reproduce.

在idf commit 134170a6fc25c72ecb03912273e761213a332ef9 后出现

xiongweichao, 2022-08-30:
Hi @fzboffice ，

If `esp_bluedroid_disable` is called in the `btc task`, a deadlock will occur. Suggest you create a new task and call it

Thanks

fzboffice, 2022-08-30:
@xiongweichao hello,i call esp_bluedroid_disable in my task.The same code behaves differently under different commit.This question only apper after commit 134170a6fc25c72ecb03912273e761213a332ef9

xiongweichao, 2022-08-30:
Hi @fzboffice ,

Please verify if this patch solves your problem?

Thanks

[fix_deadlock_after_call_esp_bluedroid_disable.patch.txt](https://github.com/espressif/esp-idf/files/9451352/fix_deadlock_after_call_esp_bluedroid_disable.patch.txt)

fzboffice, 2022-08-30:
@xiongweichao  Thank you , The problem has been fixed.

closed 2022-09-05 by espressif-bot