espressif/esp-idf #9674: esp32c3: esp_partition_read_raw silent fail with mixed QIO/DIO

HiFiPhile, 2022-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-361-gae256e9ca2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-DevKitC-02

### Power Supply used.

USB

### What is the expected behavior?

While a bootloader using DIO and an app using QIO is present, esp_partition_read_raw used by NVS should either:

- Work correctly
- Return fail with a precise error code

### What is the actual behavior?

While doing OTA update form old firmware (DIO) to new firmware (QIO), nvs configuration is lost.

New firmware can't read nvs_key partition:
```
I (557) nvs: NVS key partition empty, generating keys
I (667) nvs: NVS partition "nvs" is encrypted.
```
It's repeated after following reset.

So I did an investigation:

Here is my partition table:
```
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 coredump         Unknown data     01 03 00010000 00010000
I (87) boot:  3 factory          factory app      00 00 00020000 00140000
I (94) boot:  4 ota_0            OTA app          00 10 00160000 00140000
I (102) boot:  5 ota_1            OTA app          00 11 002a0000 00140000
I (109) boot:  6 otadata          OTA data         01 00 003e0000 00002000
I (117) boot:  7 nvs_key          NVS keys         01 04 003e2000 00001000
```

1. With a logic analyzer hooked into SPI chip:

- Old bootloader(DIO) + old app (DIO): nvs_key is read from 0x003e2000 correctly
- New bootloader(QIO) + new app (QIO): nvs_key is read from 0x003e2000 correctly
- Old bootloader(DIO) + new app (QIO): no flash reading is done from 0x003e2000 ! Instead nvs_key partition is erased directly

2. In nvs_api.cpp `nvs_flash_read_security_cfg` I've added more log:
```
extern "C" esp_err_t nvs_flash_read_security_cfg(const esp_partition_t* partition, nvs_sec_cfg_t* cfg)
{
    if (cfg == nullptr || partition == nullptr) {
        return ESP_ERR_INVALID_ARG;
    }

    uint8_t buff[128];
    auto ret = esp_partition_read(partition, 0, buff, 128);
    ESP_LOGI(TAG, "Read part status: %d",ret);
    esp_log_buffer_hex(TAG, buff, 128);

    ret = esp_partition_read_raw(partition, 0, buff, 128);
    ESP_LOGI(TAG, "Read raw part status: %d",ret);
    esp_log_buffer_hex(TAG, buff, 128);
```

It turns out `esp_partition_read_raw` silent failed but returned `ESP_OK`, so NVS thinks the partition is not initialized.
```I (447) coexist: coexist rom version 9387209
I (452) cpu_start: Starting scheduler.
I (457) nvs: Read part status: 0
I (457) nvs: 48 ba 52 13 79 67 c3 06 39 a3 51 df 7a 85 53 6b
I (467) nvs: ad da 9a 70 9e 9a cf 90 ab 71 52 38 07 2e 0c 20
I (467) nvs: d7 f2 35 53 63 7b 85 b7 01 1a 3f d3 62 f9 18 6b
I (477) nvs: 83 89 71 b2 e8 f1 d3 99 53 57 97 ef f7 af 3f a5
I (477) nvs: af 71 28 d6 ff ff ff ff ff ff ff ff ff ff ff ff
I (487) nvs: ea f6 b6 a0 a8 41 30 35 c4 fd e8 20 07 b3 4f df
I (497) nvs: 72 f5 cd bc a4 d1 f8 9d f1 f3 0b 0e c6 87 04 86
I (497) nvs: 42 e1 18 1e b2 60 82 4e 54 12 33 e0 ef c9 0e b8
I (507) nvs: Read raw part status: 0
I (507) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (517) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (527) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (527) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (537) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (547) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (547) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (557) nvs: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
I (557) nvs: NVS key partition empty, generating keys
I (667) nvs: NVS partition "nvs" is encrypted.
```

### Steps to reproduce.

[read_raw.zip](https://github.com/espressif/esp-idf/files/9446284/read_raw.zip)

With this example project:
1. Build and flash with DIO and verify NVS works correctly.
2. Build and flash with QIO and verify NVS works correctly.
3. Leave DIO bootloader and flash QIO app and observe non-working NVS.

HiFiPhile, 2022-08-31:
@igrr, @mythbuster5 could you take a look? This is a major issue which makes QIO app brick devices with DIO bootloader if NVS is used.

mythbuster5, 2022-08-31:
> 

Yes, I looked into this issue just now. The reason is because flash itself need some configurations when it works under QIO mode. And the configuration is done in 2nd bootloader. So that's why DIO bootloader + QIO app does not work. BTW we don't recommend to do that because both  flash and memory spi configurations are continuous from bootloader to app, I'm afraid it will bring some unstable bugs...

If you strongly want to do that, my suggestion is to extern function `bootloader_enable_qio_mode` and put in function `esp_flash_init_default_chip` in esp_flash_spi_init.c

liike:

```
extern void bootloader_enable_qio_mode(void);
esp_err_t esp_flash_init_default_chip(void)
{
    bootloader_enable_qio_mode();
    const esp_rom_spiflash_chip_t *legacy_chip = &g_rom_flashchip;
    memspi_host_config_t cfg = ESP_FLASH_HOST_CONFIG_DEFAULT();

```

But you need to do some tests after applying this changes.

igrr, 2022-08-31:
Another thing we can do for now to prevent such issues is to check that the flash mode of the new image is same as the current one, while doing the OTA.

As mentioned in another issue https://github.com/espressif/esp-idf/issues/9542#issuecomment-1211317354 and by @mythbuster5 above, combining a DIO bootloader with a QIO app is not supported, so we might just as well try to prevent such combinations.

HiFiPhile, 2022-08-31:
>  The reason is because flash itself need some configurations when it works under QIO mode.

That's what I'm thinking about.

> Another thing we can do for now to prevent such issues is to check that the flash mode of the new image is same as the current one, while doing the OTA.

That's a possible solution. And also add flash mode check on app boot to prevent edge cases, such as manually flash mixed bootloader & app.

It's also a good idea to add documentation on spi mode menuconfig to clarify that flash mode shouldn't be changed.

Actually I've DIO devices running in the field and new validated QIO app , issue is only found trying to update old devices.

HiFiPhile, 2022-09-02:
Actually I think the preferred solution is fallback to DIO and give a warning, if QIO is not enabled.

(Hope there is a easy way to check if QIO is not enabled, as normal code/data read is running in DIO)

Thus compatibility is guaranteed and no risk of enabling QIO in App.

What do you think ? @mythbuster5 @igrr