espressif/esp-idf #9675: Poor iperf/network performance though ethernet/wifi bridge.

KenWatsonRL, 2022-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF V4.4  v4.4-367-gc29343eb94

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

Custom board  - Bridging ethernet to wifi with Lan8710 PHY.

### Power Supply used.

External 5V

### What is the expected behavior?

Expected to have iperf performance greater the 40Mbps in our test environment  (Ethernet/Wifi bridge).

### What is the actual behavior?

Boards recently built are only getting 25% of the iperf performance that older boards from 2019 are getting.

### Steps to reproduce.

steps to produce
1. Use iperf application to test though put.

### More Information.

Our device is based on the ESP32-WROOM32U.  All devices I've tested have version 1 of this chip.
The hardware uses an XR22802chip bridged to the ESP32 with a Lan8710 PHY chip with an external 50 Mhz clock.
The XR chip exposes an ethernet interface to our linux box.
The software built for the device is based on the esp-idf/examples/ethernet/eth2ap code. 

When I run an iperf test using our devices built from 2019, performance is great  (greater than 40 Mbps in my test enivornment).
Device recently built, are getting only 25% of this performance at best.

I've been trying to debug the issues with the newer devices and have tracked it down to two areas that I have questions about.

Below is email I sent to sales support email.
I did compile the IPERF example in the ESP-IDF 4.4 code base and loaded it onto all of the devices I have.  All these devices worked at > 20Mpbs.  This would suggest that the Wi-Fi interface is working on all our devices and issue is somehow in the Ethernet to Wi-Fi interface.

Our code is based on the eth2ap example in the V4.4 branch.   I added debug code to the firmware and have tracked down issues so far to two areas. First is the transmission of Ethernet 802.3 pack out the Wi-Fi interface using the ‘esp_wifi_internal_tx()’ function.
Second, Ethernet DMA appears to stop functioning on some boards, and not function at all on a couple my devices.

First issue:

esp_wifi_internal_tx() – called to transmit 802.2 Ethernet packet out the Wi-Fi interface.   On the boards that always have good performance, the average time to transmit packet during iperf test is 15ms, but on boards with poor performance, this average goes up to over 150ms.
The function returns an error code of 257 (no memory), but when I dump the heap statistics at this failure point, it has over 50K available.   What would cause this error to occur?   (Have no access to code as it is only supplied as part of the libnet80211.a static library.  Why would  esp_wifi_internal_tx() need to be called many times to transmit a packet on boards with poor performance?

This is code calling this function.   
     do {
            vTaskDelay(pdMS_TO_TICKS(timeout));
            total_current += timeout;
            timeout += 2;
            ret = esp_wifi_internal_tx(get_wifi_interface(bridge_init_config.wifi_mode),
                                       msg->buffer, msg->len);
        } while (ret && timeout < FLOW_CONTROL_WIFI_SEND_TIMEOUT_MS);

Second issue:

Another point of failure that I’ve found testing some of our boards is that the Ethernet DMA seems to stop functioning.    The esp_eth_transmit() calls start returning an error, which basically says it can’t allocating TX_DMA descriptors for the transfer.  I’ve put delays and issued retries for the packet, but the function just keeps returning the error.   What would cause the DMA to stop functioning?

MaxwellAlan, 2022-08-30:
Hi @KenWatsonRL 

>The function returns an error code of 257 (no memory), but when I dump the heap statistics at this failure point, it has over 50K available. 

WiFi internal have some limitation for buffer count to avoid too much heap usage, so will return no memory when there is no buffer in wifi(not no heap). 

- Can you share the sdkconfig of your application?
- Can you capture the wireless packets? If yes, please also upload the packets when the performace is low.
- How about have a try disable wifi modem sleep
`esp_wifi_set_ps(0);// call this after esp_wifi_start()`

Thanks.

KenWatsonRL, 2022-08-30:
> Hi @KenWatsonRL
>
>
> WiFi internal have some limitation for buffer count to avoid too much heap usage, so will return no memory when there is no buffer in wifi(not no heap).
>
> * Can you share the [...]

* Will share sdkconfig shortly.
* I'll set up test and provide a wireshark capture.
* I already call esp_wifi_set_ps(0) to turn off power saving feature.

My question with esp_wifi_internal_tx() is why does it take so long, on average, to transfer a 1514 byte packet, on our devices that are exhibiting this poor performance.  On good devices, this average is less that 15ms, on bad device, over 150-200ms.

MaxwellAlan, 2022-08-30:
Hi @KenWatsonRL 

> My question with esp_wifi_internal_tx() is why does it take so long, on average, to transfer a 1514 byte packet, on our devices that are exhibiting this poor performance. On good devices, this average is less that 15ms, on bad device, over 150-200ms.

IMO, there must be much rertry when tx these pkts, need to be check by wireless packets. Do you test this in shield box or open environments?

KenWatsonRL, 2022-08-30:
Its' in open.  The problem I have is that the environment is the same for all my tests - the only thing that changes is the device (our device) - the test environment is same for all tests.   Devices produced in 2019 have good through put, devices produced this year are having issues.  Why are the newer devices requiring more retries than older devices in same test enivroment.

negativekelvin, 2022-08-30:
>Boards recently built are only getting 25% of the iperf performance

Did you change any components, sourcing, chip revisions? Did you compare rf emissions?

KenWatsonRL, 2022-08-30:
Iperf results for sled/device  with poor performance.

[iperf_results_sled4.txt](https://github.com/espressif/esp-idf/files/9454668/iperf_results_sled4.txt)

Average time for esp_wifi_internal_tx() to succeed on transmitting ethernet packet out wifi interface.

[esp_wifi_internal_tx_average.txt](https://github.com/espressif/esp-idf/files/9454672/esp_wifi_internal_tx_average.txt)

Wireshark capture of test - was too big to upload.

KenWatsonRL, 2022-08-30:
> > Boards recently built are only getting 25% of the iperf performance
> 
> Did you change any components, sourcing, chip revisions? Did you compare rf emissions?

Components look to be the same.  The newer boards are built with ESP32-WROOM-32U Rev 1 chips with 4MB flash part.   The older boards are built with ESP32-WROOM-32U Rev chips, but with 16MB flash.

I personally have no equipment for measuring RF emissions.

KenWatsonRL, 2022-08-30:
Iperf results for older sled/device with good performance.

[iperf_results_sled1.txt](https://github.com/espressif/esp-idf/files/9454869/iperf_results_sled1.txt)

Average time for esp_wifi_internal_tx() to succeed on transmitting ethernet packet out wifi interface.

[esp_wifi_internal_tx_average_sled1.txt](https://github.com/espressif/esp-idf/files/9454873/esp_wifi_internal_tx_average_sled1.txt)

KenWatsonRL, 2022-08-30:
Is there a debug version on the libnet80211.a library?

KenWatsonRL, 2022-08-31:
Is there a date code for when the ESP32-VROOM-U32 chip was made?

KenWatsonRL, 2022-09-01:
There does appear to be a QR code on the ESP32 - would this contain a data code of when it was manufactured?

KenWatsonRL, 2022-09-01:
My next move is to swap a couple on ESP32 chips from older devices (with the good performance) with ones from the newly produced devices and see if the issue follows the chip or the board.

KenWatsonRL, 2022-09-01:
I loaded the example/wifi/iperf application onto 10 of my devices, configured each, one a time, as AP, and connected my laptop.
I ran the iperf -c 192.168.2.4 from the iperf CLI.
Performance was 50 Mbits/second on each of these devices.
This is only testing the 802.11 packets, but shows that the Wifi portion seems to be working just fine.

Question still is why does esp_wifi_internal_tx() take so much longer on these boards - I believe this function is just re-packaging the 802.3 raw frame that was received on from the ethernet side of the bridge into 802.11 frames and sending them.

negativekelvin, 2022-09-01:
When your Ethernet interface (which has its own clock source[s]) is active there could be noise generated that reduces the performance of the radio in the wroom32 causing packet loss, retries, and full buffers.  That's why it would be helpful to scan rf emissions or at least build some rf shielding over suspected parts of the board using copper tape.

KenWatsonRL, 2022-09-05:
I modified the example/wifi/iperf code to include the initialization of the ethernet hardware.   There is a callback function defined that reports that it receives ethernet frames, and then just frees the buffer.  
 
 This does not change the behavior for the wifi-only iperf results. 
 
 Would this not imply that the ethernet  hardware is not causing 'noise' that interferes with radio?  Or because I'm only processing ARP requests, that the noise might not be generated?
 
 Will be working with hardware team this week.

KenWatsonRL, 2022-09-05:
I also compiled the eth2ap example with configuration settings set for our setup (PHY address, SMI gpios).   I set the IP addresses for the two endpoints manually.   This example code behaves the the same way as our code - older boards get good IPERF performance, boards built recently do not.

KenWatsonRL, 2022-09-18:
This actually may be a production issue with the board - some poor solder connections on ESP32.

KenWatsonRL, 2022-10-12:
We did find an issue with soldering, but that has not totally fixed the issue.    The iperf performance for our boards behaves differently when the board is not in its enclosure ( bare board)  than when it is enclosed in it's case.

When it's a bare board, the iperf performance is greater than 40Mbps through our bridge.  When the board is in it's case, the numbers go way down, sometimes to sub 1Mbps.

Normally, the ESP32 antenna is connected from the ESP32 to an antenna connector on our PCB board.  It's strange that it behaves differently when in a case, than when it is bare board (no case).

One experiment we've done is to directly connect the ESP32 with an external antenna, by passing the antenna connector on our board.  This seems to work for both bare board/ in case boards.

Are there any tools or API functions that can be used to diagnose the behavior when using board mounted in case and using our onboard antenna connector?

MaxwellAlan, 2022-10-17:
Hi @KenWatsonRL , according to ESP RF engineer,

>The iperf is greater than 40Mbps with bare board,  but 1Mbps when the board in it's case.

Please double check whether the antenna deformed or damaged when installed in case. 
    Please make sure that the case is not metallic, and if we put the external antenna outside the case, the ipef will change to 40Mbps again.
    Are there any other cicult that can affect the performace when the board installed in it's case. 

It will be better to claffity the problem if you can provide the photos before and after installed in the case.

KenWatsonRL, 2022-10-17:
The case is metallic - actually a heat sink connected to the circuit board using thermal pads.

I will look into providing some photos of the hardware (original setup and with our current use of an external antenna connected directly with the ESP32.

KenWatsonRL, 2022-10-18:
[image]

KenWatsonRL, 2022-10-18:
[image]

KenWatsonRL, 2022-10-18:
If I remove the thermal pad completely, the iperf test also work fine.

MaxwellAlan, 2022-10-19:
Hi @KenWatsonRL 

Please just remove the left part of the thermal pad, test without and with case. 

[image]

KenWatsonRL, 2022-10-19:
With just the blue thermal pad covering the chips at the bottom of picture, including the XR chip to the left, the throughput with iperf is the same in or out of the case.

MaxwellAlan, 2022-10-20:
Hi @KenWatsonRL 

The thermal pad itself would not cause issues with the RF cable/signal, According to the deep indent from the photo below, I think that the RF cable is in a bad connection when the case is installed.
[image]

negativekelvin, 2022-10-20:
Thermal pad pushes down on cable, creates torque on connector, causes partial disconnection, bad rf performance.

closed 2023-02-17 by Alvin1Zhang