espressif/esp-idf #9677: How to subscribe Group address in BLE Mesh In esp32 without mobile app

JegadeeshXaglabs, 2022-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all,
I am tiring to create a BLE mesh with esp32 which can send and receive msg to all nodes.

What i done:
I can send a msg to all nodes with nrfApp after i group the nodes
What i cant
I cant send a msg to all nodes without App from client (When Vendor client automatically bind the key with nodes). Because i cant group the nodes.

Suggest me the way to subscribe group address to all nodes or any other way to send and receive msg or binding key programmatically.

JiaLWang, 2022-09-18:
Hi @JegadeeshXaglabs 

Sorry for the late reply. If you want to send message to all nodes, you don't need to subscribe any group address, you only need to  set the DST of the message to 0xFFFF. 

If you want to subscribe group address, you can use the function of `esp_ble_mesh_model_subscribe_group_addr()`. And config client send `config model app bind` message to config server to bind the appkey to the model.

Hope it can help you.

JegadeeshXaglabs, 2022-09-20:
Thanks for the help @JiaLWang .it worked

closed 2022-09-20 by JegadeeshXaglabs

ShahinHaque97-CT, 2023-09-12:
@JiaLWang @JegadeeshXaglabs 
Can you guys explain to me how this command works and potentially point out the issue I am having

I have an ESP provisioning an nRF58232 and I am able to get temperature requests no issue. Now I want to subscribe the nRF to a group address of   0xC000, Nordic default company is 0x058.

AT-BMNS_TEST is just the ESP_LOG name

To test I have tried all the available models

```
I (17102) BMNS_COMMON: ********************** Composition Data Start **********************
I (17112) BMNS_COMMON: * CID 0x0059, PID 0x0000, VID 0x0000, CRPL 0x000a, Features 0x0007 *
I (17122) BMNS_COMMON: * Loc 0x0001, NumS 0x03, NumV 0x00 *
I (17132) BMNS_COMMON: * SIG Model ID 0x0000 *
I (17132) BMNS_COMMON: * SIG Model ID 0x1100 *
I (17142) BMNS_COMMON: * SIG Model ID 0x1101 *
I (17142) BMNS_COMMON: *********************** Composition Data End ***********************
I (17162) BMNS_PROV: ESP_BLE_MESH_PROVISIONER_STORE_NODE_COMP_DATA_COMP_EVT, err_code 0
W (17382) BMNS_CONFIG: Provision and config successfully VENDOR
W (24802) BLE_MESH: Receive configuration status message timeout
I (32212) AT-BMNS_TEST: argument count 1
I (32212) AT-BMNS_TEST: primary address 1
I (32212) AT-BMNS_TEST: element_address 1
I (32222) AT-BMNS_TEST: sig_model_count 3
I (32222) AT-BMNS_TEST: vnd_model_count 1
I (32232) AT-BMNS_TEST: [0] sig_models->model_id 0x0000
I (32232) AT-BMNS_TEST: [1] sig_models->model_id 0x0001
I (32242) AT-BMNS_TEST: [2] sig_models->model_id 0x1102
E (32242) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x0000
E (32252) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x0001
E (32262) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x1102
E (32272) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x0059
E (32272) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x0006

```