espressif/esp-idf #9678: How to use esp_event_loop_create_default() function with more than one event handler

marco711, 2022-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I am using an ESP32-C3 to connect to Wi-Fi and send data in MQTT protocol. For the Wi-Fi part I'm basing in the following example:
https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c

And for the MQTT part:
https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/tcp/main/app_main.c

The problem I'm having is that the `esp_event_loop_create_default()` function seems to call only one of the event handlers (`mqtt_event_handler` or `event_handler` for Wi-fi). It is not clear for me how do I have to declare the event handlers so both of them are called after the program starts.

Is there any example for this?

igrr, 2022-08-30:
Hi @marco711,
Could you please post the code where you are calling `esp_event_handler_register` or `esp_event_handler_instance_register`? This could help pinpoint the problem.

Normally all the event handlers registered for the particular type of event will be called when this event occurs.

I think one particular difference of the mqtt component is that it has its own event loop; it doesn't use the default system event loop created using esp_event_loop_create_default. Therefore you need to use the function `esp_mqtt_client_register_event` when registering an event handler for mqtt events. Registering an event handler for mqtt events using `esp_event_handler_register` likely will not work.

marco711, 2022-08-30:
Hello @igrr 
I call the `esp_event_handler_instance_register` only for the Wi-Fi start function:

```
/**
  * \brief  		Connect to the Wi-Fi access point.
	* \note				The SSID and pasword are in the 'station.h' file.
  */
static void start_station(void)
{
    s_wifi_event_group = xEventGroupCreate();

    //ESP_ERROR_CHECK(esp_netif_init());

   // ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    esp_event_handler_instance_t instance_any_id;
    esp_event_handler_instance_t instance_got_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_any_id));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        IP_EVENT_STA_GOT_IP,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_got_ip));

		/* Set the Wi-Fi access point parameters. */
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS,
            /* Setting a password implies station will connect to all security modes including WEP/WPA.
             * However these modes are deprecated and not advisable to be used. Incase your Access point
             * doesn't support WPA2, these mode can be enabled by commenting below line */
	     .threshold.authmode = WIFI_AUTH_WPA2_PSK,

            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI(TAG, "wifi_init_sta finished.");

    /* Waiting until either the connection is established (WIFI_CONNECTED_BIT) or connection failed for the maximum
     * number of re-tries (WIFI_FAIL_BIT). The bits are set by event_handler() (see above) */
    EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
            WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
            pdFALSE,
            pdFALSE,
            portMAX_DELAY);

    /* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
     * happened. */
    if (bits & WIFI_CONNECTED_BIT) {
        ESP_LOGI(TAG, "connected to ap SSID:%s password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else if (bits & WIFI_FAIL_BIT) {
        ESP_LOGI(TAG, "Failed to connect to SSID:%s, password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else {
        ESP_LOGE(TAG, "UNEXPECTED EVENT");
    }

    /* The event will not be processed after unregister */
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
    vEventGroupDelete(s_wifi_event_group);
}
```
For the MQTT I am not calling this function as I have not found the event name.

igrr, 2022-08-30:
Thanks for the code, it looks find and I have checked that it works, when placed into the wifi/getting_started/station example, with some lines commented out.

Is your program by any chance also calling `example_connect()`?
```c++
    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());
```
If yes, you need to remove that part, since the [example_connect function](https://github.com/espressif/esp-idf/tree/master/examples/protocols#about-the-example_connect-function) sets up its own event handlers responsible for the Wi-Fi connection. Since you have added this logic to your application, the example_connect call is not needed anymore.

marco711, 2022-08-30:
No, I am not using `example_connect()`. When calling this function I have not any problem. Nevertheless, later I will need to configure the Wi-Fi parameters (ID and password) without ESP configuration tool. That's why I'm using the wifi_station example.

igrr, 2022-08-30:
I have put together this code by merging your snippet with the wifi_station example: https://gist.github.com/igrr/2af72987a51802bb8b1643f37e1490af. The log is attached, from the log I can see that the event handler gets triggered for both Wi-Fi and IP events.

If this doesn't work on your end, could you please do the same and upload the entire source file as well as the log to gist.github.com, and leave the link here?

marco711, 2022-08-30:
This works. I have the same code. The problem I have is when I try to integrate the MQTT layer.

igrr, 2022-08-30:
Could you please post the entire source file which demonstrates the issue, or if it is too big, reduce to the smaller example which still has the same issue? The part of the code which you posted doesn't have an issue, as we have found, so the issue must be somewhere else in the code. Please also upload the log output, like I did in that gist above.

marco711, 2022-08-30:
Below the link with the project:

https://gist.github.com/marco711/04ce7decc597d7d4667d48cf92ad8f75.js

I can connect to the Wi-Fi AP. The problem is that the MQTT communication is executed only once. If I delete the Wi-Fi code and use the `example_connect()` function the MQTT publshing is executed several times.

igrr, 2022-08-30:
Thank you for the code! Couple observations:

* `wifi_station` function doesn't seem to be called anywhere (at least I don't see the call in `app_main` or any other function which is called by `app_main`). If you have removed some code before sharing it, please check if you have accidentally removed that call.
* Based on the log, the sequence of events seems to be:
  - `wifi_init_sta finished` — logged when Wi-Fi gets enabled
  - MQTT connection starts, without waiting for the Wi-Fi connection to finish or for the IP address to be received:
    ```
    E (640) esp-tls: couldn't get hostname for :toolbox.strong.blue: getaddrinfo() returns 202, addrinfo=0x0
    E (650) TRANSPORT_BASE: Failed to open a new connection: 32769
    ```
  - `I (730) wifi:connected with TP-Link_Outdoor_022078` — Wi-Fi actually connects

So it seems that you are starting the MQTT connection without waiting for the `wifi_station` function to finish. Since I don't see where `wifi_station` is called, I can't say whether that's actually the case or not...

marco711, 2022-08-31:
Hello @igrr ,
Sorry for the mistake. When I reduced the code I skipped the call to the `wifi_station()` function. I have added it and below you can see the log:

```
Executing action: monitor
Running idf_monitor in directory /home/msurca/esp/IoTliftv1_0_1/mqtt-tcp
Executing "/home/msurca/.espressif/python_env/idf5.0_py3.9_env/bin/python /home/msurca/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB1 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace /home/msurca/esp/IoTliftv1_0_1/mqtt-tcp/build/mqtt_tcp.elf -m '/home/msurca/.espressif/python_env/idf5.0_py3.9_env/bin/python' '/home/msurca/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB1'"...
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6108,len:0x1970
load:0x403ce000,len:0x904
load:0x403cf600,len:0x2eb0
entry 0x403ce000
I (30) boot: ESP-IDF v5.0-dev-3290-g01d014c42d-dirty 2nd stage bootloader
I (30) boot: compile time 11:39:58
I (30) boot: chip revision: 3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=22dd0h (142800) map
I (122) esp_image: segment 1: paddr=00032df8 vaddr=3fc8f000 size=02c98h ( 11416) load
I (124) esp_image: segment 2: paddr=00035a98 vaddr=40380000 size=0a580h ( 42368) load
I (135) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=9ab40h (633664) map
I (235) esp_image: segment 4: paddr=000dab68 vaddr=4038a580 size=04a30h ( 18992) load
I (239) esp_image: segment 5: paddr=000df5a0 vaddr=50000010 size=00010h (    16) load
I (245) boot: Loaded app from partition at offset 0x10000
I (247) boot: Disabling RNG early entropy source...
I (264) cpu_start: Pro cpu up.
I (273) cpu_start: Pro cpu start user code
I (273) cpu_start: cpu freq: 160000000 Hz
I (273) cpu_start: Application information:
I (276) cpu_start: Project name:     mqtt_tcp
I (281) cpu_start: App version:      1
I (285) cpu_start: Compile time:     Aug 29 2022 17:06:02
I (291) cpu_start: ELF file SHA256:  350b19ae3f7e1299...
I (297) cpu_start: ESP-IDF:          v5.0-dev-3290-g01d014c42d-dirty
I (305) heap_init: Initializing. RAM available for dynamic allocation:
I (312) heap_init: At 3FC95ED0 len 0002A130 (168 KiB): DRAM
I (318) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (325) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (331) spi_flash: detected chip: generic
I (336) spi_flash: flash io: dio
W (340) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (353) sleep: Configure to isolate all GPIO pins in sleep state
I (360) sleep: Enable automatic switching of GPIO sleep configuration
I (367) cpu_start: Starting scheduler.
I (377) wifi station: ESP_WIFI_MODE_STA
I (377) pp: pp rom version: 9387209
I (377) net80211: net80211 rom version: 9387209
I (397) wifi:wifi driver task: 3fc9ec84, prio:23, stack:6656, core=0
I (397) system_api: Base MAC address is not set
I (397) system_api: read default base MAC address from EFUSE
I (407) wifi:wifi firmware version: 0ca6d0b
I (407) wifi:wifi certification version: v7.0
I (407) wifi:config NVS flash: enabled
I (417) wifi:config nano formating: disabled
I (417) wifi:Init data frame dynamic rx buffer num: 32
I (427) wifi:Init management frame dynamic rx buffer num: 32
[... 22 lines cut ...]
I (597) wifi:state: assoc -> run (10)
I (617) wifi:connected with TP-Link_Outdoor_022078, aid = 1, channel 6, 40D, bssid = b0:a7:b9:02:20:78
I (617) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (617) wifi:pm start, type: 1

I (617) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (627) wifi:BcnInt:102400, DTIM:1
I (1377) esp_netif_handlers: sta ip: 192.168.1.63, mask: 255.255.255.0, gw: 192.168.1.254
I (1377) wifi station: got ip:192.168.1.63
I (1377) wifi station: connected to ap SSID:TP-Link_Outdoor_022078 password:@nep1234
I (1387) MQTT_DATA: Other event id:7
W (1397) wifi:<ba-add>idx:0 (ifx:0, b0:a7:b9:02:20:78), tid:0, ssn:0, winSize:64
I (1467) MQTT_DATA: MQTT_EVENT_CONNECTED
I (1467) MQTT_DATA: sent publish successful, msg_id1=9778
I (1467) MQTT_DATA: sent subscribe successful, msg_id2=51611
I (1467) MQTT_DATA: sent subscribe successful, msg_id3=13613
I (1477) MQTT_DATA: sent unsubscribe successful, msg_id4=18699
I (1487) MQTT_DATA: MQTT_EVENT_PUBLISHED, msg_id7=9778
I (1517) MQTT_DATA: MQTT_EVENT_SUBSCRIBED, msg_id=51611
I (1517) MQTT_DATA: sent publish successful, msg_id5=0
I (1527) MQTT_DATA: MQTT_EVENT_SUBSCRIBED, msg_id=13613
I (1527) MQTT_DATA: sent publish successful, msg_id5=0
I (1537) MQTT_DATA: MQTT_EVENT_UNSUBSCRIBED, msg_id6=18699
```
But when I run the MQTT example using the `example_connect()` function the MQTT events are published several times.

igrr, 2022-08-31:
I see, thanks for this log. Indeed it seems that while `sent publish successful` is printed a couple times, only one MQTT_EVENT_PUBLISHED is shown. I'll ask my colleagues working on the MQTT component to help here.

chegewara, 2022-08-31:
Just an observation. `msg_id` is a random id and shall not be 0, especially 2 times in a row:
```
I (1517) MQTT_DATA: sent publish successful, msg_id5=0
I (1527) MQTT_DATA: sent publish successful, msg_id5=0
```

Im not sure i understand OP properly, but you dont have to call `esp_event_loop_create_default` few times:
```
esp_err_t esp_event_loop_create_default(void)
{
    if (s_default_loop) {
        return ESP_ERR_INVALID_STATE;
    }

    esp_event_loop_args_t loop_args = {
        .queue_size = CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE,
        .task_name = "sys_evt",
        .task_stack_size = ESP_TASKD_EVENT_STACK,
        .task_priority = ESP_TASKD_EVENT_PRIO,
        .task_core_id = 0
    };

    esp_err_t err;

    err = esp_event_loop_create(&loop_args, &s_default_loop);
    if (err != ESP_OK) {
        return err;
    }

    return ESP_OK;
}
```
as you can see, if `esp_event_loop_create_default` has been created, then next call will return error, which in my opinion is wrong way to do it. Function should return `ESP_OK` since default handler already exist and can be used, or any `ESP_ERROR_CHECK` will cause app failure even if it code is valid, but that just my opinion.

chegewara, 2022-08-31:
To be honest i dont see the code which can be used to recreate the problem.
Code from https://github.com/espressif/esp-idf/issues/9678#issuecomment-1231877264 works good and logs https://github.com/espressif/esp-idf/issues/9678#issuecomment-1232593210 does not show duplicated events.
Another thing is that you can have multiple event listeners listening exactly the same or mix of events, so its possible to receive the same events multiple times.

marco711, 2022-08-31:
@chegewara, the problem is that the code in https://github.com/espressif/esp-idf/issues/9678#issuecomment-1231877264 sends the MQTT message only once. I'd like to know how can I do to send the MQTT messages periodically.

chegewara, 2022-09-01:
Make the client global:
https://gist.github.com/marco711/04ce7decc597d7d4667d48cf92ad8f75#file-mqtt_communication-c-L55
call publish from anywhere?
https://gist.github.com/marco711/04ce7decc597d7d4667d48cf92ad8f75#file-mqtt_communication-c-L95

marco711, 2022-09-01:
@chegewara this works, thank you. If I can call `esp_mqtt_client_publish` function from anywhere in the code, then which is the use of the `mqtt_event_handler` function?

chegewara, 2022-09-01:
You can use event handler for many purposes in your code.

euripedesrocha, 2022-09-05:
Hi @marco711,
In the example we use the `mqtt_event_handler` to publish, but as you figured it out you don't need to do it in the event handler. 
For publishing, you can call either `esp_mqtt_client_publish` or `esp_mqtt_client_enqueue` from anywhere in your code.

About the `event_handler` you can see the description of all events that will trigger the call of the registered function [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#events)

marco711, 2022-09-06:
Hello @euripedesrocha,
Thanks for your answer. Now it is clearer for me how to use the MQTT library. I will close this issue.

closed 2022-09-06 by marco711