espressif/esp-idf #9680: Custom error message sent by httpd_resp_send_err reaches browser only half of the time

kriegste, 2022-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2 release

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

TTGO

### Power Supply used.

USB

### What is the expected behavior?

In a http server project, I use

```
httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "CUSTOM MESSAGE");
return ESP_FAIL;
```

to inform the user if e.g. a file upload went wrong (and what went wrong). In the html file I use the JavaScript XMLHttpRequest API.
I expect the message "CUSTOM MESSAGE" to arrive in the browser (in the responseText field of the XMLHttpRequest object) every time.

### What is the actual behavior?

My problem is that this does not work every time. About half the time the onload event is triggered and the custom message gets through. But in just as many cases onerror gets triggered any no message is received. No message and no status number.
I tried Firefox and Chrome, both on Windows.

### Steps to reproduce.

The file_server example can demonstrate the problem easily.

Just try to provoke upload errors:
Upload a file which already exists. Do it repeatedly (ten times or so) and you will see the error message does not appear every time.
Often the alert window just says "Server closed the connection abruptly!".

### More Information.

Option CONFIG_HTTPD_ERR_RESP_NO_DELAY is enabled. I guess this option was once designed to prevent the issue from happening, but it still occurs.

I found out, if I return ESP_OK instead of ESP_FAIL, the message is transferred correctly. The problem with this is then one has to wait for the whole file to be uploaded.

hmalpani, 2022-09-06:
Hello @kriegste

I tried to reproduce the issue with the file_serving example. I was able to see that the error message was not received on the browser when I was using Firefox. But when I used Google Chrome, I could see the error message.
I also verified Wireshark capture for response from the server, and it contained the error message. PTAL at images attached:

I think it is not an issue on the server side rather it is on the client/browser side.

Captured packets:
[image]

[image]

kriegste, 2022-09-07:
That is very strange. I get this on Edge browser too and even under Android (Chrome, but it takes a few more attempts to see it). Next I'll try accessing via the SoftAP to rule out my router as the problem's cause.

kriegste, 2022-09-07:
Trying this via the SoftAP, the issue shows up much more frequently, both under Windows (Chrome, Firefox, Edge) and Android (Chrome).
I do not believe in a client issue.

hmalpani, 2022-09-08:
Hello @kriegste,

Can you share sdkconfig file as well as debug logs for your example? Also if possible send the Wireshark capture of the response from the server. I can see the packets sent by the server containing the error message at my end.

Can you also elaborate on what do mean by accessing via the SoftAP?

kriegste, 2022-09-08:
The SoftAP is the access point the ESP32 opens after e.g. esp_wifi_set_mode(WIFI_MODE_APSTA);

I attached the Wireshark capture. The last upload attempt is the one where the error message did not reach the browser (Chrome).

[wireshark.zip](https://github.com/espressif/esp-idf/files/9530610/wireshark.zip)

hmalpani, 2022-09-09:
Hello @kriegste,
In the Wireshark capture you sent, I can see the error message present in the packet (see attached screenshot).

[image]

kriegste, 2022-09-09:
I know. I am no Wireshark expert, but isn't ZeroWindow a hint to something being wrong (busy server or the like)?

kriegste, 2022-09-09:
The problem also occurs when there is no ZeroWindow hint.
I accidentally cut out the packets that were sent by the client side. I noticed that the final ACK is missing when the problem occurs. Maybe I'll ask in the Chrome bug tracker, too. And the Firefox bug tracker..... but I cannot believe two different browsers having the same issue.

hmalpani, 2022-09-14:
I tried to reproduce the issue with CURL.
[image]

Here we can see that custom error is received. This is the same file through which I reproduced the error using firefox. And using the same file, error was not reproduced using chrome and CURL. I think this issue is on the client side and not on server side.

I am closing this issue as per above conclusion. Please feel free to reopen.

closed 2022-09-14 by hmalpani

kriegste, 2024-11-01:
Problem was still present two years after opening this issue, even in IDF 5.3.1 and after countless Firefox and Chrome updates. So I still was sure it is on the server side.
But I could recently solve it by setting

enable_so_linger = true;
linger_timeout = 5;

in httpd_config_t. The error message gets through now every time.