espressif/esp-idf #9682: Hardcoded (128 byte) UART buffer size.

ammaree, 2022-08-30:
### Is your feature request related to a problem?

I believe the UART support in general needs some improvement.
The hardware FIFO is 1024 bytes big but only 256 bytes are used by default, 128 for RX and 128 for TX. In fact, the 128 byte limit is hardcoded

Many ESP32 projects (majority?) uses only a single UART and I would guess very very few uses more than 2 UART ports.
IDF does not provide any API to change the use of the FIFO buffer to better use the full 1024 bytes available.

### Describe the solution you'd like.

We would like to see 1 or more API's to specify sizes for RX and TX buffers on a per UART basis.
In the case of an application using a single UART and largely for console output TX=1024/RX=0 or TX=896/RX=128
For 2x UART and a more balanced communication UART0 TX=256/RX=256 and UART1 TX=256/RX=256 

### Describe alternatives you've considered.

This restriction has forced us to avoid the UART APIs and provide our own (limited) single UART support for buffers of TX/896 and RX/128 bytes

This has allowed the benefit of the full 1024 bytes and 2Mbps speeds, without any SRAM based buffering, but has forced us to wrap numerous IDF APIs at link to to overcome the restrictions.

Another small benefit of this has been the removal of the need for all UART APIs somewhat reducing code size.

ginkgm, 2022-08-31:
Hi @ammaree ,

We didn't enable this feature, because this will affect all UART channels, thus not so very convenient.

Since the HW FIFO is not scalable and not economic, we don't expect HW FIFO to deal with most user problems. Usually, the HW FIFO is designed to overcome the interval that ISR can't re-fill TX data/handle RX data. While the SW FIFO is to deal with the interval that tasks can't re-fill TX data/handle RX data.  

We have seen several requests from customers to user larger UART FIFO size in the driver, but it turns out that they don't have enough SW FIFO size compared to their task schedule interval - In this case, even if we use the largest FIFO size, let's say, 512 byte, it still can't meet the requirement. The issue was finally solved by increase the SW FIFO size.

So do you have analyzed what's the problem you meet? Is it a HW FIFO overflow, or a SW FIFO overflow? What's your baudrate, and what's the no-response interval of your ISR/task? Is it really solvable by increasing HW FIFO size?

Alvin1Zhang, 2022-09-21:
@ammaree Thanks for reporting, would you please help share if any updates? Thanks.

AxelLin, 2022-09-21:
> Hi @ammaree ,
>
> We didn't enable this feature, because this will affect all UART channels, thus not so very convenient.
>
> Since the HW FIFO is not scalable and not economic, we don't expect HW FIFO to [...]

@ginkgm 
I understand your comment and the hardcoded 128 bytes probably works for most use cases.
But the users have no chance to make a comparison with the largest FIFO size without modifying esp-idf code now.
AFAICT, the hardware FIFO is 1024 bytes, then the esp-idf should support it.
You cannot add artificial limitation on h/w usage and then say that feature is not enabled. It's simply a bug that needs fix.

ammaree, 2022-09-21:
I agree for @AxelLin , the limitation imposed is unnecessarily restrictive and can be avoided.
uart_dev_t member mem_conf rx_size & tx_size can easily be used to determine the actual buffer allocation for each UART port.
In this way the fixed limit of 128 bytes can be removed and the full buffer can be utilised.

beginner140BB, 2023-01-05:
@AxelLin @ammaree 

"I was wondering if you could help me out by sharing some sample code on how to modify the value? 
As the field creates a bit-field with a width of 4 bits as seen in the library.

```
uint32_t rx_size:   4; /*This register is used to configure the amount of mem allocated to receiver's fifo. the default byte num is 128.*/
uint32_t tx_size:   4; /*This register is used to configure the amount of mem allocated to transmitter's fifo.the default byte num is 128.*/
```

how will someone change the value according to there desire if i want rx buff as 512bytes
It would be a huge help to see how the code works in practice.

Thanks for your time"

ammaree, 2023-01-06:
@beginner140BB, from what I can remember....

The RX and TX buffer size is specified in blocks of 128 bytes with 0=0bytes, 1=128 bytes and 8=1024 bytes

To read/write the current RX buffer size:
`
u32_t Blocks = HAL_FORCE_READ_U32_REG_FIELD(psUart->mem_conf, rx_size)
HAL_FORCE_MODIFY_U32_REG_FIELD(psUart->mem_conf, rx_size, Blocks)
`

To do the same for the TX buffer substitute `rx_size` with `tx_size`

To properly read the full number of bytes in either RX or TX FIFO you need to:
`
 u32_t xUsed = HAL_FORCE_READ_U32_REG_FIELD(psUart->mem_cnt_status, rx_cnt) << 8;
 xUsed += HAL_FORCE_READ_U32_REG_FIELD(psUart->status, rxfifo_cnt);
`

Again, to do the same for the TX FIFO substitute `rx_cnt` with `tx_cnt` and `rxfifo_cnt` with `txfifo_cnt`

I hope this helps you.

beginner140BB, 2023-01-09:
@ammaree 
thanks for  the guidance👍.
the value is changing as per your direction.

But there is some data missing in the UART which is not directly linked to changing the value i.e changing the value didn't solve the issue.

from your experience and expertise based on probelms related to uart ,could you please check on to the issue if  I provide you with the code flow and a brief explanation of the problem?

ammaree, 2023-01-09:
Happy to have a look but no promises. I am not a UART

beginner140BB, 2023-01-09:
code is having 3 tasks
task 1=> handle uart data
task2 and 3 ==> handle bluetooth related functions

**uart init function**

```
void uart_init(int BaudRate)
{
    const uart_config_t uart_config = {
    		.baud_rate = BaudRate,
    		.data_bits = UART_DATA_8_BITS,
    		.parity = UART_PARITY_DISABLE,
    		.stop_bits = UART_STOP_BITS_1,
    		.flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
			.source_clk = UART_SCLK_APB,
    };
    // We won't use a buffer for sending data.
    uart_driver_install(UART_NUM_0, 2048, 2048, 20, &uart0_queue, 0);
    uart_param_config(UART_NUM_0, &uart_config);
    uart_set_pin(UART_NUM_0, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);

}
```

task1 

 ```
 xTaskCreatePinnedToCore(
	    Task1,     /* Task function. */
	    "Task1",       /* String with name of task. */
	    21000,            /* Stack size in words. */
	    NULL,             /* Parameter passed as input of the task */
	    2,                /* Priority of the task. */
	    &xHandle1,                /* Task handle. */
	    1);/* Core where the task should run */

	
void Task1(void *pvParameters)
{
  char Serial_data1[1024];
  char validation=0;
  char *cTkptr;
  uart_event_t event;
  //printf((char*)"task1:%u\n\r", xPortGetFreeHeapSize());
  for (;;)
  {

	//printf("Task 1\n");
    memset(Serial_data, 0, sizeof(Serial_data));
    memset(Serial_data1, 0, sizeof(Serial_data1));
    memset(copy_buff, 0, sizeof(copy_buff));
    if(xQueueReceive(uart0_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
	switch( event.type)
	{
		case UART_DATA:
			//printf("UART_DATA\r\n");
			xSemaphoreTake( xMutex, portMAX_DELAY );// Reading serial commands from palmtec
			uart0_rx(Serial_data1);
			/*		
			int uart0_rx(char *data)
			{
			int length = 0;
			ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_0,(size_t*)&length));
			length=uart_read_bytes(UART_NUM_0,data,1024, 20 / portTICK_RATE_MS);
			uart_flush_input(UART_NUM_0);
			return length;
			}
			*/
			xSemaphoreGive( xMutex );
			break;
		case UART_FIFO_OVF:

			// If fifo overflow happened, you should consider adding flow control for your application.
			// The ISR has already reset the rx FIFO,
			// As an example, we directly flush the rx buffer here in order to read more data.
			printf("UART_FIFO_OVF\r\n");
			uart_flush_input(0);
			xQueueReset(uart0_queue);
			break;
		//Event of UART ring buffer full
		case UART_BUFFER_FULL:
			// If buffer full happened, you should consider encreasing your buffer size
			// As an example, we directly flush the rx buffer here in order to read more data.
			printf("UART_BUFFER_FULL\r\n");
			uart_flush_input(0);
			xQueueReset(uart0_queue);
			break;
		case UART_BREAK:
			printf("UART_BREAK\r\n");
		break;
		//Event of UART parity check error
		case UART_PARITY_ERR:
			printf("UART_PARITY_ERR\r\n");
		break;
		//Event of UART frame error
		case UART_FRAME_ERR:
			printf("UART_FRAME_ERR\r\n");
		break;
		//UART_PATTERN_DET
		case UART_PATTERN_DET:
		break;
		default:
			break;
	}
    }
	
	// process data from Serial_data1 for further uses
	
	}
```

xSemaphoreTake( xMutex, portMAX_DELAY ); is used to block other uart related functionalities when reading uart data

beginner140BB, 2023-01-09:
not getting serial data means sometime UART_FIFO_OVF event occurs

```
curr_data_len :225
uart data:$1,211,172,0,SENDDATA,1,1,1,G,HTTPGET,HTTPGET,SaveTicket?data=tkt|50609012023151839|506|195|1|3|6|1|1|3|257|1|0|0|0|0|24.00|0.00|0.00|0.00|0.00|0.00|0|24.00|09/01/2023|15:18:39|2|E261706111||0.00|0.00|0|0|50|552|10163|,14883
$0,OK,0,0,0,0,0,0,0,798*
curr_data_len :225
uart data:|0.00|0.00|0.00|0.00|0|24.00|09/01/2023|15:18:39|2|E261706111||0.00|0.00|0|0|50|552|10164|,14885
UART_FIFO_OVF
```

here we can see when data is on uart at the second instant data length is showing 225bytes but  when we print the data buffer its not showing the full data and  UART_FIFO_OVF  event occurs

do you have any suggestions on this to avoid this situation?

kayzen9, 2023-01-09:
Try using api:

https://github.com/espressif/esp-idf/blob/4c98bee8a4417d92a387ef2416be14ff1fdfa92a/components/driver/include/driver/uart.h#L693-L706

default threshold is 120, set lower value than that. it will allow hardware fifo to be emptied  erlier and hence avoid loss of data if there is delay in uart rx int proces.

beginner140BB, 2023-01-09:
> @kayzen9 
still there is uart overflow issue
[image]

ammaree, 2024-02-14:
@ginkgm 
@Alvin1Zhang 

+1 for getting this fixed so full 1024 bytes of FIFO buffer can be used.
Please!!!