espressif/esp-idf #9684: Enable spiram and call esp_now_send() causes dhcp timeout

sxsx2yzyz, 2022-08-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### Development Kit.

esp32-wrover-e

### Power Supply used.

External 3.3V

### What is the expected behavior?

Dhcp got ip.

### What is the actual behavior?

With spiram enable, calling esp_now_send() before calling esp_wifi_connect() will cause dhcp to time out on getting the ip.

### Steps to reproduce.

This is my code:
```
static const char *TAG = "app_main";
static void ip_event_handler(void *arg, esp_event_base_t event_base,
                             int32_t event_id, void *event_data)
{
    ESP_LOGE(TAG, "ip_event_handler %d", event_id);
    return;
}
static void wifi_event_handler(void *arg, esp_event_base_t event_base,
                               int32_t event_id, void *event_data)
{
    ESP_LOGE(TAG, "wifi_event_handler %d", event_id);
    if(event_id == WIFI_EVENT_STA_START || event_id == WIFI_EVENT_STA_DISCONNECTED){
        esp_wifi_connect();
    }
    return;
}

static void wifi_init()
{
    ESP_ERROR_CHECK(esp_netif_init());
    esp_event_loop_create_default();
    esp_netif_create_default_wifi_sta();
    esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL);
    esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &ip_event_handler, NULL);
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
    ESP_ERROR_CHECK(esp_wifi_start());

    wifi_config_t config = {0};
    strncpy((char *)config.sta.ssid, "ChinaNet-2.4G-3648", strlen("ChinaNet-2.4G-3648") + 1);
[... 374 lines cut ...]
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
transaction id xid(5a53c937)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 15000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
```
```

### More Information.

[https://github.com/espressif/esp-now/issues/44#issue-1355176080](url)

sxsx2yzyz, 2022-08-31:
[https://github.com/espressif/esp-now/issues/44#issue-1355176080](url)

zhangyanjiaoesp, 2022-09-01:
@sxsx2yzyz Are you using this project https://github.com/espressif/esp-now ?

sxsx2yzyz, 2022-09-01:
> https://github.com/espressif/esp-now

yes

zhangyanjiaoesp, 2022-09-29:
@sxsx2yzyz We have found the root cause, and will merge the bug fix ASAP. The fix will be backported to v4.4 also.

zhangyanjiaoesp, 2022-10-12:
@sxsx2yzyz This fix has been merged into master branch 3b7d9faf78053ffc9ef0a6f518caac4050c18610, backport for v4.4 still need time.

sxsx2yzyz, 2022-10-18:
@zhangyanjiaoesp hi,How do I use him in my project?
```
bash-3.2$ idf.py --version
ESP-IDF v4.4.1-dirty
```

zhangyanjiaoesp, 2022-10-19:
The fix has been merged into [v4.4](488c79a0036aa7c8424cd1e9d6c9052d8929f405). @LJYSP Please help to sync to [esp-now](https://github.com/espressif/esp-now) project.

closed 2022-10-20 by sxsx2yzyz