espressif/esp-idf #9686: Ability to modify HSM configuration for esp-tls

hamedmo, 2022-08-31:
### Is your feature request related to a problem?

We are using a custom ATECC608B chip, in which the I2C address and slot numbers are not the same as what is already provided in esp-tls library. In order to make the HSM chip work, we need to modify the SDK.

### Describe the solution you'd like.

HSM chip configurations (I2C pin numbers and address, different slot numbers, ability to validate slots before use, etc.) can be added to sdkconfig, so that the user can modify all the details. It would also be nice if the tls library accepts pointer to functions when slot validation or authentication is needed.

AdityaHPatwardhan, 2022-09-01:
@hamedmo  Thanks for the issue.

Can you elaborate more on slot validation ? What do you mean by slot validation ? Does it mean authenticating the ATECC608 module ?
I agree that as of now, the key slot has been fixed to a single value. We can add that flexibility.

hamedmo, 2022-09-01:
> @hamedmo Thanks for the issue.
> 
> Can you elaborate more on slot validation ? What do you mean by slot validation ? Does it mean authenticating the ATECC608 module ? I agree that as of now, the key slot has been fixed to a single value. We can add that flexibility.

In our custom configuration, the client private key at Slot 0 needs authentication before it can be used. Current TLS library does not support this mode, so the authorization needs to be done manually before calling the TLS functions. It might be added as a pointer in the configuration structure if the private key needs authentication before use or as a weak function in the library.
In addition, we have 3 slots for ECDH and they are supposed to be used alternatively to increase wear level. Is there any solution for such situations?

AdityaHPatwardhan, 2022-09-12:
Hi @hamedmo sorry for the delayed reply.
We had a discussion internally, I think this authentication should not be done through the esp-tls layer. Considering that this would lead to a policy change for the esp-tls component that authentication of the key slot of ATECC element.
I think it would be better if the application handles this.
As for the configurable key slot, I will update that change

hamedmo, 2022-10-08:
Hello @AdityaHPatwardhan,
The problem with slot authorization is that if the private key slot needs authorization to be used, it should be done before every action on the slot. That is, before generating a public key, and also before using the key to sign. As a result, a one-time authorization by the user before calling the connect function is not enough to use the key. I had to modify esp_tls_mbedtls.c file and add authorization process before the call to atca_mbedtls_pk_init and mbedtls_ssl_conf_own_cert to make it work.

hamedmo, 2022-10-08:
Hello @AdityaHPatwardhan,
The problem with slot authorization is that if the private key slot needs authorization to be used, it should be done before every action on the slot. That is, before generating a public key, and also before using the key to sign. As a result, a one-time authorization by the user before calling the connect function is not enough to use the key. I had to modify esp_tls_mbedtls.c file and add authorization process to make it work.

AdityaHPatwardhan, 2022-10-12:
@hamedmo I see!, Thanks for explaining the situation. IIUC, In that case multiple attempts for a TLS connections using the same key should require separate authorizations. If possible can you share the patch that you have applied which works in your case, also if you have a demo code where I could test authorisation then it would indeed be helpful for me to test it on my side.
Thanks,
Aditya

hamedmo, 2022-10-14:
Dear @AdityaHPatwardhan,
If the private key slot is configured as require authorization (_ReqAuth_), the chip should go through authorization process (which requires an authorization key) in order for the user to use the private key. So basically before every call to `atcab_get_pubkey` and `atcab_sign`, the authorization process need to be done. In order to manipulate just a single source file, I added the calls to authorization function inside esp_tls_mbedtls.c right before `ret = atca_mbedtls_pk_init(&tls->clientkey, 0);` line and then after that to make sure the key is ready when sign function is used later on. I'm not sure if I can share the authorization code here since it might be proprietary.

AdityaHPatwardhan, 2022-10-14:
@hamedmo I will figure it out and share a patch for you to check out.
Thanks,
Aditya

hamedmo, 2022-10-14:
@AdityaHPatwardhan thank you for your support. As a suggestion, the authorization function might be a weak definition that the user can override if needed.
Thanks,
Hamed

hamedmo, 2022-12-05:
Hi @Harshal5,

Could you please let me know if there is any progress with this issue?

mahavirj, 2022-12-05:
@hamedmo 

> So basically before every call to `atcab_get_pubkey` and `atcab_sign`, the authorization process needs to be done.

I think we can use `--wrap` LDFLAG to achieve this. We have an [example](https://github.com/espressif/esp-idf/tree/master/examples/build_system/wrappers) in IDF for this. So, basically the flow would be:

- Wrap `atcab_get_pubkey` and `atcab_sign`
-  Implement wrapper functions with

```
__wrap_atcab_sign()
{ 
     custom_auth_func();
     return  __real_atcab_sign();
}
```

This way we can keep change outside of the cryptoauthlib and specific to the application. Please try once.

hamedmo, 2022-12-05:
Hello @mahavirj 

Thank you for the quick response. I added a `wrapper.c` file inside the `main` folder of my project with the following code:

```
extern ATCA_STATUS __real_atcab_get_pubkey(uint16_t key_id, uint8_t* public_key);

ATCA_STATUS __wrap_atcab_get_pubkey(uint16_t key_id, uint8_t* public_key)
{
    printf("Authorizing...\n");
    ATCA_STATUS result = authorizeDevice();
    printf("Done with result %d\n", result);

    return __real_atcab_get_pubkey(key_id, public_key);
}
```

This is the CMakeLists.txt in the same folder:

```
set(COMPONENT_SRCS
	"app_main.c"
	"mqtt_demo_mutual_auth.c"
	"wrapper.c"
	)

set(COMPONENT_ADD_INCLUDEDIRS
	"."
	"${CMAKE_CURRENT_LIST_DIR}"
	)

idf_component_register(SRCS "${COMPONENT_SRCS}"
					INCLUDE_DIRS ${COMPONENT_ADD_INCLUDEDIRS}
				       )

target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--wrap=atcab_get_pubkey")
```

Yet still the main atcab_get_pubkey gets called instead of the wrapper function. Am I doing something wrong?

mahavirj, 2022-12-05:
@hamedmo 

Since you have kept the wrappers in a separate source file (compilation unit), and there are no direct callers for any of its symbols, probably linker is excluding entire object file.

Please try adding following to your CMakeLists.txt (at the end)

```
target_link_libraries(${COMPONENT_LIB} INTERFACE "-u __wrap_atcab_get_pubkey")
```

This is to direct linker to force this symbol to be entered in the output file. Probably it should fix the issue, if not then please look through your project map file once, it should give some clue.

hamedmo, 2022-12-05:
@mahavirj 

I used the flag and when I check the map file, both the original function and the one with __wrap_ prefix are produced, however the original function is called again.

hamedmo, 2022-12-06:
@mahavirj 

I also checked calling the function from my `main()`. Interestingly, the call from `main()` runs my wrapped function, but then again other calls from the mbedTLS wrappers of `esp-cryptoauthlib` library still goes to the original function.

mahavirj, 2022-12-06:
@hamedmo 

I tried building the example `examples/peripherals/secure_element/atecc608_ecdsa` from IDF v5.0 locally and found that in default configuration `calib_get_pubkey` symbols gets linked and not the `atcab_get_pubkey`. Maybe you are using different configuration here. I have pushed a branch based on this [here](https://github.com/mahavirj/esp-idf/commits/feature/secure_element_wrap_pubkey), please try it out. Example built out of this feature branch had correct usage of `--wrap` at all places.

hamedmo, 2022-12-08:
Hi @mahavirj 

Thank you so much for your support. I guess the problem with my first attempts was that one of the header file included in my wrapper actually had the original function declaration in it. Including only the required headers for struct and enum definitions solved the problem. I also had to change the wrapped `calib_get_pubkey(uint16_t key_id, uint8_t* public_key)` function from your sample code to `calib_get_pubkey(ATCADevice device, uint16_t key_id, uint8_t *public_key)` version that also gets the pointer to a device. Without passing the pointer, the `device` references inside the first function cause core panic.

closed 2022-12-08 by hamedmo