espressif/esp-idf #9688: Custom TLS common name in esp_http_client

RostakaGmfun, 2022-08-31:
### Is your feature request related to a problem?

Sometimes it could be useful to check against a common name that differs from the hostname while establishing a TLS session via `esp_http_client`. The `esp_tls` layer already supports having a different hostname and common name, but the `esp_transport` and `esp_http_client` layers can only use the hostname as a common name or skip the common name check completely.   

### Describe the solution you'd like.

It would be great to have a `tls_common_name` config option, along with the `skip_cert_common_name_check`, that allows passing a specific common name to check against during TLS handshake. The `esp_http_client` would pass this option down to the `esp_transport`, which in turn would pass it down to the `esp_tls`, just like it is done with the `skip_common_name` configuration option.

### Describe alternatives you've considered.

While it is theoretically possible to hack into private structures and modify the SSL config structure, this is certainly not the solution one would look for.

AdityaHPatwardhan, 2022-09-06:
Hi @RostakaGmfun Can you please test this patch if it solves your issue ?
Thanks
[cn_patch.zip](https://github.com/espressif/esp-idf/files/9493113/cn_patch.zip)

RostakaGmfun, 2022-09-06:
Hi @AdityaHPatwardhan, thanks for your reply. Unfortunately I can't unzip the archive you attached. `unzip` tells me this:
`End-of-central-directory signature not found.`

AdityaHPatwardhan, 2022-09-06:
Please try this one set_tls_common_name.patch.zip, There might have been some error while compressing the file earlier.

AdityaHPatwardhan, 2022-09-06:
set_tls_common_name.patch.zip

AdityaHPatwardhan, 2022-09-06:
[set_tls_common_name.patch.zip](https://github.com/espressif/esp-idf/files/9497272/set_tls_common_name.patch.zip)

@RostakaGmfun  Please try this one.
Thanks

RostakaGmfun, 2022-09-06:
Thanks @AdityaHPatwardhan the patch looks good to me, I could give it a try if I knew the git commit I should apply it onto. Do you want me to test it on master? Also, what IDF release can I expect the patch to be included in?

AdityaHPatwardhan, 2022-09-12:
Hi @RostakaGmfun Sorry for the delayed reply, You can apply it on the master branch and check if it fixes your issue. Since it is a feature this will only be available on master branch

AdityaHPatwardhan, 2022-09-19:
This feature request should now have been completed by https://github.com/espressif/esp-idf/commit/4904d57fd98a787e41c268cfa19f590a0607b3c9