espressif/esp-idf #9690: Trigger NVS "cleanup" to forcefully remove erased items

WolfspiritM, 2022-09-01:
### Is your feature request related to a problem?

I'm currently working on a project which requires the ESPs to be given away after initial setup. For initial setup some internal things are stored (like Wifi Passwords and so on) which are removed via `nvs_erase_key` after initial setup.

The problem now is that the person who receives the ESP can just export the NVS content. Everyone who is able to read the NVS flash content is able to extract that wifi passwords cause the NVS Entry is just marked as "erased" but the content is not really removed. Overwriting will not work aswell as according to docs overriding just means marking the old value as "erased" and creating a new value.

In my opinion this can be a security issue.

### Describe the solution you'd like.

As freeing is probably resource intensive task I suggested to provide a "cleanup" API function that forcefully triggers the "Freeing" method for all pages similar to here:
https://github.com/espressif/esp-idf/blob/73ec743a0a7d48685a8614de70ecf2bcb3286715/components/nvs_flash/src/nvs_pagemanager.cpp#L160-L172

This could then be called like `nvs_cleanup();`

negativekelvin, 2022-09-01:
Possible alternative is to implement nvs_secure_erase_key which does not just mark the entry as erased but also sets all bits to zero.  That way you don't have to copy/erase potentially a whole page of data just to securely remove one key.

0xjakob, 2022-09-02:
@WolfspiritM Is the WiFi access data stored automatically (via the WiFi library) or are you manually storing them?

WolfspiritM, 2022-09-02:
@negativekelvin That would work, too. It however requires access to the code that removes the key. In my case that would be fine but maybe a lib removes a "secret" from NVS and someone wants to completly get rid of it aswell.

@0xjakob I'm storing it manually

0xjakob, 2022-09-02:
@WolfspiritM Why is that necessary? I'm asking because in all device provisioning scenarios I recall the device wouldn't need to save any temporary secrets to non-volatile memory.

Could you erase the entire NVS partition or are some values stored there for the time after the provisioning process?

Another alternative might be creating a custom NVS partition with just 3 pages only for the WiFi credentials, which can be erased before delivery.

WolfspiritM, 2022-09-02:
@0xjakob In our case the provisioning requires multiple reboots of the ESP as it happens in multiple steps. The initial setup wifi password is flashed with the NVS partition, the main configuration is then pulled from a config server (in multiple steps) and stored in NVS and the last step then removes the Wifi SSID and Password from NVS. A special NVS partition might be a way, yes. I'm also thinking about pulling all configuration into RAM in the last step, erasing the NVS partition completly and then storing everything again.

I still think that there should be a "secure erase" feature or cleanup feature in the future and probably it should be made more obvious that keys are not actually removed from NVS until a cleanup happens. Depending on the use case it can be a security issue in my opinion.

0xjakob, 2022-09-02:
@WolfspiritM Thanks for the information.

We'll discuss this issue internally on Monday and come back to you afterwards.

0xjakob, 2022-09-02:
Another method to retain memory would be RTC memory which doesn't change during reboots, IIRC.

pacucha42, 2022-09-06:
Hi @WolfspiritM. Regarding your feature request: we have started the discussion as suggested by @0xjakob but I'm afraid it will take a while. Your request is proper and there are several ways to implement it, which must be done carefully especially in the NVS case to avoid unwanted side-effects (which might also result in denying the request completely) . We will keep you posted. Thank you for your effort.

closed 2022-09-19 by pacucha42