espressif/esp-idf #9696: ESP32 + 16GB SD vSPI card mount issue.

fifyf, 2022-09-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Module : esp wroom 32 development board.
executing sd spi from example code.
Connected to vSPI (5/18/19/23)

Reports following error, while mounting the card.

D (1797) spi_master: device5 release bus
V (1807) bus_lock: dev 5 released.
V (1807) sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0x00000000
V (1817) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x0 state=0
D (1827) sdmmc_cmd: sdmmc_send_cmd_send_if_cond: received=0x0 expected=0xaa
E (1827) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
D (1837) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_if_cond returned 0x108
E (1847) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (1847) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1857) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.

Enabled verbosity and the following is the logs since boot to failing to mount SD card.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7012
load:0x40078000,len:15384
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080648
I (27) boot: ESP-IDF v5.1-dev-437-g5c1044d84d 2nd stage bootloader
I (27) boot: compile time 16:54:23
I (27) boot: chip revision: V100
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0e59ch ( 58780) map
I (132) esp_image: segment 1: paddr=0001e5c4 vaddr=3ffb0000 size=01a54h (  6740) load
I (135) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=25f34h (155444) map
I (194) esp_image: segment 3: paddr=00045f5c vaddr=3ffb1a54 size=01144h (  4420) load
I (196) esp_image: segment 4: paddr=000470a8 vaddr=40080000 size=0d5b4h ( 54708) load
I (222) esp_image: segment 5: paddr=00054664 vaddr=50000000 size=00010h (    16) load
I (229) boot: Loaded app from partition at offset 0x10000
I (229) boot: Disabling RNG early entropy source...
I (242) cpu_start: Pro cpu up.
I (242) cpu_start: Starting app cpu, entry point is 0x400811d4
0x400811d4: call_start_cpu1 at /home/lenovo/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
D (250) clk: RTC_SLOW_CLK calibration value: 3163187
I (261) cpu_start: Pro cpu start user code
I (261) cpu_start: cpu freq: 160000000 Hz
I (261) cpu_start: Application information:
I (266) cpu_start: Project name:     uart_echo
[... 348 lines cut ...]
V (1747) spi_master: polling trans
V (1747) spi_master: polling trans done
V (1757) spi_master: polling trans
V (1757) spi_master: polling trans done
V (1757) spi_master: polling trans
V (1767) spi_master: polling trans done
V (1767) spi_master: polling trans
V (1767) spi_master: polling trans done
V (1777) spi_master: polling trans
V (1777) spi_master: polling trans done
D (1787) sdspi_host: poll_busy: timeout
V (1787) spi_master: polling trans
V (1787) spi_master: polling trans done
V (1797) spi_master: polling trans
V (1797) spi_master: polling trans done
D (1797) spi_master: device5 release bus
V (1807) bus_lock: dev 5 released.
V (1807) sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0x00000000
V (1817) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x0 state=0
D (1827) sdmmc_cmd: sdmmc_send_cmd_send_if_cond: received=0x0 expected=0xaa
E (1827) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
D (1837) sdmmc_init: sdmmc_card_init: sdmmc_init_sd_if_cond returned 0x108
E (1847) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (1847) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1857) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.

igrr, 2022-09-01:
Hi @fifyf,
Based on the logs (`cmd response`), the card's output line (MISO) is always low. I suggest checking the connections, it looks like that line might be shorted to ground. Normally even if the card is not connected, MISO line would be pulled up, and the ESP would receive all bits equal to 1.

fifyf, 2022-09-02:
Once I changed to a newer esp, things started to work. Let me debug the older dev board to see, if there is any issues in it. Thank you @igrr , your comment helped me suspect the dev kit, and the sd module.

closed 2022-09-02 by fifyf