espressif/esp-idf #9697: assert failed: twai_handle_tx_buffer_frame twai.c:183 (p_twai_obj->tx_msg_count >= 0)

diplfranzhoepfinger, 2022-09-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### Development Kit.

Atom M5

### Power Supply used.

USB

### What is the expected behavior?

Not Chrashing

### What is the actual behavior?

Instead it crash 
assert failed: twai_handle_tx_buffer_frame twai.c:183 (p_twai_obj->tx_msg_count >= 0)

### Steps to reproduce.

1. Clone Repo: (coming)
2. let it run a while.
3. disturb CAN-Bus by eigher shorting CAN-H to GND or CAN-H to CAN-L
4. it will crash instead of a recovery. 

### Debug Logs.

```plain
I (100100) TWAI Master: Surpassed Error Warning Limit
I (100100) TWAI Master: Entered Error Passive state
I (100100) TWAI Master: Bus Off state
W (100100) TWAI Master: Initiate bus recovery in 50ms
E (100110) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
W (100120) TWAI Master: Initiate bus recovery in 40ms
W (100130) TWAI Master: Initiate bus recovery in 30ms
W (100140) TWAI Master: Initiate bus recovery in 20ms
W (100150) TWAI Master: Initiate bus recovery in 10ms
E (100210) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (100310) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (100410) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (100510) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (100610) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (100710) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (100810) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (100910) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101010) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101110) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101210) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101310) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101410) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101510) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101610) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101710) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101810) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (101910) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102010) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102110) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102210) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102310) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102410) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102510) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102610) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102710) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102810) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (102910) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (103010) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
E (103110) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
I (103150) TWAI Master: Initiate bus recovery

assert failed: twai_handle_tx_buffer_frame twai.c:183 (p_twai_obj->tx_msg_count >= 0)

Backtrace:0x400818de:0x3ffb0a000x40085841:0x3ffb0a20 0x4008ae2d:0x3ffb0a40 0x4008313d:0x3ffb0b60 0x40082869:0x3ffb0ba0 0x400e8257:0x3ffb62f0 0x400d1c6f:0x3ffb6310 0x40086ea2:0x3ffb6330 0x400883d1:0x3ffb6350 
0x400818de: panic_abort at /home/franz/esp-idf-v4.4.2/components/esp_system/panic.c:402

0x40085841: esp_system_abort at /home/franz/esp-idf-v4.4.2/components/esp_system/esp_system.c:128

0x4008ae2d: __assert_func at /home/franz/esp-idf-v4.4.2/components/newlib/assert.c:85

0x4008313d: twai_handle_tx_buffer_frame at /home/franz/esp-idf-v4.4.2/components/driver/twai.c:183
 (inlined by) twai_intr_handler_main at /home/franz/esp-idf-v4.4.2/components/driver/twai.c:226

0x40082869: _xt_lowint1 at /home/franz/esp-idf-v4.4.2/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x400e8257: cpu_ll_waiti at /home/franz/esp-idf-v4.4.2/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /home/franz/esp-idf-v4.4.2/components/esp_pm/pm_impl.c:837

0x400d1c6f: esp_vApplicationIdleHook at /home/franz/esp-idf-v4.4.2/components/esp_system/freertos_hooks.c:63

0x40086ea2: prvIdleTask at /home/franz/esp-idf-v4.4.2/components/freertos/tasks.c:3973 (discriminator 1)

0x400883d1: vPortTaskWrapper at /home/franz/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131

ELF file SHA256: a42730679ffbb41a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
```

diplfranzhoepfinger, 2022-09-01:
if

```
#
# TWAI configuration
#
# CONFIG_TWAI_ISR_IN_IRAM is not set
# CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC is not set
# CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST is not set
# CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID is not set
# CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT is not set
# end of TWAI configuration
```

then it does NOT crash.

diplfranzhoepfinger, 2022-09-01:
It seems

```
CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=y
```

is the Reason for the Crashing.

EmbeddedDevver, 2022-10-04:
Stumbled across the same problem. This did not happen in prior (CAN not TWAI) libs. I do not think there is a solution when using the Arduino environment right?

abombay, 2023-01-16:
I am having the same issue. In my case I powered up my device without any power on the other side of the CAN transceiver (ISO1042 Isolated). Is there any update on this issue?

f-hoepfinger-hr-agrartechnik, 2023-01-16:
@Dazza0 any news ?

Dazza0, 2023-02-27:
> disturb CAN-Bus by eigher shorting CAN-H to GND or CAN-H to CAN-L

@diplfranzhoepfinger @abombay @f-hoepfinger-hr-agrartechnik  By disturbing the CAN/TWAI Bus, you are likely generating errors that trigger the HW errata conditions. The errata fixes have been fixed on master and backported all the way back to ESP-IDF v4.2.x.  However, these errata workarounds are not enabled by default until ESP-IDF v5.0 onwards. If you are using an ESP-IDF version older than v5.0, please enable all of the `CONFIG_TWAI_ERRATA_FIX_...` options and see if the issue is resolved.

@EmbeddedDevver Arduino should also have these [workarounds enabled](https://github.com/espressif/arduino-esp32/blob/2.0.5/tools/sdk/esp32/sdkconfig) starting from v2.0.5

StehlikPhotoneo, 2023-06-06:
I have the same problem. My fix is to not use the twai_initiate_recovery() function, but instead of this uninstall, install and start the TWAI driver back. This fix might help you until the problem is fixed in the TWAI library.

Dazza0, 2023-06-06:
@StehlikPhotoneo Please ensure that the HW errata workarounds are enabled in your build (by enabling the various `CONFIG_TWAI_ERRATA_FIX_...` configurations. If you are using ESP-IDF, simply enable them via `menuconfig`. If you are using Arduino, this may require you upgrade to a newer version.

Dazza0, 2023-06-14:
@diplfranzhoepfinger @StehlikPhotoneo @abombay @f-hoepfinger-hr-agrartechnik 

I suspect what's happening is the `CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST` workaround is triggering a false positive TX done event when a bus off occurs.

In [`twai_hal_iram.c:twai_hal_decode_interrupt()`](https://github.com/espressif/esp-idf/blob/5181de8ac5ec5e18f04f634da8ce173b7ef5ab73/components/hal/twai_hal_iram.c#L32), if you change...

```c
#ifdef CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST
    if ((interrupts & TWAI_LL_INTR_TI || hal_ctx->state_flags & TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED) && status & TWAI_LL_STATUS_TBS) {
#else
```
to 

```c
#ifdef CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST
    if ((interrupts & TWAI_LL_INTR_TI || hal_ctx->state_flags & TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED)
        && status & TWAI_LL_STATUS_TBS
        && !(state_flags & TWAI_HAL_STATE_FLAG_BUS_OFF)) {
#else
```
does this end up resolving the issue?

franz-ms-muc, 2023-06-14:
i will check !

travis012, 2023-06-15:
I am also having this issue.  All of my TWAI_ERRATA fixes are enabled.  I'm using ESPIDF 4.4.4 with a ESP32MINI-01 rev3 device.

I will try [Dazza0](https://github.com/Dazza0) code patch right now and let you know.

travis012, 2023-06-15:
Unfortunately this doesn't fix it.  I should note that I am using single shot mode with no tx queue.  I handle retries myself.  This is the only function I use to send out CAN messages.

My transmit function is as follows:
```C
IRAM_ATTR esp_err_t tts_twai_send_message(twai_message_t* msg, uint32_t paceTime100us, uint32_t timeoutMS)
{
#define MAX_TX_RETRIES (16)
	uint32_t alerts;
	uint32_t paceTimeTicks;
	esp_err_t err;
	int retryCount = MAX_TX_RETRIES;
	
	xSemaphoreTake(_semTxLock, portMAX_DELAY);
	
	msg->flags = TWAI_MSG_FLAG_SS;
	
	TickType_t tickTimeout = xTaskGetTickCount() + pdMS_TO_TICKS(timeoutMS);
	if (twai_read_alerts(&alerts, 0) == ESP_OK)
	{
#if TWAI_REPORT_AND_HANDLE_ALERTS
		tts_twai_report_alerts(alerts);
#endif			
	}
	
RetryTx:
	err = twai_transmit(msg, pdMS_TO_TICKS(100));
	if (err != ESP_OK)
	{
		ESP_LOGE(EXAMPLE_TAG, "twai_transmit failed! Error=%08X", err);
		alerts = 0;
		twai_read_alerts(&alerts, pdMS_TO_TICKS(100));
#if TWAI_REPORT_AND_HANDLE_ALERTS
		tts_twai_report_alerts(alerts);
#endif		
		if (err == ESP_ERR_INVALID_STATE)
		{
			if (twai_initiate_recovery() == ESP_OK)
			{
				ESP_LOGW(EXAMPLE_TAG, "Bus recovery initiated..");
				// wait for bus to recover
				while (xTaskGetTickCount() <= tickTimeout)
				{
					if (twai_read_alerts(&alerts, pdMS_TO_TICKS(100)) == ESP_OK)
					{
#if TWAI_REPORT_AND_HANDLE_ALERTS
						tts_twai_report_alerts(alerts);
#endif							
						if (alerts & TWAI_ALERT_BUS_RECOVERED)
						{
							if (--retryCount <= 0) 
							{
								ESP_LOGE(EXAMPLE_TAG, "Max TX retry count exceeded!");
								goto Done; 
							}
							
							err = twai_start();
							ESP_LOGW(EXAMPLE_TAG, "Bus Recovered. twai_start err=%i", err);
							vTaskDelay(1);
							goto RetryTx;
						}						
					}
				}
				ESP_LOGE(EXAMPLE_TAG, "Timeout waiting for bus recovery");
			}
			else
			{
				ESP_LOGW(EXAMPLE_TAG, "TX failed for unknown reason");
				if (--retryCount <= 0) 
				{
					ESP_LOGE(EXAMPLE_TAG, "Max TX retry count exceeded!");
					goto Done; 
				}
				vTaskDelay(1);
			}
		}
		else
		{
			ESP_LOGE(EXAMPLE_TAG, "TX failed Error=%08X", err);
			if (--retryCount <= 0) 
			{
				ESP_LOGE(EXAMPLE_TAG, "Max TX retry count exceeded!");
				goto Done; 
			}
			vTaskDelay(1);
			goto RetryTx;
			
		}
		// this is caused by the loss of a TX interrupt. (see errata)
		// it should not occur anymore as there is a work-a-round in place deep in the TWAI drvier stack.
		//ESP_LOGE(EXAMPLE_TAG, "tts_twai_send_message1 Error=%08X", err);
		//twai_stop();
		//vTaskDelay(pdMS_TO_TICKS(20));
		//twai_start();
		//vTaskDelay(pdMS_TO_TICKS(20));
		//err = twai_transmit(msg, pdMS_TO_TICKS(100));
		
	}
	else
	{
		// TX was successful. we should receive either an idle alert or some error promptly
		
		// clear any alerts
		if (twai_read_alerts(&alerts, pdMS_TO_TICKS(1000)) == ESP_OK)
		{
#if TWAI_REPORT_AND_HANDLE_ALERTS
			tts_twai_report_alerts(alerts);
#endif	
			if (alerts & TWAI_ALERT_TX_FAILED)
			{
				if (--retryCount <= 0) 
				{
					ESP_LOGE(EXAMPLE_TAG, "Max TX retry count exceeded!");
					err = ESP_ERR_TIMEOUT;
					goto Done; 
				}
				vTaskDelay(1);
				goto RetryTx;
			}
		}
		
		// honor pacetime
		paceTimeTicks = pdMS_TO_TICKS(((TickType_t)paceTime100us + (TickType_t)9) / (TickType_t)10);
		if (paceTimeTicks > 0)
		{
			vTaskDelay(paceTimeTicks);
		}
		else
		{
			uint64_t timeEnd = esp_timer_get_time() + (paceTime100us * 100);
			while (timeEnd > esp_timer_get_time())
			{
				// taskYIELD() is used to request a context switch to another task. However, if there are 
				// no other tasks at a higher or equal priority to the task that calls taskYIELD() then 
				// the RTOS scheduler will simply select the task that called taskYIELD() to run again. 
				taskYIELD();
			}

		}
	}
Done:
	xSemaphoreGive(_semTxLock);
	if (retryCount != MAX_TX_RETRIES)
	{
		//ESP_LOGI(EXAMPLE_TAG, "TX took %d retries", MAX_TX_RETRIES - retryCount);
	}
	return err;
}
```

travis012, 2023-06-16:
If I use this:
```C
#ifdef CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST
	// 
	if ((interrupts & TWAI_LL_INTR_TI || hal_ctx->state_flags & TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED) && status & TWAI_LL_STATUS_TBS && p_twai_obj->tx_msg_count) {
#else
```
it solves the problem.  This is not a good fix but it works in my case because I'm not using any kind of TX queuing.  It does provide more proof that the problem is with this errata "fix".  

FYI, You have to manipulate some code to gain access to ```p_twai_obj``` in-order to use this.

Hopefully this will help someone else find a proper fix. :)

andrew-elder, 2023-12-14:
I'm running into this after 40 minutes of runtime (more than 120,000 CAN messages sent).
I have `CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST` set.
I'm using SDK 5.0 and I don't see any fixes for this posted in later implementations. Have I missed a patch?

andrew-elder, 2023-12-15:
@igrr - any comments?

wanckl, 2023-12-18:
@andrew-elder Have you tried code changes from Dazza0 above https://github.com/espressif/esp-idf/issues/9697#issuecomment-1591011346.    
And do you have some custom setting like `no_queue` ? and your chip version ?

andrew-elder, 2023-12-18:
@wanckl I have not yet tired the fix from https://github.com/espressif/esp-idf/issues/9697#issuecomment-1591011346. I will try it.

I am using default setting. ie not using `no_queue`. Chip version is v3.0
```
I (31) boot: chip revision: v3.0
```

andrew-elder, 2023-12-18:
@wanckl - the test failed after an hour with
```
assert failed: twai_handle_tx_buffer_frame twai.c:185 (p_twai_obj->tx_msg_count >= 0)
```
The suggested change checked bus off, but that isn't expected to happened in my setup. The CAN bus is hardwired to another device that is continuously sending messages to the ESP32.

Thank you for your help so far. Any other suggestions?

travis012, 2023-12-18:
> @wanckl - the test failed after an hour with
>
> ```
> assert failed: twai_handle_tx_buffer_frame twai.c:185 (p_twai_obj->tx_msg_count >= 0)
> ```
>
> The suggested change checked bus off, but that isn't [...]

Have you tried the fix I posted above?  To use it, you have to disable the TX queue.  It is not a good fix but it does solve the problem.  I have been using it in a production product for a while now.

andrew-elder, 2023-12-18:
@travis012 - I have not tried the fix you posted above. I will do so.

andrew-elder, 2023-12-19:
@travis012 - an implementation very close to yours works for me. ie, I no longer observe the assert() error. I don't observe that `twai_initiate_recovery()` ever gets called either. I think I am ok to continue development with what I have. Thank you for your help.

@wanckl - does espressif have plans to release a fix for this?

travis012, 2023-12-19:
> @travis012 - an implementation very close to yours works for me. ie, I no longer observe the assert() error. I don't observe that `twai_initiate_recovery()` ever gets called either. I think I am ok to continue development with what I have. Thank you for your help.

Is it better than mine?  Can you use the TX queue? I'd like to improve what I have in our production devices.  Please share your fix.

andrew-elder, 2023-12-19:
No, it's not better than yours. I removed the sem for example because I have a single thread sending messages. I have a hardwired CAN connection to another device, so there is never a BUS-OFF condition, so twai_initiate_recovery() always returns ESP_ERR_INVALID_STATE if I force it to trigger.

italocjs, 2024-07-31:
Any update on this? having the same issue

diplfranzhoepfinger, 2024-08-01:
i have to check.

italocjs, 2024-09-20:
Hi @willianaraujo , here's an example of how i did it.   Using twai_initiate_recovery() causes crash.
I have tested the code below with short circuit only, and has worked well.

```c
twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(CAN_TX_PIN, CAN_RX_PIN, TWAI_MODE_NORMAL);
twai_timing_config_t t_config;

port_err_t start_driver(int can_speed)
{
	switch (can_speed)
	{
		case 25000:
			t_config = TWAI_TIMING_CONFIG_25KBITS();
			break;
		case 50000:
			t_config = TWAI_TIMING_CONFIG_50KBITS();
			break;
		case 100000:
			t_config = TWAI_TIMING_CONFIG_100KBITS();
			break;
		case 125000:
			t_config = TWAI_TIMING_CONFIG_125KBITS();
			break;
		case 250000:
			t_config = TWAI_TIMING_CONFIG_250KBITS();
			break;
		case 500000:
			t_config = TWAI_TIMING_CONFIG_500KBITS();
			break;
		case 800000:
			t_config = TWAI_TIMING_CONFIG_800KBITS();
			break;
		case 1000000:
			t_config = TWAI_TIMING_CONFIG_1MBITS();
			break;
		default:
			t_config = TWAI_TIMING_CONFIG_500KBITS();
			ESP_LOGE(tagCanDriver, "Default speed is NOT STANDARD!!!, using fallback to 500KBITS");
			break;
	}

	twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();

	esp_err_t install_result = twai_driver_install(&g_config, &t_config, &f_config);
	if (install_result != ESP_OK)
	{
		ESP_LOGE(tagCanDriver, "twai_driver_install failed: %s", esp_err_to_name(install_result));
		return PORT_FAIL;
	}
	if (twai_start() != ESP_OK)
	{
		ESP_LOGE(tagCanDriver, "twai_start failed");
		return PORT_FAIL;
	}

	ESP_LOGI(tagCanDriver, "Driver started with speed %d", can_speed);
	return PORT_OK;
}

port_err_t stop_driver()
{
	esp_err_t error = twai_stop();
	if (error != ESP_OK && error != ESP_ERR_INVALID_STATE)
	{
		ESP_LOGE(tagCanDriver, "Failed to stop can, error was %s", esp_err_to_name(error));
		return PORT_FAIL;
	}
	else if (error == ESP_ERR_INVALID_STATE)
	{
		ESP_LOGI(tagCanDriver, "Driver was not started");
	}

	error = twai_driver_uninstall();
	if (error != ESP_OK && error != ESP_ERR_INVALID_STATE)
	{
		ESP_LOGE(tagCanDriver, "Failed to uninstall can, error was %s", esp_err_to_name(error));
		return PORT_FAIL;
	}
	else if (error == ESP_ERR_INVALID_STATE)
	{
		ESP_LOGI(tagCanDriver, "Driver was not installed");
	}

	ESP_LOGI(tagCanDriver, "Driver Stopped");
	return PORT_OK;
}
bool handleBusStatus(twai_status_info_t &status_info)
{
	// Print the status info
	ESP_LOGI(tagCanDriver, "State: %d, RX Error Counter: %d, TX Error Counter: %d, RX Queue Length: %d, TX Queue Length: %d",
	         status_info.state, status_info.rx_error_counter, status_info.tx_error_counter, status_info.msgs_to_rx, status_info.msgs_to_tx);

	// Based on state, do something
	if (status_info.state == TWAI_STATE_RUNNING)
	{
		ESP_LOGI(tagCanDriver, "CAN bus is running");
		if (status_info.msgs_to_rx > 0)
		{
			ESP_LOGI(tagCanDriver, "Queued %d packages, ready to process...", status_info.msgs_to_rx);
		}
	}
	else if (status_info.state == TWAI_STATE_STOPPED)
	{
		stop_driver();
		delay(100);
		start_driver(api.sensors.can.serviceSpeed);
		ESP_LOGE(tagCanDriver, "Driver restarted");
	}
	else if (status_info.state == TWAI_STATE_BUS_OFF)
	{
		ESP_LOGE(tagCanDriver, "CAN bus is in bus-off state");
		stop_driver();
		delay(100);
		start_driver(api.sensors.can.serviceSpeed);
		ESP_LOGE(tagCanDriver, "Driver restarted");
	}
	else if (status_info.state == TWAI_STATE_RECOVERING)
	{
		ESP_LOGI(tagCanDriver, "CAN bus is recovering (does not really work)");
		stop_driver();
		delay(100);
		start_driver(api.sensors.can.serviceSpeed);
		ESP_LOGE(tagCanDriver, "Driver restarted");

		ESP_LOGE(tagCanDriver, "twai_stop() and twai_driver_uninstall() called");
	}
	else
	{
		ESP_LOGE(tagCanDriver, "Unknown CAN bus state");
		stop_driver();
		delay(100);
		start_driver(api.sensors.can.serviceSpeed);
		ESP_LOGE(tagCanDriver, "Driver restarted");
	}
	return true;
}

```

phatpaul, 2024-11-14:
oof! I thought enabling these errata workarounds would improve stability.
For now, I'm turning off CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST .  That stops the constant crashing when the CAN bus is unplugged from my product.
```
# CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST is not set
```

Could someone pls explain what that errata workaround tried to improve?  The help says:

    On the ESP32, when a transmit interrupt occurs, and interrupt register is read on the same APB clock cycle, the transmit interrupt could be lost. Enabling this option will add a workaround that checks the transmit buffer status bit to recover any lost transmit interrupt.

Just curious what could I be missing by not having that errata workaround?  Does it lead to a few missed Tx can frames?

JensOgorek, 2025-01-29:
Are there any updates on this? We ran into the same problem

diplfranzhoepfinger, 2025-02-04:
> [@diplfranzhoepfinger](https://github.com/diplfranzhoepfinger) [@StehlikPhotoneo](https://github.com/StehlikPhotoneo) [@abombay](https://github.com/abombay) [@f-hoepfinger-hr-agrartechnik](https://github.com/f-hoepfinger-hr-agrartechnik)
> 
> I suspect what's happening is the `CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST` workaround is triggering a false positive TX done event when a bus off occurs.
> 
> In [`twai_hal_iram.c:twai_hal_decode_interrupt()`](https://github.com/espressif/esp-idf/blob/5181de8ac5ec5e18f04f634da8ce173b7ef5ab73/components/hal/twai_hal_iram.c#L32), if you change...
> 
> #ifdef CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST
>     if ((interrupts & TWAI_LL_INTR_TI || hal_ctx->state_flags & TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED) && status & TWAI_LL_STATUS_TBS) {
> #else
> to
> 
> #ifdef CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST
>     if ((interrupts & TWAI_LL_INTR_TI || hal_ctx->state_flags & TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED)
>         && status & TWAI_LL_STATUS_TBS
>         && !(state_flags & TWAI_HAL_STATE_FLAG_BUS_OFF)) {
> #else
> does this end up resolving the issue?

i still have to test this. Sorry, @Dazza0 , forgot !

diplfranzhoepfinger, 2025-02-27:
with the fix of @Dazza0 

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6276
load:0x40078000,len:15716
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3860
entry 0x4008063c
I (31) boot: ESP-IDF v5.4 2nd stage bootloader
I (31) boot: compile time Feb 27 2025 14:06:20
I (31) boot: Multicore bootloader
I (32) boot: chip revision: v1.0
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a204h ( 41476) map
I (104) esp_image: segment 1: paddr=0001a22c vaddr=3ff80000 size=0001ch (    28) load
I (104) esp_image: segment 2: paddr=0001a250 vaddr=3ffb0000 size=023d4h (  9172) load
I (111) esp_image: segment 3: paddr=0001c62c vaddr=40080000 size=039ech ( 14828) load
I (121) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=148cch ( 84172) map
I (152) esp_image: segment 5: paddr=000348f4 vaddr=400839ec size=09c4ch ( 40012) load
I (174) boot: Loaded app from partition at offset 0x10000
I (174) boot: Disabling RNG early entropy source...
I (184) cpu_start: Multicore app
I (193) cpu_start: Pro cpu start user code
I (193) cpu_start: cpu freq: 160000000 Hz
I (193) app_init: Application information:
I (196) app_init: Project name:     canrecovery
I (201) app_init: App version:      v0.21c-6-g35ea235-dirty
I (207) app_init: Compile time:     Feb 27 2025 14:07:58
I (213) app_init: ELF file SHA256:  d6dd3d23eaa78883...
--- Warning: Checksum mismatch between flashed and built applications. Checksum of built application is c5d2cc1540809c8f4823c726e6f5d16690484d4e7bd605dbd904fcd5d07e6e00
I (219) app_init: ESP-IDF:          v5.4
I (224) efuse_init: Min chip rev:     v0.0
I (229) efuse_init: Max chip rev:     v3.99 
I (234) efuse_init: Chip rev:         v1.0
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (252) heap_init: At 3FFB2CA0 len 0002D360 (180 KiB): DRAM
I (258) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (264) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (271) heap_init: At 4008D638 len 000129C8 (74 KiB): IRAM
I (278) spi_flash: detected chip: gd
I (281) spi_flash: flash io: dio
W (285) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (299) main_task: Started on CPU0
I (309) main_task: Calling app_main()
I (309) main: Hello world!
I (309) TWAI Master: Driver installed
I (309) TWAI Master: Driver started
I (309) TWAI Master: Starting transmissions
I (4919) TWAI Master: Surpassed Error Warning Limit
I (4919) TWAI Master: Entered Error Passive state
I (4919) TWAI Master: Bus Off state
W (4919) TWAI Master: Initiate bus recovery in 50ms
E (4919) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
[... 40 lines cut ...]
W (5689) TWAI Master: Initiate bus recovery in 40ms
W (5699) TWAI Master: Initiate bus recovery in 30ms
W (5709) TWAI Master: Initiate bus recovery in 20ms
W (5719) TWAI Master: Initiate bus recovery in 10ms
I (5729) TWAI Master: Initiate bus recovery
E (5749) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
I (5829) TWAI Master: Bus Recovered
I (5829) TWAI Master: Driver started
I (5829) TWAI Master: Starting transmissions
I (5929) TWAI Master: Surpassed Error Warning Limit
I (5929) TWAI Master: Entered Error Passive state
I (5929) TWAI Master: Bus Off state
W (5929) TWAI Master: Initiate bus recovery in 50ms
W (5939) TWAI Master: Initiate bus recovery in 40ms
W (5949) TWAI Master: Initiate bus recovery in 30ms
E (5949) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
W (5959) TWAI Master: Initiate bus recovery in 20ms
W (5969) TWAI Master: Initiate bus recovery in 10ms
I (5979) TWAI Master: Initiate bus recovery
E (6049) main: twai_transmit failed, 259 ESP_ERR_INVALID_STATE
I (6079) TWAI Master: Bus Recovered
I (6079) TWAI Master: Driver started
I (6079) TWAI Master: Starting transmissions

```

diplfranzhoepfinger, 2025-02-27:
@Dazza0 Pull Request: 
https://github.com/espressif/esp-idf/pull/15484

Dazza0, 2025-03-15:
@diplfranzhoepfinger Sorry for the very late response (I don't work at Espressif anymore)

I've been able to recreate the issue. The root cause is due to the TX Done interrupt handler attempting to load a next frame for transmission even if a bus-off has occurred (see details below). So the fix for this is to update the TX done interrupt handler.

## Details

<details>
<summary>Sequence of events</summary>

1. Starting condition
    - Inside ISR, just arrived at `twai_handle_tx_buffer_frame()` after a failed TX message
    - Two or more messages previously queued for transmission (via `twai_transmit()`) in addition to the one that jsut failed, thus `tx_msg_count >= 3`.
    - The failed TX has made TEC is near to 256 (bus-off condition). The next failed TX message will cause a bus-off
2. `twai_handle_tx_buffer_frame()` handles the just failed and loads the next queued TX message
    - `tx_msg_count--;`, thus `tx_msg_count >= 2`.
    - `twai_hal_set_tx_buffer_and_transmit()` starts the next TX message, which sets the `TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED` flag
3. The TX has an error thus incrementing the TEC and causing a bus-off interrupt. We arrive at `twai_intr_handler_main() -> twai_hal_get_events() -> twai_hal_decode_interrupt()`
    - Bus-off condition detected, `TWAI_HAL_STATE_FLAG_BUS_OFF` is set and a `TWAI_HAL_EVENT_BUS_OFF` is generated.
    - TX done condition detected, `TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED` is cleared and `TWAI_HAL_EVENT_TX_BUFF_FREE` is generated
4. We return back to the main interrupt `twai_intr_handler_main()` 
    - `twai_handle_tx_buffer_frame()` handles the `TWAI_HAL_EVENT_TX_BUFF_FREE` event
        - Handles the failed TX message that caused the bus-off. `tx_msg_count--;`, thus `tx_msg_count >= 1`
        - **Doesn't check for bus-off and attempts to load another queued TX message for transmission because `tx_msg_count >= 1`**. Calls `twai_hal_set_tx_buffer_and_transmit()` which sets the `TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED` again. The underlying hardware won't actually transmit anything because the bus-off state
    - `TWAI_HAL_EVENT_BUS_OFF` is then handled, generating a `TWAI_ALERT_BUS_OFF` alert
5. User code receives the `TWAI_ALERT_BUS_OFF` alert, and calls `twai_initiate_recovery()` to start bus recovery
    - **Reset `tx_msg_count = 0`**
    - **`twai_hal_start_bus_recovery()` doesn't clear the `TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED` that was mistakenly set in the previous interrupt**
6. Bus recovery continues until completion, generating another interrupt. We arrive at We arrive at `twai_intr_handler_main() -> twai_hal_get_events() -> twai_hal_decode_interrupt()` again.
    - Bus off recovery complete detected. Generate `TWAI_HAL_EVENT_BUS_RECOV_CPLT` event
    - **`CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST` workaround sees that `TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED` is set (from step 4), thus generates a `TWAI_HAL_EVENT_TX_BUFF_FREE` event even if there is no completed TX.**
7. We return back to the main interrupt `twai_intr_handler_main()`
    - `twai_handle_tx_buffer_frame()` handles the `TWAI_HAL_EVENT_TX_BUFF_FREE` event
        - **`tx_msg_count` is already `0` as it was cleared in step 5**
        - `tx_msg_count--`, thus is now `-1`
        - `assert(tx_msg_count >= 0);` fails, leading to a crash

</details>

## Proposed Fix

The fix for the issue would simply be to update `twai_handle_tx_buffer_frame()` so that it doesn't attempt to load another message for transmission if in a bus-off state (even if `tx_msg_count > 0`). So something like:

```c
static inline void twai_handle_tx_buffer_frame(BaseType_t *task_woken, int *alert_req)
{
    //Handle previously transmitted frame
    ...

    //Update TX message count
    p_twai_obj->tx_msg_count--;
    assert(p_twai_obj->tx_msg_count >= 0);      //Sanity check

    //If not bus-off, check if there are more frames to transmit
    if (!bus_off && p_twai_obj->tx_msg_count > 0 && p_twai_obj->tx_queue != NULL) {
        ...
    } else {
        //No more frames to transmit
        twai_alert_handler(TWAI_ALERT_TX_IDLE, alert_req);
    }
}
```

However, for the sake of completeness, it might also be a good idea to address...

- `twai_hal_start_bus_recovery()` not clearing the `TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED`
- guarding against `twai_transmit()` calls during a bus-off/recovery state

Dazza0, 2025-03-15:
@diplfranzhoepfinger Pushed the proposed to [this branch](https://github.com/Dazza0/esp-idf/tree/bugfix/twai_transmit_after_bus_off).

@wanckl PTAL

wanckl, 2025-03-17:
@Dazza0 Thank you very much darain 😭

closed 2025-03-24 by espressif-bot

boborjan2, 2025-04-15:
Hi, is there a reason why it is not merged in 5.3 and 5.4 as well? Thx, Viktor.

diplfranzhoepfinger, 2025-04-15:
> Hi, is there a reason why it is not merged in 5.3 and 5.4 as well? Thx, Viktor.

Agree, Valid Question !

Alvin1Zhang, 2025-04-16:
@boborjan2 @diplfranzhoepfinger Thanks for the question, the fixes on release/5.3 and release/5.4 have been merged internally, and will be available on the next sync to GitHub, thanks.