espressif/esp-idf #9698: ERROR: Could not find a version that satisfies the requirement pyparsing>=3.0.3

hzhh110, 2022-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### What is the expected behavior?

ERROR: Could not find a version that satisfies the requirement pyparsing>=3.0.3
ERROR: Could not find a version that satisfies the requirement pyparsing>=3.0.3 (from -c /home/huangzhenhui/.espressif/espidf.constraints.v5.1.txt (line 20)) (from versions: 1.4.6, 1.4.7, 1.4.8, 1.4.11, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 1.5.4, 1.5.5, 1.5.6, 1.5.7, 2.0.0, 2.0.1, 2.0.2, 2.0.3, 2.0.4, 2.0.5, 2.0.6, 2.0.7, 2.1.0, 2.1.1, 2.1.2, 2.1.3, 2.1.4, 2.1.5, 2.1.6, 2.1.7, 2.1.8, 2.1.9, 2.1.10, 2.2.0, 2.2.1, 2.2.2, 2.3.0, 2.3.1, 2.4.0, 2.4.1.1, 2.4.2a1, 2.4.2, 2.4.3, 2.4.4, 2.4.5, 2.4.6, 2.4.7)
ERROR: No matching distribution found for pyparsing>=3.0.3 (from -c /home/huangzhenhui/.espressif/espidf.constraints.v5.1.txt (line 20))

### What is the actual behavior?

pip -V
pip 20.3.4 from /home/huangzhenhui/.local/lib/python2.7/site-packages/pip (python 2.7)
pip3 -V
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6
pip3 install pyparsing
Collecting pyparsing
  Using cached https://files.pythonhosted.org/packages/6c/10/a7d0fa5baea8fe7b50f448ab742f26f52b80bfca85ac2be9d35cdd9a3246/pyparsing-3.0.9-py3-none-any.whl
Installing collected packages: pyparsing
Successfully installed pyparsing-3.0.9

pip install pyparsing
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyparsing in /home/huangzhenhui/.local/lib/python2.7/site-packages (2.4.7)

### Steps to reproduce.

ERROR: Could not find a version that satisfies the requirement pyparsing>=3.0.3 (from -c /home/huangzhenhui/.espressif/espidf.constraints.v5.1.txt (line 20)) (from versions: 1.4.6, 1.4.7, 1.4.8, 1.4.11, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 1.5.4, 1.5.5, 1.5.6, 1.5.7, 2.0.0, 2.0.1, 2.0.2, 2.0.3, 2.0.4, 2.0.5, 2.0.6, 2.0.7, 2.1.0, 2.1.1, 2.1.2, 2.1.3, 2.1.4, 2.1.5, 2.1.6, 2.1.7, 2.1.8, 2.1.9, 2.1.10, 2.2.0, 2.2.1, 2.2.2, 2.3.0, 2.3.1, 2.4.0, 2.4.1.1, 2.4.2a1, 2.4.2, 2.4.3, 2.4.4, 2.4.5, 2.4.6, 2.4.7)
ERROR: No matching distribution found for pyparsing>=3.0.3 (from -c /home/huangzhenhui/.espressif/espidf.constraints.v5.1.txt (line 20))

### Build or installation Logs.

```plain
1
```

### More Information.

1

dobairoland, 2022-09-02:
May I ask how are you trying to install ESP-IDF? I'm asking because if you followed the official guide then you should have received a message that you have available only unsupported versions of Python.

```
pip 20.3.4 from /home/huangzhenhui/.local/lib/python2.7/site-packages/pip (python 2.7)
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6
```

Neither 2.7 nor 3.6 are supported. The Python community dropped a while ago and packages are not available for them.

For recent ESP-IDF versions only Python 3.7 and newer are supported.

dobairoland, 2022-09-14:
I'm closing this for inactivity and under the assumption that this is caused by a broken system.

closed 2022-09-14 by dobairoland