espressif/esp-idf #9699: Please tell me heap_caps_malloc API is reentrancy or not?

WangShuoran, 2022-09-02:
### Is your feature request related to a problem?

When I watch the manual (https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/system/mem_alloc.html).
I don't know the malloc/heap_caps_malloc is reentrancy or not.
But I think it is very impontant in freertos task. I don't find any explain in docs.
So I want to know the malloc/heap_caps_malloc is or not reentrancy.

ESP-Marius, 2022-09-02:
There is a section that mention it here:

https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/system/mem_alloc.html#thread-safety

ESP-Marius, 2022-09-02:
I've also pushed an internal MR to make this info more visible, instead of hiding it in the middle of the API documentation. So hopefully it should be easier to find this information in the future.

WangShuoran, 2022-09-02:
Thank you! You've really helped me a lot.

WangShuoran, 2022-09-03:
@ESP-Marius . Hello, I have a question about reentrancy and thread safey.

When I search reentrancy and thread safey different. I see a blog that is https://blog.csdn.net/acs713/article/details/20034511.

So I think malloc/heap_caps_malloc is cool of the thread safey. But reentrancy?

ESP-Marius, 2022-09-05:
It is also reentrant.

closed 2022-09-05 by WangShuoran

WangShuoran, 2022-09-06:
@ESP-Marius Hello. I think you can add "reentrant" detail in docs. Because I think https://www.ibm.com/docs/en/aix/7.2?topic=programming-writing-reentrant-threadsafe-code tell me the reentrant is different with threadsafe.

igrr, 2022-09-06:
@WangShuoran could you explain what difference does reentrant vs thread-safe makes in this case for the application developer? In other words, what would you have to do differently in your application code if the function was only thread-safe but not reentrant?

WangShuoran, 2022-09-08:
@igrr 
我不是一个很熟练使用英文的人，我用中文回答，用英文翻译吧
我只是说这件事，并不代表言辞锋利。请看内容，不必付诸感情
我是一个RTOS新手，对多线程概念仅仅有个大概认识：

- 重入代表函数被要求不能使用外部的数据
- 而线程安全代表不用锁也可以使用这个API

在malloc这里，因为不存在指针，所以也不需要考虑重入，但是不能认为它不重要，我想如果补充文档，也是更严谨些，优秀的文档不应该把使用者当作大神，而应该把使用着当作菜鸟。

还有，我认为比如memcpy这些使用指针的，就要考虑重入。
多问几个问题，我看Espressif的工具链头文件的时候，看到是有一些newlib库的头文件，所以乐鑫libc是使用的newlib吗？
newlib中的标准库实现大多可以重入(百度百科这样介绍)，那么如果我使用一个标准库，或者查找一个标准库函数是否线程安全或可重入，那么我应该去newlib的官网https://sourceware.org/newlib/libc.html去查找吗？

I'm not a very proficient English speaker. I'll answer in Chinese and translate in English

Just because I said this does not mean that my words are sharp. Look at the content, don't put your feelings into it

I am a newcomer to RTOS, and I only have a general understanding of the concept of multithreading:

-Reentry delegate function is required not to use external data

-Thread safety representatives can use this API without locks

In malloc, because there is no pointer, there is no need to consider reentry, but it can't be considered unimportant. I think if you supplement the document, it will be more rigorous. Excellent documents should not treat users as gods, but should treat users as novices.

Also, I think memcpy, which uses pointers, should consider reentry.

Ask a few more questions. When I looked at the tool chain header file of espressif, I saw that there were some newlib library header files, so is Lexin libc using newlib?

Most of the standard library implementations in newlib can be reentrant (as Baidu Baike introduced). If I use a standard library or find out whether a standard library function is thread safe or reentrant, then I should go to the official website of newlib https://sourceware.org/newlib/libc.html Do you want to look it up?

WangShuoran, 2022-09-08:
https://github.com/espressif/esp-idf/issues/1997
我看到了这个问题，我看了一些i2c_master_write实现源码，我认为虽然这个API是非线程安全的，但它是可重入。因为就算是它在执行过程中被打断，只要传入的第一个参数不会被其他线程使用，那么依然执行起来是正确的。所以只要保证第一个参数所使用的句柄在系统中只有一个线程使用，那么就是正确的。

caidamao, 2023-11-30:
如果函数是可重入的，那其实就应该是线程安全的。可重入函数一定是线程安全的，但线程安全不一定是可重入的。

可重入函数可以在任何时候被中断，一段时间后又可以运行，而相应的数据不会丢失。而线程安全函数能够使不同的线程访问同一块地址空间。二者的区别在于，可重入函数强调不同执行流对数据的操作互不影响，而且结果相同，但是线程安全允许在调用该函数时，出现互相影响的情况，只要使用互斥锁保证安全性即可。