espressif/esp-idf #9703: 关于重连wifi时不能扫描wifi的问题

Szeroy, 2022-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

当密码错误重连的时候可以扫描wifi，或者有没有回调函数的标志位知道当前在连接wifi，这样我就可以断连之后再去扫描，我使用了esp_wifi_disconnect之后再去扫描wifi也是没用的，同样死机

### What is the actual behavior?

我设置了连接wifi重试十五次，如果当用户输入wifi密码错误时，esp32去连接wifi，但是因为密码错误是一直连不上的，当这时用户又去扫描wifi，mcu就死机了。

### Steps to reproduce.

```
static char wifi_scan(wifi_ap_record_t *ap_info, uint16_t enum_number)
{
    uint16_t ap_count = 0;

    if(g_network_state == 2)
        esp_wifi_disconnect();

    ESP_ERROR_CHECK(esp_wifi_scan_start(NULL, true));                     // 开始扫描WIFI(默认配置，阻塞方式)
    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&enum_number, ap_info)); // 获取搜索的具体AP信息
    ESP_ERROR_CHECK(esp_wifi_scan_get_ap_num(&ap_count));                 //接入点的数量
    ESP_LOGI(TAG, "Total APs scanned = %u", ap_count);
    for (int i = 0; (i < enum_number) && (i < ap_count); i++)
    {
        ESP_LOGI(TAG, "SSID \t\t%s", ap_info[i].ssid); // 打印WIFI名称
        ESP_LOGI(TAG, "RSSI \t\t%d", ap_info[i].rssi); // 打印信号强度
        ESP_LOGI(TAG, "Channel \t\t%d\n", ap_info[i].primary);
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
    return 1;
}

void wifi_scan_task(void *parm)
{
    uint16_t ap_count = 0;
    scan_semaphore_handle = xSemaphoreCreateBinary();
    while (1)
    {
        xSemaphoreTake(scan_semaphore_handle,portMAX_DELAY);
        // if(get_network_state() == 2)
        // {
        //     ESP_ERROR_CHECK(esp_wifi_disconnect());
        // }
        scan_flag = 1;
        memset(ap_info_buf,0,sizeof(ap_info_buf));
        if(wifi_scan(ap_info_buf,DEFAULT_SCAN_LIST_SIZE))
        {
            esp_wifi_scan_get_ap_num(&ap_count);                 //接入点的数量
            if(ap_count <= DEFAULT_SCAN_LIST_SIZE)
                refrash_wifi_name(ap_info_buf, ap_count);
            else
                refrash_wifi_name(ap_info_buf, DEFAULT_SCAN_LIST_SIZE);
        }
        scan_flag = 0;
        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
}
```

### Debug Logs.

```plain
I (89215) smartconfig_example: Got SSID and password
I (89215) smartconfig_example: SSID:XAZN_84:f7:03:d8:9c:ce
I (89215) smartconfig_example: PASSWORD:edff
I (89225) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (89225) wifi:state: init -> auth (b0)
I (89235) wifi:state: auth -> assoc (0)
I (89245) wifi:state: assoc -> run (10)
I (89285) smartconfig_example: Total APs scanned = 22
I (89755) Sensor: ff 01 90 01 03 cc 02 83 00 dc 00 00 00
I (89755) Sensor: CO2 = 400,temp = 30.3,humi = 52.3,TVOC = 220,HCHO = 0,PM2.5 = 0.00
ui_WifiNameText =
W (91035) wifi:sta_scan: STA is connecting, scan are not allowed!
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4037d5bc
0x4037d5bc: _esp_error_check_failed at D:/esp-idf/esp-idf-4.4v/components/esp_system/esp_err.c:42

file: "./main/netConfig/netConfig.c" line 130
func: wifi_scan
expression: esp_wifi_scan_start(NULL, true)

abort() was called at PC 0x4037d5bf on core 0
0x4037d5bf: _esp_error_check_failed at D:/esp-idf/esp-idf-4.4v/components/esp_system/esp_err.c:43

Backtrace:0x40375c62:0x3fcca1400x4037d5c9:0x3fcca160 0x40384896:0x3fcca180 0x4037d5bf:0x3fcca1f0 0x4200888f:0x3fcca210 0x420089ae:0x3fcca250 0x40380811:0x3fcca280
0x40375c62: panic_abort at D:/esp-idf/esp-idf-4.4v/components/esp_system/panic.c:402

0x4037d5c9: esp_system_abort at D:/esp-idf/esp-idf-4.4v/components/esp_system/esp_system.c:128

0x40384896: abort at D:/esp-idf/esp-idf-4.4v/components/newlib/abort.c:46

0x4037d5bf: _esp_error_check_failed at D:/esp-idf/esp-idf-4.4v/components/esp_system/esp_err.c:43

0x4200888f: wifi_scan at D:/WorkProject/WorkProject/OutstandingProject/Hand_held_air_detector/main/netConfig/netConfig.c:130 (discriminator 1)

0x420089ae: wifi_scan_task at D:/WorkProject/WorkProject/OutstandingProject/Hand_held_air_detector/main/netConfig/netConfig.c:158

0x40380811: vPortTaskWrapper at D:/esp-idf/esp-idf-4.4v/components/freertos/port/xtensa/port.c:131
```

zhangyanjiaoesp, 2022-09-07:
@Szeroy 
1. 目前没有 API 或者回调函数的标志位可以判断 wifi 是否正在连接
2. > 设置了连接wifi重试十五次，

你是怎么设置重连操作的？是在每次 disconnect event 发生的时候调用 `esp_wifi_connect()` 吗？如果是这样的话可以在调用 connect 之前先进行 scan 操作
3. `ESP_ERROR_CHECK()` 会判断 API 的返回值，如果不是 ESP_OK，就会强制 assert，即出现你提供的 log 里面的程序挂掉的情况。你可以在代码里面不加这个判断，直接 `int ret = esp_wifi_scan_start(NULL, false)`, 然后判断 ret 的值是不是 ESP_OK，再进行相应的操作。这样可以保证在 scan fail 的情况下程序还能正常运行。

AxelLin, 2022-12-24:
> @Szeroy
> 
> 1. 目前没有 API 或者回调函数的标志位可以判断 wifi 是否正在连接

It would be helpful to add such API.

closed 2023-08-29 by Szeroy