espressif/esp-idf #9704: when I use nimble example bleprph, it will assert in ble_svc_gap_init function

luacerbb, 2022-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.1.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-DevKitC-V4_

### Power Supply used.

USB

### What is the expected behavior?

example can run right

### What is the actual behavior?

assertion "rc == 0" failed: file "/mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/bt/host/nimble/nimble/nimble/host/services/gap/src/ble_svc_gap.c", line 297, function: ble_svc_gap_init
abort() was called at PC 0x401159c3 on core 0
0x401159c3: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: b6c1c8451b07a9f2

Backtrace: 0x4008ebc1:0x3ffbb590 0x4008ef65:0x3ffbb5b0 0x401159c3:0x3ffbb5d0 0x400d5f97:0x3ffbb600 0x400d5ab9:0x3ffbb620 0x400d5972:0x3ffbb640 0x400d1bf2:0x3ffbb660 0x4009029d:0x3ffbb680
0x4008ebc1: invoke_abort at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/esp32/panic.c:157

0x4008ef65: abort at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/esp32/panic.c:174

0x401159c3: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400d5f97: ble_svc_gap_init at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/bt/host/nimble/nimble/nimble/host/services/gap/src/ble_svc_gap.c:297 (discriminator 1)

0x400d5ab9: gatt_svr_init at /mnt/hgfs/test/bleprph/build/../main/gatt_svr.c:196

0x400d5972: app_main at /mnt/hgfs/test/bleprph/build/../main/main.c:372 (discriminator 2)

0x400d1bf2: main_task at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/esp32/cpu_start.c:597 (discriminator 2)

0x4009029d: vPortTaskWrapper at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/freertos/port.c:143

### Steps to reproduce.

1、build the nimble example,bleprph
2、flash to  esp32 devc board
3、 start and run assert
log:
I (12) boot: ESP-IDF v4.1.3-dirty 2nd stage bootloader
I (12) boot: compile time 09:47:29
I (12) boot: chip revision: 1
I (14) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (21) boot.esp32: SPI Speed      : 40MHz
I (26) boot.esp32: SPI Mode       : DIO
I (30) boot.esp32: SPI Flash Size : 4MB
I (35) boot: Enabling RNG early entropy source...
I (40) boot: Partition Table:
[... 129 lines cut ...]
test
svr init
assertion "rc == 0" failed: file "/mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/bt/host/nimble/nimble/nimble/host/services/gap/src/ble_svc_gap.c", line 297, function: ble_svc_gap_init
abort() was called at PC 0x401159c3 on core 0
0x401159c3: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: b6c1c8451b07a9f2

Backtrace: 0x4008ebc1:0x3ffbb590 0x4008ef65:0x3ffbb5b0 0x401159c3:0x3ffbb5d0 0x400d5f97:0x3ffbb600 0x400d5ab9:0x3ffbb620 0x400d5972:0x3ffbb640 0x400d1bf2:0x3ffbb660 0x4009029d:0x3ffbb680
0x4008ebc1: invoke_abort at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/esp32/panic.c:157

0x4008ef65: abort at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/esp32/panic.c:174

0x401159c3: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400d5f97: ble_svc_gap_init at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/bt/host/nimble/nimble/nimble/host/services/gap/src/ble_svc_gap.c:297 (discriminator 1)

0x400d5ab9: gatt_svr_init at /mnt/hgfs/test/bleprph/build/../main/gatt_svr.c:196

0x400d5972: app_main at /mnt/hgfs/test/bleprph/build/../main/main.c:372 (discriminator 2)

0x400d1bf2: main_task at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/esp32/cpu_start.c:597 (discriminator 2)

0x4009029d: vPortTaskWrapper at /mnt/hgfs/Project/gate_code/esp-idf-v4.1.3/components/freertos/port.c:143
```

rahult-github, 2022-09-02:
Hi @luacerbb ,

  I quickly tried on mentioned tag (v4.3.1) and the application is executing normally. Attached console log for same. 
   
  Can you please ensure you have a clean build ? Please share the sdkconfig of your setup. 

  Also, it looks like failure happens during registration of ble_svc_gap_defs .  Kindly confirm, if there are any other modifications done, that can give us better understanding of the issue. 

Thanks. 

[console_log.txt](https://github.com/espressif/esp-idf/files/9479467/console_log.txt)

rahult-github, 2022-09-02:
Hi @luacerbb , kindly discard the above request. I will update on this asap.

rahult-github, 2022-09-02:
Hi @luacerbb  ,      Please check if attached patch helps. 

[0001-Nimble-Check-stack-initialization-status-before-exec.txt](https://github.com/espressif/esp-idf/files/9479622/0001-Nimble-Check-stack-initialization-status-before-exec.txt)

luacerbb, 2022-09-05:
thanks,when I attach patch,it work well.

closed 2022-09-05 by luacerbb