espressif/esp-idf #9705: ADC readings on idf-5.0+ are offset by -30mV compared to IDF-4.4

pavlohamov, 2022-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release5.0/master VS release4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

esp32s2-saola

### Power Supply used.

USB

### What is the expected behavior?

128 mV are sourced to GPIO_3 (ADC1_CH2) ADC conversion vis calibration results in 128mV

### What is the actual behavior?

128 mV are sourced to GPIO_3 (ADC1_CH2) ADC conversion vis calibration results in 98mV on idf5.0

### Steps to reproduce.

1. Source 128 mV to GPIO_3 (ADC1_CH2) (e.g,. voltage divider 10K POT)
2. build and execute IDF-4.4 examples/peripherals/adc/single_read/single_read
3. build and execute IDF-5.0+ (or master 91e8312) examples/peripherals/adc/single_read/single_read
4. Observe that IDF 5.0 readings have -30 mV offset compared to idf4.4

6. Step
7. Step
...

### Debug Logs.

```plain
IDF-4.4
I (261) cpu_start: Starting scheduler on PRO CPU.
I (267) ADC1_CH2: raw  data: 401
I (267) ADC1_CH2: cali data: 129 mV
I (2267) ADC1_CH2: raw  data: 404
I (2267) ADC1_CH2: cali data: 130 mV
I (4267) ADC1_CH2: raw  data: 397
I (4267) ADC1_CH2: cali data: 128 mV
I (6267) ADC1_CH2: raw  data: 404
I (6267) ADC1_CH2: cali data: 130 mV
I (8267) ADC1_CH2: raw  data: 401
I (8267) ADC1_CH2: cali data: 129 mV
I (10267) ADC1_CH2: raw  data: 400
I (10267) ADC1_CH2: cali data: 128 mV
I (12267) ADC1_CH2: raw  data: 400
I (12267) ADC1_CH2: cali data: 128 mV
I (14267) ADC1_CH2: raw  data: 393
I (14267) ADC1_CH2: cali data: 126 mV

IDF-5.0
I (264) cpu_start: Starting scheduler on PRO CPU.
I (269) EXAMPLE: calibration scheme version is Line Fitting
I (269) EXAMPLE: Calibration Success
I (279) EXAMPLE: calibration scheme version is Line Fitting
I (279) EXAMPLE: Calibration Success
I (289) EXAMPLE: ADC1 Channel[2] Raw Data: 329
I (289) EXAMPLE: ADC1 Channel[2] Cali Voltage: 106 mV
I (3299) EXAMPLE: ADC1 Channel[2] Raw Data: 328
I (3299) EXAMPLE: ADC1 Channel[2] Cali Voltage: 105 mV
I (6299) EXAMPLE: ADC1 Channel[2] Raw Data: 329
I (6299) EXAMPLE: ADC1 Channel[2] Cali Voltage: 106 mV
I (9299) EXAMPLE: ADC1 Channel[2] Raw Data: 324
I (9299) EXAMPLE: ADC1 Channel[2] Cali Voltage: 104 mV
```

### More Information.

128 mV are sourced to GPIO_3 (ADC1_CH2) additional 100uF capacitor (aluminium) is soldered to GPIO (leads ~ 2cm)
ADC readings differs much for latest idf.

Also I find out that it depends on HW calibration data. it is reproduced only for devices that reports "With ADC calibration V2"
One of IC's has readings calibration table  
```
=== Run "adc_info" command ===

Temperature Sensor Calibration = -4.5C

ADC1 readings stored in efuse BLOCK2:
    MODE0 D1 reading  (250mV):  -56
    MODE0 D2 reading  (600mV):  -44
    MODE1 D1 reading  (250mV):  -44
    MODE1 D2 reading  (800mV):  -36
    MODE2 D1 reading  (250mV):  -36
    MODE2 D2 reading  (1000mV): -24
    MODE3 D1 reading  (250mV):  -20
    MODE3 D2 reading  (2000mV): -8

ADC2 readings stored in efuse BLOCK2:
    MODE0 D1 reading  (250mV):  -60
    MODE0 D2 reading  (600mV):  -76
    MODE1 D1 reading  (250mV):  -44
    MODE1 D2 reading  (800mV):  -68
    MODE2 D1 reading  (250mV):  -36
    MODE2 D2 reading  (1000mV): -56
    MODE3 D1 reading  (250mV):  -20
    MODE3 D2 reading  (2000mV): -56
```

other one just:
```
=== Run "adc_info" command ===

BLOCK2_VERSION = With ADC calibration V2
```

Device with ADC readings table provides proper readings (offset is couple of mV). This issue is crucial for our product, a lot of production devices are affected by this issue.

pavlohamov, 2022-09-05:
@Icarus113 I've managed to find the root case by my own. Please take a look to PR https://github.com/espressif/esp-idf/pull/9715

Icarus113, 2022-09-06:
@pavlohamov  Hi, thanks for this. I checked your PR, which is a correct fix. Will deal with your PR.

closed 2022-09-12 by espressif-bot