espressif/esp-idf #9710: Is there a way to load dynamic library with esp-idf?

lanistor, 2022-09-03:


chipweinberger, 2022-09-03:
The closest thing you will find is Micropython

https://docs.micropython.org/en/latest/esp32/tutorial/intro.html

chegewara, 2022-09-03:
https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake

lanistor, 2022-09-05:
> The closest thing you will find is Micropython
> 
> https://docs.micropython.org/en/latest/esp32/tutorial/intro.html

Thanks a lot, it seems a way to flash bins on different address, then how to load a bin dynamiclly in another?

0xjakob, 2022-09-22:
@lanistor In IDF, we currently don't publicly support this. We have an internal solution that hasn't been published yet. Please contact the Espressif business support for that.

There are also some third-party projects, e.g. https://github.com/joltwallet/elfloader.

If you are able to switch to NuttX on ESP32, then you can use dynamically loadable libraries, as NuttX supports this.

0xjakob, 2022-11-10:
@lanistor There hasn't been any activity on this issue for some time, so we assume it can be closed. If you still have questions regarding the original issue, please feel free to re-open the issue.

closed 2022-11-10 by 0xjakob

chipweinberger, 2022-11-10:
There is a new company working on dynamic apps on esp32: **Toit**

"Live Reloading of Micro-Services on Your ESP32"
https://www.youtube.com/watch?v=2jWVfiibhmc

If you need dynamic code, I'd highly recommend them. Ex-Google engineer.

FranckFreiburger, 2024-05-03:
> ... we currently don't publicly support this. We have an internal solution that hasn't been published yet.

Does the word “yet” suggest that this API will be public one day?

Even after many size optimizations, I run out of flash and my project lends itself well to this kind of shared object breaking down.

0xjakob, 2024-05-03:
@FranckFreiburger I'm not sure if dynamic library loading would help in your case. The flash is usually much larger than the internal memory of our chips. And unless you're loading the code from somewhere else, the additional code needs to be stored in flash anyway.

Does your project contain mostly code or also a considerable amount of data (e.g., UI graphics)? If yes, you might want to check additional storage options available in IDF, such as [SDcards](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/sdmmc.html).