espressif/esp-idf #9712: Chapter #gpio-matrix-and-io-mux seems missing for ESP32S3

franz-ms-muc, 2022-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux

https://github.com/espressif/esp-idf/blob/91e8312f03/docs/en/api-reference/peripherals/spi_master.rst

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_master.html

according these Documents: 

the Chapter #gpio-matrix-and-io-mux seems missing for ESP32S3 right ?

Icarus113, 2022-09-05:
@franz-ms-muc  Hi, thanks for this report. We'll update this doc recently.

wanckl, 2022-10-27:
Hi, documentations you need have updated, welcome to discuss if still anything confused. thank you!
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux

closed 2023-01-12 by Alvin1Zhang