espressif/esp-idf #9713: The OTA error is in ESP32-S3-WROOM-1.

lingdangsmoke, 2022-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.3.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I want to OTA my board.  My board is all ready for OTA, but the log show the error,"**Mismatch chip id, expected 9, found 4**",when the load from server.

### What is the actual behavior?

OTA my board. used OTA function in esp_https_ota.h 。

### Steps to reproduce.

1. Make the ESP32-S3-WROOM-1 function of OTA.
2. build the xxx.bin and show the server.
3. ESP32-S3-WROOM-1 used OTA.

### Debug Logs.

```plain
[0;32mI (2105) esp_https_ota: Starting OTA...[0m
[0;32mI (2105) esp_https_ota: Writing to partition subtype 17 at offset 0x210000[0m
[0;32mI (2105) ESP32_OTA: Running firmware version: 1[0m
[0;31mE (2115) esp_https_ota: Mismatch chip id, expected 9, found 4[0m
[0;31mE (2115) ESP32_OTA: other err:266[0m
[0;31mE (2125) ESP32_OTA: OTA_perform_error[0m
[0;31mE (2125) ESP32_OTA: Complete data was not received.[0m
[0;31mE (2135) ESP32_OTA: ESP_HTTPS_OTA upgrade failed[0m
```

### More Information.

no more things.

chipweinberger, 2022-09-04:
v4.3.1 is pretty old. I'd recommend upgrading to v4.4.2 for better S3 support.

lingdangsmoke, 2022-09-04:
> v4.3.1 已经很旧了。我建议升级到 v4.4.2 以获得更好的 S3 支持。

Thanks for answer my question. This issue has  been  sloved  and  i will be update my esp-idf to v4.4.2.

closed 2022-09-04 by lingdangsmoke