espressif/esp-idf #9714: Modbus tcp enc28j60

Silvesterrr, 2022-09-04:
Hi, Im trying to setup Modbus TCP Slave on my esp32s3 using enc28j60 module. 
Ive combined enc28j60 example with modbus slave tcp example. 
Device gets address from dhcp without any problem.
.
.
When I want to read some data from device I do not get any response. Console clearly indicates that it recieved frame and accepted client communication but nothing else happends.
this is log on eth starup:
```
I (3563) eth_example: Ethernet Got IP Address
I (3573) eth_example: ~~~~~~~~~~~
I (3573) eth_example: ETHIP:192.168.1.34
I (3583) eth_example: ETHMASK:255.255.255.0
I (3583) eth_example: ETHGW:192.168.1.1
I (3593) eth_example: ~~~~~~~~~~~
D (5673) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.
D (5673) MB_PORT_COMMON: vMBPortSetMode: Port exit critical
W (5673) eth_example: Input reg. set to 000000
I (5673) MB_TCP_SLAVE_PORT: Protocol stack initialized.
D (5683) MB_PORT_COMMON: TCP Slave port enable.
I (5683) MB_TCP_SLAVE_PORT: Socket (#54), listener 192.168.1.34 on port: 502, errno=0
```
.
As you can see there seems to be no problem but after trying to communicate with device I do not get any response.
[image]
.
.
My console looks like this while trying to commuincate:
```
 (165093) MB_TCP_SLAVE_PORT: Socket (#55), accept client connection from address: 192.168.1.12
D (165093) MB_TCP_SLAVE_PORT: | TID = 0001 | PID = 0000 | LEN = 0006 | UID = 01 | FUNC = 04 | DATA = 00000001 |
D (165103) MB_TCP_SLAVE_PORT: Socket (#55)(192.168.1.12), get packet TID=0x1, 12 bytes.
D (165103) MB_PORT_COMMON: EV_FRAME_RECEIVED
D (165113) MB_PORT_COMMON: 04 00 00 00 01 
D (165123) MB_PORT_COMMON: eMBPoll:EV_EXECUTE
D (165123) MB_TCP_SLAVE_PORT: Client 0, Socket(#55), processing time = 22852 (us)
```
.
.
So it clearly recives frame but it doesn't do anything with it.
next data recieve attempt just drops the connection:
`E (163033) MB_TCP_SLAVE_PORT: Socket (#55)(192.168.1.12), connection closed by peer
`
Here is my Code:
```
#include <stdio.h>
#include <string.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_netif.h"
#include "esp_eth.h"
#include "esp_event.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "driver/spi_master.h"
#include "eth_enc28j60/esp_eth_enc28j60.h"

#include "mbcontroller.h"
void modbus_tcp(void);
void modbus_slave_task(void *pvParameter);

static const char *TAG = "eth_example";

/** Event handler for Ethernet events */
static void eth_event_handler(void *arg, esp_event_base_t event_base,
                              int32_t event_id, void *event_data)
{
    uint8_t mac_addr[6] = {0};
    /* we can get the ethernet driver handle from event data */
    esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;
[... 173 lines cut ...]
        if (event & MB_EVENT_INPUT_REG_RD)
        {
            rc += mbc_slave_get_param_info(&reg_info, MB_PAR_INFO_GET_TIME_OUT);
            ESP_LOGI(TAG, "INPUT READ:, ADDR:%u, SIZE:%u",
                     (uint32_t)reg_info.address,
                     (uint32_t)reg_info.size);

            /* discrete register read */
        }
    }
    // Destroy of Modbus controller on alarm
    ESP_LOGE(TAG, "Modbus controller destroyed.");
    vTaskDelay(100);
}
void app_main(void)
{
    init_eth();
    vTaskDelay(pdMS_TO_TICKS(5000));
    modbus_tcp();
    modbus_slave_task(NULL);
}
```
.
.
I don't know is it a some sort of bug, or my basic misconception, or did I missed something important?

alisitsyn, 2022-09-05:
Hi @Silvesterrr ,

> So it clearly recives frame but it doesn't do anything with it.

From quick review of the code and logs I can say that the physical layer works and the slave Modbus stack was able to receive your command to read input register from address 0x00. The stack executed the handler for this command successfully and returned exception report back because you configured your example a bit incorrectly. The error is in `setup_input_register(0x0, &test, 1);`- you try to configure the input register area which contains just one byte (see the [configure Modbus slave data access](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/slave_api_overview.html#configuring-slave-data-access))but in the master you read one register (2 bytes instead). This is a reason for the exception.

> next data recieve attempt just drops the connection:

In order to figure out the issue I need to know which library version you are using to follow with the dropped connection issue. It is recommended to use the [ESP-Modbus library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/modbus.html) . Unfortunately I can not check this with `ENC28J60` PHY now. I will try it as soon as possible but expect some delay.

Thanks.

Silvesterrr, 2022-09-05:
Thank you for the response. Surely it was my trivial mistake. Thank you for reply. With correct number of registers comunication works without any issue.

> he E (163033) MB_TCP_SLAVE_PORT: Socket (#55)(192.168.1.12), connection closed by peer - means the the master close the connection after send. I think it is expected behavior of your QModbus client.

Yes trying it with python library this thing does not occure.

> In order to figure out the issue

No need for futher investigation. Closing the issue.

closed 2022-09-05 by Silvesterrr

DishankJogi, 2023-06-19:
(165093) MB_TCP_SLAVE_PORT: Socket (#55), accept client connection from address: 192.168.1.12
D (165093) MB_TCP_SLAVE_PORT: | TID = 0001 | PID = 0000 | LEN = 0006 | UID = 01 | FUNC = 04 | DATA = 00000001 |
D (165103) MB_TCP_SLAVE_PORT: Socket (#55)(192.168.1.12), get packet TID=0x1, 12 bytes.
D (165103) MB_PORT_COMMON: EV_FRAME_RECEIVED
D (165113) MB_PORT_COMMON: 04 00 00 00 01 
D (165123) MB_PORT_COMMON: eMBPoll:EV_EXECUTE
D (165123) MB_TCP_SLAVE_PORT: Client 0, Socket(#55), processing time = 22852 (us) 

Hello Friends , Anybody know how we can print this Modbus TCP frame data like this " | TID = 0001 | PID = 0000 | LEN = 0006 | UID = 01 | FUNC = 04 | DATA = 00000001 |"  on the serial monitor ?  Let me elaborate i am using Modbus TCP slave example code.

Thanks.