espressif/esp-idf #9717: Classic Bluetooth HID need a API to change HID_DEV_MTU_SIZE

elmagnificogi, 2022-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

latest

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-Wroom

### Power Supply used.

USB

### What is the expected behavior?

Classic Bluetooth HID need a API to change HID_DEV_MTU_SIZE

### What is the actual behavior?

When i send a 363 bytes report , it show like that:
`BT_APPL: BTA_HdSendReport, report len (363) > MTU len (64), can't send report. Increase value of HID_DEV_MTU_SIZE to send larger reports[0m`

I know most HID APP just use a small report like 64. But sometime it need send a larger report for just a few times.
so i think you could make a API to change it ,maybe better

### Steps to reproduce.

send anything that larger than 64 bytes
`esp_bt_hid_device_send_report(ESP_HIDD_REPORT_TYPE_INTRDATA, 0xa1, 363, (uint8_t *)&reply);`

Wth-Esp, 2022-09-06:
hi @elmagnificogi 
 
I went through the HID profile and find that there were no such big scenarios that HID Device should support.
[image]

But HID host is assumed to support relatively large MTU like 672 bytes.

[image]

I suspect that it's valuable to enable a variable HID device MTU. BTW, your blogs on ESP32 and Bluetooth are very good.  Could you please leave me a message, maybe we can learn together?

Thanks
Tianhua Wei

elmagnificogi, 2022-09-06:
@Wth-Esp Thank you for your reply. I am a new ESP32 developer. Now i am working on simulating NS JoyCon or NS Pro Controller in ESP32. Normal hid funcs work well.
When i simulate the Amiibo transfer, i found NS could transfer 362 bytes in HID reporter. It just need exchange about 60 package that 362 bytes in a few time about 30s. After that it change to normal size that's why i need go change it's MTU size..

My contact is in blog or github page

boblane1, 2022-09-06:
Hi @elmagnificogi，

Thanks for your reporting. I suggest you can just change the macro HID_DEV_MTU_SIZE in your project, please remember to not exceed 672 bytes. If more customers want to change the HID device MTU, we will think about adding an API like what you suggested.