espressif/esp-idf #9718: Add date to bootloader header

josesimoes, 2022-09-05:
### Is your feature request related to a problem?

When building the bootloader only the TIME information is added. I believe it's valuable to have the date too.

### Describe the solution you'd like.

Simply changing [this code line ](https://github.com/espressif/esp-idf/blob/4eb98adf43c945e67db8eb6f89451e8c5964378a/components/bootloader_support/src/bootloader_init.c#L94) to:
`ESP_LOGI(TAG, "compile time " __DATE__ " " __TIME__);` 

### Additional context.

Can't see any reason for not including the date information to the build timestamp. And that makes it more complete.

o-marshmallow, 2022-09-06:
Hi @josesimoes,

You can already do this thanks to wrappers. In your project directory, create a bootloader component:
```
mkdir -p bootloader_components/my_boot_message
cd bootloader_components/my_boot_message
```
Then, create a `CMakeLists.txt` like:
```
idf_component_register(SRCS "boot_msg.c"
                       WHOLE_ARCHIVE)

target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--wrap=bootloader_print_banner")
```

Finally, the `boot_msg.c` file should look like this:
```
#include "esp_log.h"

static const char *TAG = "boot";

void __wrap_bootloader_print_banner(void)
{
    ESP_LOGI(TAG, "ESP-IDF %s 2nd stage bootloader", IDF_VER);
    ESP_LOGI(TAG, "compile time " __DATE__ " " __TIME__);
}
```

After compiling and running, the bootloader should now show:
```
I (30) boot: ESP-IDF v5.1-dev-636-g24c31bd7fb-dirty 2nd stage bootloader
I (30) boot: compile time Sep  6 2022 11:08:17
```

josesimoes, 2022-09-06:
Hey @o-marshmallow ! 

Interesting approach you're suggesting. Wasn't aware of that extension mechanism for bootloader.
I've implemented this as you've suggested and it's working.

Nevertheless, I keep my suggestion on adding the date to the official source, as (to me) it makes a more complete information if one can see the date and time of the build, not only the time.

Thanks for your help!

o-marshmallow, 2022-09-07:
Hi @josesimoes ,

I also personally think it makes sense to add the date to the bootloader log. I am going to create an internal merge request for it.

In the meanwhile, you can use the wrap workaround :smile:

closed 2022-09-12 by espressif-bot

josesimoes, 2022-11-10:
I was testing this with a build for an ESP32_C3 but it's failing with 

`E:/GitHub/esp-idf/components/bootloader_support/src/esp32c3/bootloader_esp32c3.c:78: undefined reference to `__wrap_bootloader_print_banner'`

Note that I have not changed anything and this is building and working just fine for builds for ESP32 and ESP32_S2.
I can see that the source file it's being picked up by the build as if I add a syntax error on the source file the compiler complains about it. 

Appreciate your help on this!

o-marshmallow, 2022-11-11:
Hi @josesimoes ,

This error sounds odd as there is no reference to such strong symbol in `bootloader_support ` components. I just tried compiling the `examples/build_systems/wrappers` which wraps `__wrap_bootloader_print_banner` for the C3 and it compiles fine.
I tried on the latest master commit.

Which version of IDF are you using? Which example are you compiling? Have you modified any IDF file? In short, how can I reproduce your issue?

josesimoes, 2022-11-11:
I'm building from the recent v4.4.3 branch.

Worth noting that there is no examples/build_systems/wrappers in this version...
Could this be related with the  WHOLE_ARCHIVE option that doesn't seem to exist in 4.4.3?
If yes,can you suggest a workaround for it?

o-marshmallow, 2022-11-11:
@josesimoes Oh okay thanks for the details!

Indeed, as v4.4.3 doesn't have the WHOLE_ARCHIVE option, when it comes to linking, the wrapper implementation is discarded. To fix this issue, you will have to manually tell the linker to **not** ignore tyour library containing your wrapper.

Let's take back the `wrappers` example, copy it to your v4.4.3 branch. Open the `CMakeLists.txt` located in `bootloader_components/my_boot_message`, remove the `WHOLE_ARCHIVE` option, and replace the former `target_link_libraries` function call to this:
```
target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--whole-archive"
                     $<TARGET_FILE:${COMPONENT_LIB}>
                    "-Wl,--no-whole-archive"
                    "-Wl,--wrap=bootloader_print_banner")
```
This will tell the linker that the library we are currently defining should be a "whole archive".

In the end, your file should look like this:
```
idf_component_register(SRCS "wrapper.c")

target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--whole-archive"
                     $<TARGET_FILE:${COMPONENT_LIB}>
                    "-Wl,--no-whole-archive"
                    "-Wl,--wrap=bootloader_print_banner")
```

josesimoes, 2022-11-11:
@o-marshmallow brillant! 👏🏻 🥳 
 That was it, it's now building fine! Thanks.

(Still wondering why this was building fine for the xtensa builds and this only manifests in the riscv one)